evcc-io/evcc #1660: What do I have to return using http Plugin?

doppelrittberger, 2021-09-28:
**Is your feature request related to a problem? Please describe.**
It is unclear what a http plugin should return as SoC if the current SoC cannot be read.

**Describe the solution you'd like**
A clear documentation on what to return as SoC value if the SoC reader is not ready yet.

**Additional context**
I use an Arduino and a ODB Bluetooth reader to read the SoC of my vehicle. The Arduino presents the value as JSON in form "{soc:12, range: 50}". Since the ODB reader needs some time and the car is not always in connection range the Arduino must return a value which represents "no current SoC value". I already tried "NaN" (which is not supported) and "-1" but this way EVCC seems to get confused (not loading, improper SoC readings etc).

andig, 2021-09-28:
I don‘t think the http plugin can handle this case.

doppelrittberger, 2021-09-28:
Ok. In this case I always return the last value until I receive a new one.

closed 2021-09-28 by andig