evcc-io/evcc #1668: Report Charged (Total)Energy to the MQTT-API

PaddyX, 2021-09-29:
It would be fine to report the Charged Total Energy value of a loadpoint also to the MQTT-API, to further evaluate it elsewhere.

```
./evcc -c evcc.yaml charger
[main  ] INFO 2021/09/29 19:18:56 evcc 0.64 (a90c69e)
[main  ] INFO 2021/09/29 19:18:56 using config file evcc.yaml
Power:          1390W
Energy:        91.1kWh
Current L1..L3: 5.8A 0A 0A
Charge status:  C
Enabled:        true
Charged:        0.6kWh
```
I could not find it at my configured MQTT-Broker, only the charged value of a current loading process:

```
evcc/updated 1632935785
evcc/site/pvPower 0
evcc/site/gridPower 1774
evcc/loadpoints/1/mode minpv
evcc/updated 1632935786
evcc/loadpoints/1/chargePower 1390
evcc/loadpoints/1/chargeCurrents/l1 5.9
evcc/loadpoints/1/chargeCurrents/l2 0
evcc/loadpoints/1/chargeCurrents/l3 0
evcc/loadpoints/1/chargeCurrents 5.9
evcc/loadpoints/1/activePhases 1
evcc/loadpoints/1/chargeCurrent 6
evcc/loadpoints/1/chargedEnergy 521.04
evcc/loadpoints/1/chargeDuration 1340
evcc/loadpoints/1/connected true
evcc/loadpoints/1/charging true
evcc/loadpoints/1/enabled true
evcc/loadpoints/1/vehicleSoc 37
evcc/loadpoints/1/chargeRemainingDuration 57333
evcc/loadpoints/1/range 146
evcc/loadpoints/1/timerSet false
evcc/loadpoints/1/timerActive true
evcc/loadpoints/1/timerProjectedEnd 1632935760
```
Thanks a lot

andig, 2021-09-29:
chargedEnergy?

PaddyX, 2021-09-30:
ChargedEnergy is only the Energy of the current loading process. In my example above i need the value 91.1 kWh of the charger also in the mqtt API which describe the sum value of Energy of all loading processes.

andig, 2021-09-30:
That‘s not available and currently not planned.

PaddyX, 2021-09-30:
Thanks andig. Since evcc manages the value read out from my charger internally, I thought that evcc would also make it available via mqtt. I will then have to read the value directly from the charger myself.

premultiply, 2021-09-30:
This is just the raw energy total meter value.
In general it is NOT the sum of the charged energy of this loadpoint. Only in special cases they are equal.

closed 2021-09-30 by andig