evcc-io/evcc #1679: Verbindung zur Tesal PowerWall 2

RettungsTim, 2021-10-01:
Eine Verbindung zur Tesal PowerWall 2 ist nicht möglich.

EVCC Startet damit nicht.
```
[main  ] INFO 2021/10/01 08:14:10 evcc 0.65 (270a29a)
[main  ] INFO 2021/10/01 08:14:10 using config file /etc/evcc.yaml
[main  ] INFO 2021/10/01 08:14:10 listening at 0.0.0.0:7070
[main  ] FATAL 2021/10/01 08:14:10 cannot create meter 'battery': cannot create meter 'tesla': login failed: unexpected status: 401

```
Meine Config der Meters:

```
# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters
meters:
- name: battery
  type: tesla
  uri: http://192.168.178.57/
  usage: battery
  password: ****
- name: grid
  type: tesla
  uri: http://192.168.178.57/
  usage: grid
  password: ****
- name: pv
  type: tesla
  uri: http://192.168.178.57/
  usage: pv
  password: ****
```

premultiply, 2021-10-01:
Da kann ich auch nur raten, da ich das System selbst nicht kenne.

HTTP Code 401 heisst aber normalerweise dass die Zugangsdaten nicht stimmen. Der hier im Modul hinterlegte Benutzername ist `customer`. Gilt das angegebene Passwort für diesen Benutzer?

Versuche doch mal im ersten Schritt `http://` durch `https://` zu ersetzen. Da gab es vielleicht ein Softwareupdate welches nur noch HTTPS zulässt. Das lässt sich zumindest aus https://github.com/vloschiavo/powerwall2 lesen.

RettungsTim, 2021-10-01:
Wie kann ich den den Benutzenamen übergeben? Da fehlt mir die Dokumentation

premultiply, 2021-10-01:
Derzeit gar nicht. Der Benutzername ist im Code fest mit `customer` hinterlegt:
https://github.com/evcc-io/evcc/blob/370a1fedb97ac291ef8c31316b1083f6f4114226/meter/tesla.go#L100-L105

Ob das für die API so sein muss oder ob Tesla da zwischenzeitlich etwas geändert so dass man auch andere Benutzer dafür benutzen kann kann ich dir leider nicht sagen.

Probier ggf. mal aus einen Benutzer `customer` in der TPW2 zu erstellen bzw. dessen Kennwort zu ändern und zu hinterlegen.

RettungsTim, 2021-10-01:
Das finde ich spannenden. Es war wirklich das https. Jetzt muss ich weiter schauen, da scheint das Automatische Routing von http zu https nicht zu funktionieren. Danke dir!!

closed 2021-10-01 by RettungsTim

premultiply, 2021-10-01:
Gerne! Ich passe dann mal fix die Doku an...