evcc-io/evcc #1750: Easee: vehicle remote charge start: unexpected status: 403

suhl-cloudian, 2021-10-16:
**Describe the bug**
I build the latest version from 15.10.2021 to test the Easee per Wallbox implementation by cloning the repo and building it.
When I start/stop the charge from the EVCC GUI I get the follwing error:

`[lp-1  ] ERROR 2021/10/16 11:52:19 vehicle remote charge start: unexpected status: 403`
or
`[lp-1  ] ERROR 2021/10/16 11:51:50 vehicle remote charge stop: unexpected status: 403`

It looks like everything is working like expected. Means the charging stops and starts.
Just this error is popping up in the interface. The logs are set to "trace", but no other info is logged:

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'EVCC GUI'
2. Click on 'STOP' or 'NOW'
4. See error

**Expected behavior**
Surpress the error message if everything is working fine.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.65 (270a29a)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc log output with <code>docker logs <container id></code>:</summary><br><pre>
[lp-1  ] INFO 2021/10/16 11:51:50 set charge mode: off
[lp-1  ] DEBUG 2021/10/16 11:51:50 charge power: 10688W
[lp-1  ] DEBUG 2021/10/16 11:51:50 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2021/10/16 11:51:50 charger status: C
[lp-1  ] DEBUG 2021/10/16 11:51:50 next soc poll remaining time: 14m46s
[lp-1  ] DEBUG 2021/10/16 11:51:50 soc estimated: 73.45% (vehicle: 71.00%)
[lp-1  ] DEBUG 2021/10/16 11:51:50 vehicle soc: 73%
[lp-1  ] DEBUG 2021/10/16 11:51:50 vehicle range: 379km
[lp-1  ] ERROR 2021/10/16 11:51:50 vehicle remote charge stop: unexpected status: 403
[lp-1  ] DEBUG 2021/10/16 11:51:51 charger disable
[lp-1  ] DEBUG 2021/10/16 11:51:56 charge power: 9419W
[lp-1  ] DEBUG 2021/10/16 11:51:56 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2021/10/16 11:51:56 charger status: B
[lp-1  ] INFO 2021/10/16 11:51:56 stop charging <-
[lp-1  ] DEBUG 2021/10/16 11:51:56 soc estimated: 73.45% (vehicle: 71.00%)
[lp-1  ] DEBUG 2021/10/16 11:51:56 vehicle soc: 73%
[lp-1  ] DEBUG 2021/10/16 11:51:56 vehicle range: 379km
[lp-2  ] DEBUG 2021/10/16 11:52:06 charge power: 0W
[lp-2  ] DEBUG 2021/10/16 11:52:06 charge currents: [15.6 15.8 15.6]A
[lp-2  ] DEBUG 2021/10/16 11:52:06 charger status: A
[lp-1  ] DEBUG 2021/10/16 11:52:16 charge power: 0W
[lp-1  ] DEBUG 2021/10/16 11:52:16 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2021/10/16 11:52:16 charger status: B
[lp-1  ] DEBUG 2021/10/16 11:52:16 next soc poll remaining time: 14m39s
[lp-1  ] INFO 2021/10/16 11:52:18 set charge mode: now
[lp-1  ] DEBUG 2021/10/16 11:52:18 charge power: 0W
[lp-1  ] DEBUG 2021/10/16 11:52:18 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2021/10/16 11:52:18 charger status: B
[lp-1  ] DEBUG 2021/10/16 11:52:18 next soc poll remaining time: 14m37s
[lp-1  ] DEBUG 2021/10/16 11:52:18 charger enable
[lp-1  ] ERROR 2021/10/16 11:52:19 vehicle remote charge start: unexpected status: 403
[lp-2  ] DEBUG 2021/10/16 11:52:26 charge power: 0W
[lp-2  ] DEBUG 2021/10/16 11:52:26 charge currents: [0.009 0.009 0.01]A
[lp-2  ] DEBUG 2021/10/16 11:52:26 charger status: A
[lp-1  ] DEBUG 2021/10/16 11:52:36 charge power: 1305W
[lp-1  ] DEBUG 2021/10/16 11:52:36 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2021/10/16 11:52:36 charger status: C
[lp-1  ] INFO 2021/10/16 11:52:36 start charging ->
[lp-1  ] DEBUG 2021/10/16 11:52:36 soc estimated: 73.45% (vehicle: 71.00%)
[lp-1  ] DEBUG 2021/10/16 11:52:36 vehicle soc: 73%
[lp-1  ] DEBUG 2021/10/16 11:52:36 vehicle range: 379km
[lp-2  ] DEBUG 2021/10/16 11:52:46 charge power: 0W
[lp-2  ] DEBUG 2021/10/16 11:52:46 charge currents: [6.81 6.81 0.01]A
[lp-2  ] DEBUG 2021/10/16 11:52:46 charger status: A
[lp-1  ] DEBUG 2021/10/16 11:52:56 charge power: 10606W
[lp-1  ] DEBUG 2021/10/16 11:52:56 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2021/10/16 11:52:56 charger status: C
[lp-1  ] DEBUG 2021/10/16 11:52:56 next soc poll remaining time: 14m40s
[lp-1  ] DEBUG 2021/10/16 11:52:56 soc estimated: 73.45% (vehicle: 71.00%)

</pre></details>

andig, 2021-10-16:
Daten fehlen

closed 2021-10-16 by andig

suhl-cloudian, 2021-10-16:
Was meinst Du damit? Daten vom Bugreport?

suhl-cloudian, 2021-10-16:
Fehler scheint von Tronity zu kommen:

```
[tronity] TRACE 2021/10/16 12:24:47 
POST /v1/vehicles/5fcf59b7a2a3d90012df0906/charge_start HTTP/1.1
Host: api-eu.tronity.io
User-Agent: Go-http-client/1.1
Content-Length: 0
Authorization: Bearer eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJzdWIiOiI2MTRjN2E3MS02N2VlLTRiYTAtODUyNC03YzNlNGE1MzI2MmYiLCJhcHBJZCI6ImYyMzAzZjU1LTE0YjAtNGM1NC04MzdjLWMzZWFiZjRkZjk0ZSIsInZlaGljbGVJZCI6bnVsbCwic2NvcGVzIjpbInJlYWRfYmF0dGVyeSIsInJlYWRfY2FyZHMiLCJyZWFkX2NoYXJnZXBvaW50cyIsInJlYWRfaWRsZXMiLCJyZWFkX2xvY2F0aW9uIiwicmVhZF9vZG9tZXRlciIsInJlYWRfb3BlcmF0b3JzIiwicmVhZF9zbGVlcHMiLCJyZWFkX3RyaXBzIiwicmVhZF92ZWhpY2xlX2luZm8iLCJyZWFkX3ZpbiIsInJlYWRfY2hhcmdlcyIsInJlYWRfY2hhcmdlIiwicmVhZF9yZWNvcmRzIl0sInJvbGUiOiJ1c2VyIiwiaWF0IjoxNjM0Mzg3MDg3LCJleHAiOjE2MzQ0NzM0ODcsImF1ZCI6Imh0dHBzOi8vcGxhdGZvcm0udHJvbml0eS5pbyIsImlzcyI6InBsYXRmb3JtLnRyb25pdHkuaW8ifQ.VYVR3YhxKkFvyf0DLyW2lJflQMzoALZExLqqgITmSsw
Content-Type: 
Accept-Encoding: gzip

HTTP/2.0 403 Forbidden
Content-Length: 40
Access-Control-Allow-Credentials: true
Alt-Svc: h3=":443"; ma=86400, h3-29=":443"; ma=86400, h3-28=":443"; ma=86400, h3-27=":443"; ma=86400
Cf-Cache-Status: DYNAMIC
Cf-Ray: 69f1441e6b2f6946-FRA
Content-Security-Policy: default-src 'self';connect-src 'self' https://api.tronity.io;script-src 'self' https://connect-tronity.web.app;img-src * data:;style-src 'self' 'unsafe-inline' https://connect-tronity.web.app;font-src 'self' https://connect-tronity.web.app
Content-Type: application/json; charset=utf-8
Date: Sat, 16 Oct 2021 12:24:47 GMT
Etag: W/"28-U8jz4ag/pp9XnJ7D8ZTSv5nAl68"
Expect-Ct: max-age=0
Nel: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
Referrer-Policy: no-referrer
Report-To: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v3?s=q6dDLBXsWQi8cq0ldmtWlCvg5OkTxHB%2B13MCOwIAFN%2FVdROv89c4l%2BUlUC5XGW5F2XWj%2BSzqEoLxO3zmECL4DJcf7SKXpGyZsoz516XCfvt9HfwwB7roBwTKyHRknyN9mYGN"}],"group":"cf-nel","max_age":604800}
Server: cloudflare
Strict-Transport-Security: max-age=15552000; includeSubDomains
Vary: Origin, Accept-Encoding
X-Content-Type-Options: nosniff
X-Dns-Prefetch-Control: off
X-Download-Options: noopen
X-Frame-Options: SAMEORIGIN
X-Permitted-Cross-Domain-Policies: none
X-Xss-Protection: 0

{"statusCode":403,"message":"Forbidden"}
[lp-1  ] ERROR 2021/10/16 12:24:47 vehicle remote charge start: unexpected status: 403
```

andig, 2021-10-16:
Es fehlten alle relevanten Daten- Config wie auch Logfile. Dass das irgendwas mit Tronity zu tun haben könnte war völlig unklar. Für solche Suchspiele fehlt mir die Zeit.