evcc-io/evcc #1821: UI should communicate the current state

DerAndereAndi, 2021-11-02:
Current situation:

- All data is only updated depending on the interval setting in the configuration file
- Changing the charge mode also only happens depending on the interval settings and is not instant
- There are timers (PV enable and disable, 1p3p, ...) which trigger several actions, e.g. pause charging, switching phases, ...

The problem for the user:

- It often unclear when or if an action will happen or why the current state is as it is, e.g.
  - changing a mode doesn't have an instant effect
  - charging in PV mode is on Pause, but there is enough PV power available

Solution:

We need to find a way in the UI to communicate different states and when something is expected to happen/change.
First step is to list all these situations and then check how this can be presented/communicated.

See also for an example: https://github.com/evcc-io/evcc/discussions/1819

andig, 2021-11-02:
We have `pvTimer` and `phaseTimer`, both are evaluated against `enable/disable.delay`. For first experiments we could just show the timers in the UI and the seconds elapsed sind timer value.

@naltatis könnten wir sowas einbauen und zu Testzwecken z.B. für Query Param oder auch einen schöneren Mechanismus anzeigen? Schritt zwei wäre dann zu überlegen wann/welche Aussagen sich daraus generieren lassen.

premultiply, 2021-11-02:
Wie wäre es mit einer Art Debugseite im Web-UI wo diese internen Zustände und weitere regelmäßig für den Support und die Fehlersuche benötige Infos in einer halbwegs übersichtlichen und gebündelten Textform verfügbar sind?

DerAndereAndi, 2021-11-02:
@premultiply das ist wieder etwas anderes als das hier

naltatis, 2021-11-02:
@andig wenn du die beiden Timer mit in die UI gibts überlege ich mir ne kompakte Darstellung. Pro Timer sollte die Richtung (enable/disable), der aktuelle Wert und der Gesamtwert des Timers (4:22min von 5min) mitkommen. Dann können wir mit ner prozentualen Darstellung (kleiner Balken, Donut oder so) arbeiten.

closed 2021-12-13 by naltatis