evcc-io/evcc #1822: Change charge mode to now with 1p3p should always directly switch to 3p

DerAndereAndi, 2021-11-02:
Current situation:

- Charge mode changes do not happen instantly, but only depending on the current interval configuration
- When the previous charging mode was PV and the charger is capable of 1p3p, it might be in the 1p state
- Switching between 1p and 3p is also depending on timers, as this should not happen too often (it could cause damages otherwise)

The problem:
- When switching to now charging might start with 1p, and it is unclear to the user if or when it might change as the expectation is charge mode node will always charge with 3p

Solution options:
- When the user switches to now and the charger is in 1p mode, ignore the timer and switch to 3p
- Highlight in the UI that the switch will happen and how long it will take, also why. See https://github.com/evcc-io/evcc/issues/1821

See also: https://github.com/evcc-io/evcc/discussions/1819

andig, 2021-11-02:
`Now` schaltet bereits auf 3p, berücksichtigt aber die Timer. Frage wäre ob wir- technisch gesprochen- bei jedem Moduswechsel (und nicht nur bei `now`) alle Timer expired wollen i.S. von "Anwender tut was und erwartet sofort Ergebnis"?

premultiply, 2021-11-02:
Ja, bitte.

andig, 2021-12-31:
Sollte zumindest in Teilen durch https://github.com/evcc-io/evcc/pull/2168 adressiert sein

closed 2022-01-26 by github-actions[bot]