evcc-io/evcc #1825: Out of Sync easee Walbox

tmy0, 2021-11-03:
Wie hier https://github.com/evcc-io/evcc/discussions/1777#discussioncomment-1581142 besprochen der issue zum Thema.
Beim umschalten der Lade-Modi kommt es teilweise zum Out of Sync Fehlern.
Beispiel: charger out of sync: expected enabled, got disabled.

Jogging wurde aktiviert:
log: trace
levels:
  core: error
  lp-1: error
  lp-2: error

andig, 2021-11-03:
@tmy0 schickst Du noch das Log?

tmy0, 2021-11-03:
@andig Ich konnte den Fehler nicht reproduzieren heute. Ich kann das dir das Log trotzdem gern zukommen lassen. Ich müsste nur kurz wissen, wie das Log heißt und wo es liegt. Gibt es hier einen Uploadbereich? Ich kann das auch selbst hochladen und dir einen link für den Download schicken.

andig, 2021-11-03:
Ich mach mal zu. Kein Fehler, kein Log ;)