evcc-io/evcc #1827: temporary loss of MQTT Connection breaks sma meters communication

F-L-0, 2021-11-03:
**Describe the bug**
Every time mqtt server is temporary not available (eg restarted), evcc fails to update any sma component from meters section. mqtt connection is re-established once mqtt server is available (updates can be seenon broker side), but polling of sma meters (inversters) keeps failing and only recovers after evcc restart.
Issue is reproducable very easy, can provide additional debug/trace logs and configs when relevant/needed.

**To Reproduce**
Steps to reproduce the behavior:
1. Start mqtt server (in my case embedded in fhem - should be reproducable with mosquitto too)
2. Start evcc with at leasst one sma inverter configured in yaml
3. Restart MQTT server
4. sma inverter updates permanently fail until evcc is restarted

**Expected behavior**
Restart of mqtt server / failing of mqtt connection does not impact communication with sma inverters

Additional information: Tested also to stop/restart influxdb to check if there is the same behaviour. Here errors for writing to influx where logged as expected, but sma inverters continued to update flawlessly, the issue seems to be related to mqtt backend specifically.

**EVCC details:**
Show output of <code>evcc -v</code> 
```
evcc version 0.66 (9181925)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
- name: grid
  type: sma
  uri: 192.168.1.13
- name: STP10000
  type: sma
  uri: 192.168.1.11
- name: SI60H
  type: sma
  uri: 192.168.1.12

</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>
restart MQTT Broker at 18:30:59:

Nov  3 18:30:41 rpi4-1 evcc[11459]: [lp-1  ] DEBUG 2021/11/03 18:30:41 next soc poll remaining time: 56m10s
Nov  3 18:30:41 rpi4-1 evcc[11459]: [lp-1  ] DEBUG 2021/11/03 18:30:41 charge power: 0W
Nov  3 18:30:51 rpi4-1 evcc[11459]: [lp-1  ] DEBUG 2021/11/03 18:30:51 charge power: 0W
Nov  3 18:30:51 rpi4-1 evcc[11459]: [lp-1  ] DEBUG 2021/11/03 18:30:51 charger status: A
Nov  3 18:30:51 rpi4-1 evcc[11459]: [lp-1  ] DEBUG 2021/11/03 18:30:51 next soc poll remaining time: 56m0s
Nov  3 18:30:51 rpi4-1 evcc[11459]: [lp-1  ] DEBUG 2021/11/03 18:30:51 charge power: 0W
Nov  3 18:30:59 rpi4-1 evcc[11459]: [mqtt  ] ERROR 2021/11/03 18:30:59 localhost:1883 connection lost: EOF
Nov  3 18:31:01 rpi4-1 evcc[11459]: [lp-1  ] DEBUG 2021/11/03 18:31:01 charge power: 0W
Nov  3 18:31:01 rpi4-1 evcc[11459]: [lp-1  ] DEBUG 2021/11/03 18:31:01 charger status: A
Nov  3 18:31:01 rpi4-1 evcc[11459]: [lp-1  ] DEBUG 2021/11/03 18:31:01 next soc poll remaining time: 55m50s
Nov  3 18:31:01 rpi4-1 evcc[11459]: [lp-1  ] DEBUG 2021/11/03 18:31:01 charge power: 0W
Nov  3 18:31:06 rpi4-1 evcc[11459]: [sma   ] ERROR 2021/11/03 18:31:06 login failed: no packet received in timeout
Nov  3 18:31:06 rpi4-1 evcc[11459]: [sma   ] ERROR 2021/11/03 18:31:06 login failed: no packet received in timeout
Nov  3 18:31:11 rpi4-1 evcc[11459]: [sma   ] ERROR 2021/11/03 18:31:11 login failed: no packet received in timeout
Nov  3 18:31:11 rpi4-1 evcc[11459]: [sma   ] ERROR 2021/11/03 18:31:11 login failed: no packet received in timeout
Nov  3 18:31:11 rpi4-1 evcc[11459]: [site  ] ERROR 2021/11/03 18:31:11 updating pv meter 0: update timeout: 11s
Nov  3 18:31:12 rpi4-1 evcc[11459]: [site  ] ERROR 2021/11/03 18:31:12 updating battery meter 0: update timeout: 12s
Nov  3 18:31:12 rpi4-1 evcc[11459]: [site  ] ERROR 2021/11/03 18:31:12 updating battery soc 0: update timeout: 12s
Nov  3 18:31:12 rpi4-1 evcc[11459]: [lp-1  ] DEBUG 2021/11/03 18:31:12 charge power: 0W
Nov  3 18:31:12 rpi4-1 evcc[11459]: [lp-1  ] DEBUG 2021/11/03 18:31:12 charger status: A
Nov  3 18:31:12 rpi4-1 evcc[11459]: [lp-1  ] DEBUG 2021/11/03 18:31:12 next soc poll remaining time: 55m39s
Nov  3 18:31:12 rpi4-1 evcc[11459]: [lp-1  ] DEBUG 2021/11/03 18:31:12 charge power: 0W
Nov  3 18:31:16 rpi4-1 evcc[11459]: [sma   ] ERROR 2021/11/03 18:31:16 login failed: no packet received in timeout
Nov  3 18:31:16 rpi4-1 evcc[11459]: [sma   ] ERROR 2021/11/03 18:31:16 login failed: no packet received in timeout
Nov  3 18:31:21 rpi4-1 evcc[11459]: [sma   ] ERROR 2021/11/03 18:31:21 login failed: no packet received in timeout
Nov  3 18:31:21 rpi4-1 evcc[11459]: [sma   ] ERROR 2021/11/03 18:31:21 login failed: no packet received in timeout
Nov  3 18:31:21 rpi4-1 evcc[11459]: [site  ] ERROR 2021/11/03 18:31:21 updating pv meter 0: update timeout: 21s
Nov  3 18:31:22 rpi4-1 evcc[11459]: [site  ] ERROR 2021/11/03 18:31:22 updating battery meter 0: update timeout: 22s
Nov  3 18:31:22 rpi4-1 evcc[11459]: [site  ] ERROR 2021/11/03 18:31:22 updating battery soc 0: update timeout: 22s
Nov  3 18:31:22 rpi4-1 evcc[11459]: [lp-1  ] DEBUG 2021/11/03 18:31:22 charge power: 0W

</pre></details>

**If using Docker:**
Show output of <code>docker run andig/evcc -v</code>:
```
<- paste here

```

<details><summary>Show evcc log output with <code>docker logs <container id></code>:</summary><br><pre>
<- paste here

</pre></details>

andig, 2021-11-03:
Was hat das mit SMA zu tun? Ich kanns wir schwer vorstellen aufgrund https://github.com/evcc-io/evcc/blob/1ba175bd5c49e24e97b51b68e371983d78f78b82/provider/mqtt/client.go#L66, deshalb bitte Logfile.

andig, 2021-11-04:
Wenn ich es richtig verstehe, dann fällt SMA aus wenn der (unabhängigte) MQTT Server restartet wird?

@bboehmke ich kann mir keinen Reim drauf machen und es auch nicht nachstellen. Könntest Du bei Dir schauen, ob Du das Verhalten reproduzieren kannst? Vielen Dank!

F-L-0, 2021-11-04:
> Was hat das mit SMA zu tun? Ich kanns wir schwer vorstellen aufgrund
> 
> https://github.com/evcc-io/evcc/blob/1ba175bd5c49e24e97b51b68e371983d78f78b82/provider/mqtt/client.go#L66
> 
> , deshalb bitte Logfile.

Hi Andi,
das Logfile (evcc --log debug) ist oben im Issue schon dabei, brauchst du was zusätzliches, anderer Loglevel?

Im Config sind 3 SMA Geräte definiert, das EM läuft weiter (sieht man an der durchgehenden Ausgabe der Werte der 3 Phasen), die beiden Wechselrichter abzufragen funktioniert ab dem Restart des MQTT Servers nicht mehr. Der Reconnect zum mqtt Server funktioniert einwandfrei, die Kommunikation zu den Wechselrichtern bleibt gestört.
Ich weiß, klingt komisch, ist aber reproduzierbar so ;-)

Wenn ich noch was tracen kann (Loglevel, tcpdump, etc ...) - sehr gerne.

bboehmke, 2021-11-04:
Ich hab es jetzt mal mit einer erweiterten Demokonfiguration getestet und bei mir funktioniert zumindest das EnergyMeter weiterhin nach Verbindungsverlust mit dem MQTT broker. 

Leider war/ist das Wetter nicht besonders weshalb weder Strom von der PV Anlage kommt noch wurde die Batterie heute geladen weshalb ich nicht prüfen konnte ob die Wechselrichter weiterhin funktionieren.

<details>
  <summary>Getestet habe ich mit dieser Config</summary>

```yaml
log: info

javascript:
  - vm: shared
    script: |
      state = {
        residualpower: 200,
        pvpower: -3000,
        batterypower: 200,
        gridpower: -2000,
        chargepower: 0,
        maxcurrent: 0,
        battery: 63, // car
      };
      function get() {
        console.log("state:", JSON.stringify(state));
      }
      function set() {
        console.log(param+":", val);
        console.log("state:", JSON.stringify(state));
      }

meters:
  - name: grid
    type: sma
    interface: enp0s20f0u1
    uri: 192.168.4.50
  - name: pv
    type: sma
    interface: enp0s20f0u1
    uri: 192.168.4.51
  - name: battery
    type: sma
    interface: enp0s20f0u1
    uri: 192.168.4.52

  - name: charge
    type: custom
    power:
      type: js
      vm: shared
      script: state.chargepower;
    soc:
      type: js
      vm: shared
      script: "30"

chargers:
  - name: demo
    type: custom
    enable:
      type: js
      vm: shared
      script: |
        set();
        state.enabled = val;
        if (state.enabled) state.chargepower = state.maxcurrent * 230; else state.chargepower = 0;
    enabled:
      type: js
      vm: shared
      script: |
        state.enabled;
    status:
      type: js
      vm: shared
      script: |
        if (state.enabled) "C"; else "B";
    maxcurrent:
      type: js
      vm: shared
      script: |
        set();
        state.maxcurrent = val;
        if (state.enabled) state.chargepower = state.maxcurrent * 230;

vehicles:
  - name: demo
    title: e-Golf
    type: custom
    charge:
      type: js
      vm: shared
      script: |
        if (state.chargepower > 0) state.battery++; else state.battery--;
        if (state.battery < 15) state.battery = 15;
        if (state.battery > 100) state.battery = 100;
        state.battery;
    cache: 1s

site:
  title: Demo
  meters:
    grid: grid
    pv: pv
    battery: battery

loadpoints:
  - title: Carport
    charger: demo
    meters:
      charge: charge
    vehicle: demo

mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable

```
</details>

und einem lokalen mqtt broker via docker:
```
docker run --rm -it -p 1883:1883 eclipse-mosquitto mosquitto -c /mosquitto-no-auth.conf
```

andig, 2021-11-04:
Anscheinend betrifft das Problem nur die WR. Ich lausche gespannt was ihr raus findet- das ist ja mal ein wirklich fancy issue ;)

andig, 2021-11-04:
@F-L-0 kann man die WR auch über Serial statt URI konfigurieren? Ändert das was?

F-L-0, 2021-11-04:
Ich habe jetzt selbst auch nochmal auf einem zweiten Raspberry getestet und konnte mit identischer Config den Fehler nicht nachstellen. Was auch immer das Problem ist, es muss in meinem Setup begründet sein. Ich werde das noch weiter untersuchen, auf dem RPI der den Fehler zeigt läuft noch einiges mehr parallel das mit den Wechselrichtern spricht (SBFSpot, FHEM, ...). Wie auch immer, ich würde eure Zeit da nicht weiter beanspruchen und, falls Interesse besteht, mich damit melden was ich gefunden hab. Falls ich es nicht finde bekommt evcc halt einen eigenen RPI ;-)

bboehmke, 2021-11-04:
Ah da haben wirs: Die Kommunikation mit den SMA Wechselrichtern via Speedwire funktioniert nur einmal von einem Gerät aus. Also entweder evcc oder SPFSpot. 

(Hintergrund ist das beide Anwendungen auf dem gleichen UDP port lauschen -> das geht ohne weiteres nicht zweimal auf einem System)

F-L-0, 2021-11-04:
Der Vollständigkeit halber noch - SBFSpot, FHEM mit dem Modul 76_SMAInverter.pm und evcc existieren schon seit Wochen erstmal friedlich nebeneinander auf einem RPI. Wenn man allerdings fhem restartet (und erst dann) wird die Speedwire-Session beim shutdown beendet, worauf auch evcc nix mehr bekommt (?). Bin nicht ganz tief im Protokoll, aber ich meine da waren Multicast-Subscriptions (netstat -g) im Spiel, beim EM auf jeden Fall, bei den Invertern wars evtl auch nur UDP. MQTT hat definitiv nix damit zu tun, da bin ich falsch abgebogen, fhem war nur in meinem setup gleichzeitig der MQTT Broker. Der Fehler tritt auch auf wenn man mqtt in evcc.yaml deaktiviert.

Ab diesem Zeitpunkt (fhem-restart) antwortet der Inverter im Log von evcc nicht mehr auf
[sma   ] TRACE 2021/11/04 19:42:57 login for 192.168.1.11:9522
[sma   ] TRACE 2021/11/04 19:42:57 send 192.168.1.11: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]

Vorher wenn es geht steht da
[sma   ] TRACE 2021/11/04 19:42:52 login for 192.168.1.11:9522
[sma   ] TRACE 2021/11/04 19:42:52 send 192.168.1.11: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2021/11/04 19:42:52 recv 192.168.1.11: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]

Der Filedescriptor auf UDP:9522 von evcc bleibt vorhanden:
evcc    15867 evcc    3u     IPv4  209850044      0t0       UDP *:9522

Im tcpdump sieht man das der Wechselrichter auf den Login von evcc antwortet, das aber bei evcc nicht mehr ankommt:
```
20:33:13.389992 IP 192.168.1.240.9522 > 192.168.1.11.9522: UDP, length 78
        0x0000:  4500 006a 8cd5 4000 4011 2962 c0a8 01f0  E..j..@.@.)b....
        0x0010:  c0a8 010b 2532 2532 0056 84b3 534d 4100  ....%2%2.V..SMA.
        0x0020:  0004 02a0 0000 0001 003a 0010 6065 0ea0  .........:..`e..
        0x0030:  b500 0b4a 2e12 0001 7800 f001 a8c0 0001  ...J....x.......
        0x0040:  0000 0000 f180 0c04 fdff 0700 0000 8403  ................
        0x0050:  0000 7935 8461 0000 0000 b8b8 b8b8 8888  ..y5.a..........
        0x0060:  8888 8888 8888 0000 0000                 ..........
20:33:13.423860 IP 192.168.1.11.9522 > 192.168.1.240.9522: UDP, length 78
        0x0000:  4500 006a 0000 0000 4011 f637 c0a8 010b  E..j....@..7....
        0x0010:  c0a8 01f0 2532 2532 0056 124b 534d 4100  ....%2%2.V.KSMA.
        0x0020:  0004 02a0 0000 0001 003a 0010 6065 0ed0  .........:..`e..
        0x0030:  7800 f001 a8c0 0001 b500 0b4a 2e12 0001  x..........J....
        0x0040:  0000 0000 f180 0d04 fdff 0700 0000 8403  ................
        0x0050:  0000 7935 8461 0000 0000 b8b8 b8b8 8888  ..y5.a..........
        0x0060:  8888 8888 8888 0000 0000                 ..........
```

Was evcc bei der Initialisierung anders macht das der Speedwire-Login nach evcc-restart wieder geht ist mir schleierhaft, hab mal hier gelesen und versucht zu verstehen, aber meine go-Kenntnisse tendieren gegen Null:
https://gitlab.com/bboehmke/sunny/-/blob/master/connection.go

Anyway - Ich habe auf die schnelle keine Möglichkeit gefunden da was anzupassen, daher zieht evcc jetzt halt einfach auf einen anderen RPI um und gut ist  :-)

bboehmke, 2021-11-04:
Soweit ich weiß ist ein listen auf einem Port erstmal exklusiv und kann nur einmal pro IP gleichzeitig verwendet werden. Es gibt zwar mit `SO_REUSEPORT` die Option einen port mehrfach zu nutzen allerdings hab ich das bisher nie verwendet.

Was mich etwas wundert das es überhaupt möglich ist beide Anwendungen gleichzeitig zustarten. Ich hätte hier mit einer Meldung aller "port already in use" gerechnet.

Wenn ich dazu komme schaue ich da am Wochenende nochmal rein aber ich rechne nicht mit einer einfachen Lösung.

andig, 2021-11-05:
> Im tcpdump sieht man das der Wechselrichter auf den Login von evcc antwortet, das aber bei evcc nicht mehr ankommt:

Top Analyse 👍🏻

> Wenn ich dazu komme schaue ich da am Wochenende nochmal rein aber ich rechne nicht mit einer einfachen Lösung.

Das wär Klasse.

> Was mich etwas wundert das es überhaupt möglich ist beide Anwendungen gleichzeitig zustarten. Ich hätte hier mit einer Meldung aller "port already in use" gerechnet.

Ist mir auch unklar, wobei ich wenig Erfahrung mit UDP habe. Ggf. könnte man das mit einem kleinen Experiment verifizieren (2 lokale Listener).

premultiply, 2021-11-05:
Die Kommunikation wird ja bei UDP auch mit dem entsprechenden ZIELport aufgebaut. Der lokale Quellport wird im Normalfall dynamisch (beliebiger freier Port) vergeben. Die Gegenstelle antwortet dann als ZIEL auf den ursprünglichen Quellport.
Es gibt hier daher keine notwendigkeit für Unicast ausgerechnet mit dem lokalen Quellport 9522 zu arbeiten oder schreibt das Protokoll das zwingend so vor?
9522 ist doch eigentlich der Multicast-Listener-Port und den sollten sich aus Prinzip schon mehrere Anwendungen teilen können müssen, oder?

F-L-0, 2021-11-05:
Soweit ich weiß will Speedwire zwingend udp:9522
Ich teile die Meinung das es eigentlich nicht funktionieren sollte 2 Applikationen auf dem selben Port lauschen zu lassen, die zweite sollte eine "Adress already in use" ressource exception bekommen. Grad nachgeschaut, warum auch immer, es ist durchaus bei UDP möglich das mehrere Programme auf dem selben Port lauschen, hier perl (fhem)und evcc unter Raspbian:

```
root@rpi4-1:~# lsof -iudp:9522
COMMAND   PID USER   FD   TYPE    DEVICE SIZE/OFF NODE NAME
perl    15986 fhem   13u  IPv4 209859519      0t0  UDP *:9522
evcc    16961 evcc    7u  IPv4 209921525      0t0  UDP *:9522
```

Selbst wenn es geht ist es aber glaube ich nicht gut und kann wie bei meinem Setup gesehen zu komischen Fehlerzuständen führen wenn das Programm das zuerst kommt restartet wird (bei mir wird wegen der mqtt Abhängigkeit derzeit evcc nach fhem gestartet).

Soweit ich weiß für die Wechselrichter ist die Kommunikation Unicast UDP Port 9522, während für den EnergyManager eine Multicast-Gruppe auf 239.12.255.254 Port 9522 existiert und jede Sekunde ein Update vom EM geschickt wird.

```
root@rpi4-1:~# netstat -g
IPv6/IPv4-Gruppenmitgliedschaften
Schnittstelle   RefZäh Grupp
--------------- ------ ---------------------
eth0            2      239.12.255.254
[...]
```

Ich weiß nicht ob es eine gute Idee ist SO_REUSEPORT überhaupt anzugehen oder stattdessen besser in der Modulbeschreibung zu dokumentieren das keine anderen Programme auf dem selben Rechner mit den SMA Komponenten kommunizieren dürfen? Schließlich bekommt die evcc sunny lib auch jeglichen Traffic des anderen Programms oder bei passendem Timing Antwortpakete die nicht zur eigenen Anfrage passen und muss programmtechnisch damit umgehen, oder? Ich habe bisher zwar auf beiden Seiten keine komischen Effekte beobachtet, aber möglich wärs.

premultiply, 2021-11-05:
> Soweit ich weiß will Speedwire zwingend udp:9522

Aber nur eingehend auf Seiten des WR, oder? Der Quellport des Initiators sollte beliebig sein.

andig, 2021-11-05:
Ja. Die Response kommt aber immer auf 9522, nutzt dir also nix

premultiply, 2021-11-05:
Aaaarg. Wer hat denn das erfunden?

andig, 2021-11-06:
Naja, wie willst Du Multicast sonst machen?

premultiply, 2021-11-06:
Moment! Wir sprechen hier von Unicast.
Nur die Messwerte des SHM und EM werden per Multicast gesendet. Die gezielte Kommunikation zwischen SHM und WR ist Unicast.

bboehmke, 2021-11-06:
Auch dort ist der Empfangsport beim Client fix 9522.
Bei simplen Protokollen besonders bei embeded Geräten taucht sowas häufiger auf. (nicht schön aber simpel)

bboehmke, 2021-11-06:
@F-L-0 ich hab nochmal über das Thema nachgedacht und ich denke es wäre wirklich das Sinnvollste dieses Verhalten einfach zu dokumentieren.

Mir scheint das es (zumindest in go) möglich ist das selbe multicast listen mehrfach zu machen (beim normalen UDP listen geht es nicht). Vielleicht könnte man hier versuchen noch ein zusätzlichen normalen listen auf den port zu machen aber ich weiß nicht ob das wirklich sinnvoll ist.

Ich versuche bei Gelegenheit mir mal SPFSpot genauer anzuschauen vielleicht wird da ja was anders gemacht was in der go lib nützlich wäre. In der nächsten Zeit werde ich allerdings nicht dazu kommen.

andig, 2021-11-06:
Dann machen wir hier erstmal zu. Works as expected.

closed 2021-11-06 by andig