evcc-io/evcc #1828: Consumption on E3DC wrong

matthiasertl, 2021-11-04:
**Describe the bug**
The Power consumption is Same like ob Power [image]

**To Reproduce**
E3DC Template from evcc /config 

**EVCC details:**

see Screenshot

andig, 2021-11-04:
Evcc version?

matthiasertl, 2021-11-04:
0.68

matthiasertl, 2021-11-04:
WIthout PV everything seems correct - i made a battery forced charge ..
[image]
.

andig, 2021-11-04:
Bitte konkreten Fehler- richtige/falsche Werte- in https://github.com/evcc-io/evcc/issues/1787 ergänzen.

closed 2021-11-04 by andig