evcc-io/evcc #1843: Fix build on OpenBSD - update serial dependencies

geowiwi, 2021-11-08:
Currently build on OpenBSD fails with:

`../go/pkg/mod/github.com/goburrow/serial@v0.1.0/serial_bsd.go:30:10: undefined:                                                                          syscall.B460800
../go/pkg/mod/github.com/goburrow/serial@v0.1.0/serial_bsd.go:78:44: fds.X__fds_                                                                         bits undefined (type *syscall.FdSet has no field or method X__fds_bits)
../go/pkg/mod/github.com/goburrow/serial@v0.1.0/serial_bsd.go:79:45: fds.X__fds_                                                                         bits undefined (type *syscall.FdSet has no field or method X__fds_bits)
../go/pkg/mod/github.com/goburrow/serial@v0.1.0/serial_bsd.go:86:5: fds.X__fds_b                                                                         its undefined (type *syscall.FdSet has no field or method X__fds_bits)
../go/pkg/mod/github.com/goburrow/serial@v0.1.0/serial_bsd.go:92:12: fds.X__fds_                                                                         bits undefined (type *syscall.FdSet has no field or method X__fds_bits)
../go/pkg/mod/github.com/goburrow/serial@v0.1.0/termios_bsd.go:10:17: cannot use                                                                          speed (type uint32) as type int32 in assignment
../go/pkg/mod/github.com/goburrow/serial@v0.1.0/termios_bsd.go:14:17: cannot use                                                                          speed (type uint32) as type int32 in assignment`

and:

`../go/pkg/mod/github.com/grid-x/serial@v0.0.0-20191104121038-e24bc9bf6f08/serial                                                                         _bsd.go:30:10: undefined: syscall.B460800
../go/pkg/mod/github.com/grid-x/serial@v0.0.0-20191104121038-e24bc9bf6f08/serial                                                                         _bsd.go:78:44: fds.X__fds_bits undefined (type *syscall.FdSet has no field or me                                                                         thod X__fds_bits)
../go/pkg/mod/github.com/grid-x/serial@v0.0.0-20191104121038-e24bc9bf6f08/serial                                                                         _bsd.go:79:45: fds.X__fds_bits undefined (type *syscall.FdSet has no field or me                                                                         thod X__fds_bits)
../go/pkg/mod/github.com/grid-x/serial@v0.0.0-20191104121038-e24bc9bf6f08/serial                                                                         _bsd.go:86:5: fds.X__fds_bits undefined (type *syscall.FdSet has no field or met                                                                         hod X__fds_bits)
../go/pkg/mod/github.com/grid-x/serial@v0.0.0-20191104121038-e24bc9bf6f08/serial                                                                         _bsd.go:92:12: fds.X__fds_bits undefined (type *syscall.FdSet has no field or me                                                                         thod X__fds_bits)
../go/pkg/mod/github.com/grid-x/serial@v0.0.0-20191104121038-e24bc9bf6f08/termio                                                                         s_bsd.go:10:17: cannot use speed (type uint32) as type int32 in assignment
../go/pkg/mod/github.com/grid-x/serial@v0.0.0-20191104121038-e24bc9bf6f08/termio                                                                         s_bsd.go:14:17: cannot use speed (type uint32) as type int32 in assignment`
`

Code is fixed in:
https://github.com/goburrow/serial/commit/bfb69110f8dd407add15aa6de3a7b6b31c9aa4a7
https://github.com/grid-x/serial/commit/583c7356b3aa92ee5ebdba04ac1fc3a98a313cdd

Please consider updating the serial go modules.
`go get github.com/goburrow/serial/commit/bfb69110f8dd407add15aa6de3a7b6b31c9aa4a7`
`go get github.com/grid-x/serial/commit/583c7356b3aa92ee5ebdba04ac1fc3a98a313cdd`

Thanks!

andig, 2021-11-08:
Good one, I was waiting for 7c98d6e91a65c00bbf7277e5e6f770305206437a anyway.

andig, 2021-11-08:
@geowiwi out of interest: what are you doing with evcc on OpenBSD?

premultiply, 2021-11-08:
Mmmh. If it does support BSD now it should also run on TrueNAS (FreeBSD) jails… 👌

geowiwi, 2021-11-08:
evcc is replacing my hand-rolled python script commanding my goe-charger. In general, having some kind of standardized, open-source charging point solution in the OpenBSD ports-tree might be worthwhile. OTOH the overhead of a go implementation is quite big for a small-scale arm-based router (in my case).

andig, 2021-11-08:
I don‘t know how strong these boxes are, but go shouldn‘t be much less capable than cpp. The heavy lifting is the build and Go has excellent cross-build capabilities.

geowiwi, 2021-11-08:
Sure, but these are topics not specific to evcc.

Please note that andig/gosunspec still pulls in goburrow/serial, requiring a dependency update either in evcc or gosunspec. Switching gosunspec to grid-x/serial would remove one dependency from evcc. Thanks!

andig, 2021-11-08:
Thanks for the hint. I will move gosunspec to grid-x, too

andig, 2021-11-08:
Finally fixed in https://github.com/evcc-io/evcc/commit/790155634912a4ee741d401a44d29e95fb6eafdc

closed 2021-11-08 by andig