evcc-io/evcc #1852: Easee error 403 geht nicht "von selbst" wieder weg

TheSpirit, 2021-11-09:
**Describe the bug**
Hi.
Heute war vermutlich kurzzeitig die api von Easee nicht verfügbar.
Dadurch hatte ich folgendes alle 10s im log stehen:

```
Nov 09 15:39:05 loxberry evcc[18949]: [easee ] ERROR 2021/11/09 15:39:05 connect: &{POST https://api.easee.cloud/hubs/chargers/negotiate HTTP/1.1 1 1 map[Authorization:[Bearer eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJBY2NvdW50SWQiOjExMjY2OSwiVXNlcklkIjo5MTk4OCwidW5pcXVlX25hbWUiOiIxMTI2NjkiLCJyb2xlIjoiVXNlciIsIm5iZiI6MTYzNjQ0NzUyMSwiZXhwIjoxNjM2NTMzOTIxLCJpYXQiOjE2MzY0NDc1MjF9.iDAhBUojeM7s-TWfqp1CQx7APGGayLiOBUA1oDCOJ7o]] <nil> <nil> 0 [] false api.easee.cloud map[] map[] <nil> map[]   <nil> <nil> <nil> 0x3d28000} -> 403 Forbidden
Nov 09 15:39:10 loxberry evcc[18949]: [easee ] ERROR 2021/11/09 15:39:10 connect: &{POST https://api.easee.cloud/hubs/chargers/negotiate HTTP/1.1 1 1 map[Authorization:[Bearer eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJBY2NvdW50SWQiOjExMjY2OSwiVXNlcklkIjo5MTk4OCwidW5pcXVlX25hbWUiOiIxMTI2NjkiLCJyb2xlIjoiVXNlciIsIm5iZiI6MTYzNjQ0NzUyMSwiZXhwIjoxNjM2NTMzOTIxLCJpYXQiOjE2MzY0NDc1MjF9.iDAhBUojeM7s-TWfqp1CQx7APGGayLiOBUA1oDCOJ7o]] <nil> <nil> 0 [] false api.easee.cloud map[] map[] <nil> map[]   <nil> <nil> <nil> 0x3d28000} -> 403 Forbidden
```

Mittlerweile ist die cloud api aber wieder verfügbar, der Fehler ging jedoch nicht von selbst weg.
Nach einem restart von evcc läuft jetzt alles wieder ohne Fehlermeldung im log.

**Expected behavior**
Error sollte sich selbst "löschen" wenn die api wieder erreichbar ist.

**EVCC details:**
0.68

tmathes, 2021-11-09:
Bei mir trat der Fehler auch auf. Nach Neustart von evcc auch, so dass das Docker Image direkt wieder gestoppt wurde. Da in der easee Cloud System Status heute kein Fehler gemeldet wurde, muss es evtl. an etwas anderem liegen? nach 3mal Neustart läuft es jetzt wieder..

andig, 2021-11-10:
Das teitt auch auf wenn Easee grün meldet. Je mehr Leute das beim Easee Support meldne, desto größer die Chance auf einen Fix des Root Cause. Erstmal kein akutes Thema für evcc.

TheSpirit, 2021-11-10:
Naja, ist glaube ich schon ein Thema. Denn selbst wenn die Easee cloud wieder erreichbar ist schafft es evcc nicht wieder sich mit der cloud zu verbinden. 
Das der Fehler vorkommt, dafür kann evcc nichts, aber wenn die api wieder erreichbar ist sollte sich evcc wieder mit ihr verbinden können ohne das ich was dazu tun muss.

andig, 2021-11-10:
Das Problem ist dass die Cloud ein eben noch gültiges Token ablehnt. Das ist ein Bug bei Easee der gefixt werden sollte.

TheSpirit, 2021-11-10:
Ok. Dann ist mir das klar. 
Werd dann gleich mal ein Ticker bei Easee machen.

MicFre173, 2021-11-10:
Hatte das Problem auch. Hab das Ticket bei Easee hochgevotet.

andig, 2021-11-10:
Ich hab das Thema nochmal per Mail adressiert. Habt ihr das evtl. mal so lange laufen lassen, bis der Token Refresh erfolgt? Mich würde interessieren, ob dieser erfolgreich ist?

TheSpirit, 2021-11-10:
Wie lange dauert es bis zum Token refresh?

andig, 2021-11-10:
Das dürften so 24h sein:

    "expiresIn":86400

andig, 2021-11-11:
Ggf hat sich mit https://feedback.easee.com/f/changelog/api-v1771/ auch etwas bzgl. der Refresh Tokens geändert. Falls die Annahme stimmt würde das aber erst nach 24h helfen. Sicher nur zu beurteilen wenn wir davon ein Logfile hätten und löst letztlich auch nicht den Root Cause.

TheSpirit, 2021-11-12:
Sie haben ja eine Antwort auf das Ticket gegeben.
Kann da eventuell einer der Entwickler ne bessere Antwort geben, als ich das könnte?
Danke

naltatis, 2021-11-12:
Bei mir taucht aktuell wieder der gleiche Fehler im Log auf. Hatte vor ein paar Tagen parallel den technischen Support zu dem Thema angeschrieben. Hab darauf aber auch keine Antwort bekommen.

andig, 2021-11-14:
Ich bräuchte bitte ein Logfile über >24h in dem Folgendes sichtbar ist:
- Easee auf `trace`
- Token Auth für Easee
- Fehler 403 für Easee
- Token Refresh für Easee

Im Prinzip sollte es reichen das 24h laufen zu lassen, Annahme ist dass der Fehler innerhalb von 24h bzw. kurz danach auftritt.

andig, 2021-11-15:
Es scheint als würde der Fehler nach den letzten Fixes in Master nicht mehr auftreten (48h Testzeit). Falls es wieder vorkommt bitte mit Logfile (siehe oben drüber) hier melden, dann machen wir das Issue wieder auf.

closed 2021-11-15 by andig

naltatis, 2021-11-16:
Ist leider wieder da. Dieses mal aber mit anderer Lognachricht:
```
[easee ] ERROR 2021/11/16 08:29:06 connect: Post "https://api.easee.cloud/hubs/chargers/negotiate": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)
[easee ] ERROR 2021/11/16 08:29:06 connect: Post "https://api.easee.cloud/hubs/chargers/negotiate": Post "https://api.easee.cloud/api/accounts/refresh_token": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)
```
@andig Logfile habe ich dir geschickt.

andig, 2021-11-16:
Das ist erstmal anders: das Easee Api antwortet gar nicht. Danach sollte der reconnect weiter versucht werden und irgendwann auch erfolgreich sein.

naltatis, 2021-11-16:
Die Easee API ist nicht down, "nur" unser Reconnect funktionierte scheinbar nicht. Evcc Neustart hat bei mir geholfen.

andig, 2021-11-16:
Das nutzt leider nix. Im Log ist nicht zu erkennen welchen Softwarestand das betrifft. Das Log ist allerdings eindeutig:

```
[easee ] ERROR 2021/11/16 08:29:06 connect: Post "https://api.easee.cloud/hubs/chargers/negotiate": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)
[easee ] ERROR 2021/11/16 08:29:06 connect: Post "https://api.easee.cloud/hubs/chargers/negotiate": Post "https://api.easee.cloud/api/accounts/refresh_token": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)
```

Hier antwortet einfach das API nicht. Problem bei Easee...

chrisbhv, 2022-03-15:
Dort ist alles richtig.

andig, 2022-03-15:
> Dort ist alles richtig.

Ziemlich sicher nicht. Mach bitte ein neues Thema mit den relevanten Informationen.

andig, 2022-03-15:
Ich glaub Dir das, ist aber trotzdem die naheliegendste Erklärung. Mach gleich ein log trace von der Easee dazu dann sehen wir ja was los ist.

chrisbhv, 2022-03-16:
So ich habe den Fehler gefunden!
Ich habe den falschen Namen für die Wallbox eingetragen.
Jetzt läuft alles!

AlexAhrens, 2022-10-03:
Ich würde dies gern wieder öffnen, da der Fehler nun beim Hinzufügen einer neuen easee-box wieder auftritt. 

Log: 

`cannot create charger 'Loki'
cannot create charger 'template'
cannot create charger 'easee'
Post "https://api.easee.cloud/api/accounts/login"
dial tcp
lookup api.easee.cloud: i/o timeout`

andig, 2022-10-03:
Das Easee Api antwortet nicht. Bitte an den Support wenden- da können wir nix tun.