evcc-io/evcc #1861: Vehicle SoC from BMW i3s ends in 410

TheSpirit, 2021-11-11:
Aktuell kann der SoC Wert beim BMW i3s nicht mehr abgefragt werden.
Siehe Diskussion https://github.com/evcc-io/evcc/discussions/1858

Die iOS App [Pump](https://pumpapp.co/de/index.html) (Routenplanungssoftware für Elektroautos) kann sich mit Fahrzeugen verschiedener Hersteller verbinden um den aktuellen SoC des Fahrzeugs auszulesen und so die Planung genauer zu machen.
Hier funktioniert nacht wie vor die Abfrage des SoC.

Des weiteren nutze ich ein Script auf iOS und Mac für [Scriptable](https://scriptable.app), welche auch Fahrzeugdaten des i3s ausliehst.
Auch hier funktioniert noch immer die Abfrage der Fahrzeugdaten inkl. SoC.
Das Script ist in folgendem Forum zu finden: https://www.goingelectric.de/forum/viewtopic.php?f=69&t=62100

andig, 2021-11-11:
Dieses https://www.goingelectric.de/forum/download/file.php?id=106835?

TheSpirit, 2021-11-11:
genau

andig, 2021-11-11:
Könntest du noch ein Logfile vom evcc Login teilen? Wo läuft das schief?

TheSpirit, 2021-11-11:
wenn ich folgendes in der console ausführe
`evcc vehicle`
kommt zurück
`.SoC:          unexpected status: 410
Charge status: unexpected status: 410
Range:         unexpected status: 410
Odometer:      unexpected status: 410
Capacity:      42kWh`

TheSpirit, 2021-11-11:
oder was meinst du genau mit "vom evcc Login"?

andig, 2021-11-11:
Ich bräuchte Zugangsdaten zum testen (cpuidle@gmx.de)

TheSpirit, 2021-11-11:
also den BMW login?

andig, 2021-11-11:
Naklar. Welchen sonst ;)

tmeidinger, 2021-11-11:
Wenn's was hilft könnte ich testen. Credentials (nicht mein Auto) kann ich leider nicht rausgeben.

closed 2021-11-14 by andig