evcc-io/evcc #1871: Add heatpump control

johschmitz, 2021-11-13:
Some people also have a heatpump as part of their setup. Would it be possible to also ads control of those devices? That means we can control when the vehicle should be charged or when the home should be heated first.

andig, 2021-11-13:
Interessante Idee. Dann wären wir allerdings eine Art SAE? Nicht unmöglich, aber doch ein wenig Aufwand.

johschmitz, 2021-11-13:
Es macht halt schlussendlich nur Sinn alles gemeinsam zu Steuern. Wenn die Wärmepumpe nicht mit rein kommt wird es für immer mehr Leute dann halt nur begrenzt Sinn ergeben evcc-io zu nutzen, fände ich schade, da das Projekt ansonsten für mich ziemlich spannend aussieht.

TheSpirit, 2021-11-13:
Meine Hauptsteuerung ist bei mir Loxone. Die kennt auch die Wärmepumpe. Und evcc wird einfach per mqtt „eingebunden“. 
Klappt super.

johschmitz, 2021-11-13:
@TheSpirit das kling sehr gut, kann man denn dann selbst den Kompromiss zwischen Heizen und Fahrzeug laden beeinflussen?

andig, 2021-11-14:
Du kannst einfach das Heizen selbst steuern und evcc nimmt die restliche Energie fürs Auto. Dafür ist der PV Modus da.

closed 2021-11-14 by andig