evcc-io/evcc #1873: Synology Docker SMA

Schwaleimker, 2021-11-13:
Ich habe auf meiner Synology EVCC im Docker Container installiert . Es läuft soweit alles .
Nun würde ich gerne meine SMA Anlage ins Sunny Portal einbinden.

Dies habe ich ausgeführt
For use with SMA Sunny Home Manager, evcc needs to generate a unique device id. On Linux, we're using machine-id for this purpose, make sure to mount the host folders into the container:

docker run -v /etc/machine-id:/etc/machine-id -v /var/lib/dbus/machine-id:/var/lib/dbus/machine-id andig/evcc ...

dann bekomme ich beim starten des Containers  /var/lib/dbus/machine-id does not Exits.

was kann ich machen?

andig, 2021-11-13:
Ich denke die Fehlermeldung sagt alles?

Schwaleimker, 2021-11-14:
Sorry.. nicht richtig gefragt. Wie erzeuge und befülle sie auf meinem Host.

tmeidinger, 2021-11-14:
Versuch mal docker run -v /etc/machine-id:/var/lib/dbus/machine-id andig/evcc.

closed 2021-11-14 by andig