evcc-io/evcc #1878: TPLink linter error

andig, 2021-11-14:
```
❯ semgrep -f ../semgrep-go .
Running 45 rules...
charger/tplink.go
rule:semgrep-go.read-io-eof: Read() can return n bytes and io.EOF
187:	len, err := conn.Read(resp)
188:	if err != nil {
189:		return err
190:	}
```

@thierolm ggf. ist das eine Fehlerquelle?

thierolm, 2021-11-14:
Lese dazu gerade [https://github.com/golang/go/issues/40385](https://github.com/golang/go/issues/40385) ...
Muss ich `if err != nil || len != 0` Abfragen, um den Linter zu entspannen?

thierolm, 2021-11-14:
Die folgenden Programmschritte erwarten ja mindestens 6 Bytes, um mindestens ein leeres JSON Object `{}` definieren zu können.

[image]

andig, 2021-11-14:
Ich frag mich eher wie‘s in der Sache ist. Solange immer evcc die Verbindung schließen muss kann man es durchaus als Fehler sehen wenn der TPlink das macht. Insofern wäre der Code erstmal korrekt. Falls das der Fall ist würde ich schauen, wir man semgrep zum Schweigen bringen kann.

thierolm, 2021-11-14:
Ich hatte mir ja die TP-Link Basis Infos über https://www.softscheck.com/en/reverse-engineering-tp-link-hs110/ und https://github.com/softScheck/tplink-smartplug geholt. Ob die Steckdose die Verbindung eigenständig schließt, geht daraus nicht hervor.
Allerdings ist ja das JSON in der Response immer unterschiedlich lang und beim read ist mir beim Testen nie der io.EOF error unter gekommen ..., also schließt die Dose nach der Response die Verbindung auch nicht.

andig, 2021-11-14:
Dann lassen wir das einfach so... danke für's rein schauen!

closed 2021-11-14 by andig