evcc-io/evcc #1896: Tesla API neue Funktion - set_charging_amps

martinez81, 2021-11-18:
Tesla aktualisiert gerade die Fahrzeugflotte mit einer Funktion den Ladestrom per App einzustellen.
Hier ist die Funktion bereits beschrieben:
[Tesla API](https://tesla-api.timdorr.com/miscellaneous/endpoints)
In Zeile 107 bis 110 ist der relevante Bereich.

Wenn EVCC diesen Teil der API nutzt könnte man mit jeder Wallbox PV Laden beim einem Tesla durchführen.
Die Wallbox müsste nicht mehr direkt gesteuert werden. Start / Stopp, Ladestrom einstellen würde alles über die Tesla API laufen.

Was haltet ihr davon?

premultiply, 2021-11-18:
Welchen Vorteil sollte das haben?

martinez81, 2021-11-18:
EVCC würde mit jeder Wallbox funktionieren. Weil die Wallbox keine Rolle mehr spielt.
Ein Meter bei der Wallbox ist theoretisch auch nicht mehr nötig. Von der Tesla API kommt alles was nötig ist.
Jeder der einen Tesla hat und eine PV Anlage mit min. Wechselrichterwerte könnte mit PV Strom laden.

DerAndereAndi, 2021-11-18:
Siehe https://github.com/evcc-io/evcc/discussions/1505#discussioncomment-1562114

DerAndereAndi, 2021-11-18:
Ich würde dich bitten, Ideen zuerst unter Discussions zu führen. Mache das hier mal zu, da es schon besprochen wurde.

closed 2021-11-18 by DerAndereAndi

andig, 2021-11-18:
Lasst uns das bitte in dem Discussions Thread weiter besprechen wie Andi sagte.