evcc-io/evcc #1910: BMW i3 vehicle soc: unexpected length: 2 Reopen?

Clanchef, 2021-11-21:
Bei mir wird der SOC bzw. das Fahrzeug leider immer noch nicht erkannt, 
Wurde das Problem bereits gefixt?
vehicle soc: unexpected length: 2 

```
vehicles:
- name: bmwi3s
  type: bmw
  title: BMWi3s
  capacity: 42
  user: xxxxxxxxxxxx
  password: xxxxxxxxxxxxx
  vin: xxxxxxxxxxxxxxxx
  cache: 5m
```

Der Fix hier hat bei mir nicht geholfen:  #https://github.com/evcc-io/evcc/pull/1876

andig, 2021-11-21:
Hast Du 2 bmws in dem Account? Dann bitte für einen Zugangsdaten an cpuidle@gmx.de damit ixh mir das anschauen kann.

Clanchef, 2021-11-21:
Ja ich habe 2 BMWs ... aber Zugangsdaten rausgeben?

andig, 2021-11-21:
Das musst Du natürlich selbst entscheiden

Hofyyy, 2021-11-29:
> Das musst Du natürlich selbst entscheiden

Hey, brauchst du nur realen Zugriff auf das BMW API, oder muss es genau ein Auto mit 42KWh Akku sein?

andig, 2021-12-02:
> > Das musst Du natürlich selbst entscheiden
> 
> Hey, brauchst du nur realen Zugriff auf das BMW API, oder muss es genau ein Auto mit 42KWh Akku sein?

Ich brauche ein Konto mit 2 Fahrzeugen zum Test.

andig, 2021-12-05:
Closed in https://github.com/evcc-io/evcc/commit/111d499b69f13f0be502679bf897a08b16956dbb

closed 2021-12-05 by andig