evcc-io/evcc #1922: Fahrzeug nicht erkannt - Ioniq vs Twingo

andig, 2021-11-22:
### Discussed in https://github.com/evcc-io/evcc/discussions/1914

<div type='discussions-op-text'>

<sup>Originally posted by **SwaY88LA** November 21, 2021</sup>
Hallo,

ich hab seit kurzem ein zweites EVs, einmal einen Hyundai Ioniq BEV aus 2020 und nun noch einen Twingo electric.

Wenn kein Fahrzeug angesteckt ist steht nun ja nicht mehr der Ioniq als Standart drin sondern "Fahrzeug".
Ich habe ihn jedoch heute Nachmittag angesteckt und als ich später mal geguckt habe war der Ladebalken auch als "verbunden" dargestellt, aber immer noch als "Fahrzeug" betitelt. Über die Schaltfläche Start kann ich den Ladevorgang auch starten, jedoch bleibt auch nach einer Stunde noch "Fahrzeug" aktiv und "lädt" statt Ioniq und Anzeige des SOC.

Liegt das daran, dass der Hyundai keinen Charge Status ausgibt wie der Twingo?
Wobei ich den Twingo aber noch testen muss, ob dieser richtig erkannt wird, hier ist nur die Batterie noch voll.

Was mir beim Hyundai aufgefallen ist - wenn er nicht lädt steht bei der Finish time "not available". Wenn er lädt, siehe unten.
Falls es beim Hyundai an dem fehlenden Charge Status über die API liegt, ließe sich vielleicht an der Finish time erkennen, dass das Fahrzeug gerade aktiv ist und lädt?

Zum einen wäre die Frage ob ich vielleicht was falsch konfiguriert habe und der Hyundai deswegen nicht erkannt wird?
Und zum anderen, falls es wie vermutet am fehlenden Status liegt, ob sich das über die Finish time umsetzen ließe (nur für Hyundai)?

Verwendete Version ist die 0.69

VG
Holger

```
vehicles:
- name: hyundai
  type: hyundai
  title: Ioniq
  capacity: 38
  user: ...
  password: ...
  
- name: renault
  type: renault
  title: Twingo
  capacity: 22 # kWh
  user: ...
  password: ...
  vin: ...
  cache: 5m
```
```
loadpoints:
- title: Garage # display name for UI
  charger: wallbe # charger
  meters:
    charge: charge # charge meter
  # vehicle: hyundai
  vehicles: # use if multiple vehicles allowed to charge on this loadpoint
  - hyundai
  - renault
```

```
[hyundai] DEBUG 2021/11/21 17:39:30 found vehicle:
hyundai
-------
....SoC:     67%
Range:       189km
Finish time: 2021-11-21 22:29:00 +0100 CET
Capacity:    38kWh

renault
-------
SoC:            99%
Charge status:  A
Range:          181km
Odometer:       25km
Finish time:    2021-11-23 08:13:00 +0000 UTC
Climate active: false
Outside temp:   0.0°C
Target temp:    20.0°C
Capacity:       22kWh

```
</div>

andig, 2021-11-22:
@SwaY88LA hast Du bitte mal ein Logfile des Vorgangs? Bitte auch die neue Version 0.70 testen, da sich die Erkennung etwas geändert hat.

SwaY88LA, 2021-11-22:
Zuerst wollte das UI nach dem Update auf 0.70 nicht mehr funktionieren, darum bin ich erstmal wieder auf die 0.69 zurück.

Und hier gibts wohl doch ein Problem mit der Konfiguration meiner Fahrzeuge:

```
Nov 22 15:30:41 solaranzeige evcc[1283]: [main  ] FATAL 2021/11/22 15:30:41 invalid vehicle: hyundai renault
Nov 22 15:30:41 solaranzeige systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Nov 22 15:30:41 solaranzeige systemd[1]: evcc.service: Failed with result 'exit-code'.
Nov 22 15:30:41 solaranzeige systemd[1]: evcc.service: Service RestartSec=100ms expired, scheduling restart.
Nov 22 15:30:41 solaranzeige systemd[1]: evcc.service: Scheduled restart job, restart counter is at 7.
Nov 22 15:30:41 solaranzeige systemd[1]: Stopped evcc.
```

Grundsätzlich scheinen sie ja richtig konfiguriert zu sein, die Daten aus der API kommen ja an.

Nun hab ich die evcc.yaml nochmal folgendermaßen geändert:

```
loadpoints:
- title: Garage # display name for UI
  charger: wallbe # charger
  meters:
    charge: charge # charge meter
  vehicle: hyundai
 # vehicles: # use if multiple vehicles allowed to charge on this loadpoint
 # - hyundai
 # - renault
  mode: pv
```

Der Fehler lt. Log bleibt aber der Selbe wie oben. 
Jedoch ist die Anzeige im UI nun wieder in Ordnung und zeigt den Hyundai mit korrektem SoC.

Nachdem ich auch noch die vehicle definition vom Twingo mit # zu Beginn der Zeilen deaktivert hab, ist der Fehler weg.

Nun hab ich zuerst die vehicle definition vom Twingo wieder aktiviert. Dabei ist mir aufgefallen, dass ich zwischen Passwort vom Hyundai und "- renault" keine leere Zeile hatte. Die hab ich zusätzlich eingefügt und nach einem Neustart gab es den Fehler von Anfang des Beitrags nicht mehr.

Im Anschluss hab ich die Loadoints wieder auf "vehicles" umgestellt und es gibt weiterhin keinen Fehler.

Aktuell siehts also wieder so aus:

```
vehicles:
- name: hyundai
  type: hyundai
  title: Ioniq
  capacity: 38
  user: ...
  password: ...

- name: renault
  type: renault
  title: Twingo
  capacity: 22 # kWh
  user: ...
  password: ...
  vin: ...
  cache: 5m

loadpoints:
- title: Garage # display name for UI
  charger: wallbe # charger
  meters:
    charge: charge # charge meter
 # vehicle: hyundai
  vehicles: # use if multiple vehicles allowed to charge on this loadpoint
  - hyundai
  - renault
```

Und das Log sagt folgendes
```
Nov 22 15:52:19 solaranzeige systemd[1]: Started evcc.
Nov 22 15:52:31 solaranzeige evcc[663]: [lp-1  ] INFO 2021/11/22 15:52:31 loadpoint 1:
Nov 22 15:52:31 solaranzeige evcc[663]: [lp-1  ] INFO 2021/11/22 15:52:31   mode:      pv
Nov 22 15:52:31 solaranzeige evcc[663]: [lp-1  ] INFO 2021/11/22 15:52:31   charger:   power ✗ energy ✗ currents ✗ timer ✓
Nov 22 15:52:31 solaranzeige evcc[663]: [lp-1  ] INFO 2021/11/22 15:52:31   meters:    charge ✓
Nov 22 15:52:31 solaranzeige evcc[663]: [lp-1  ] INFO 2021/11/22 15:52:31     charge:  power ✓ energy ✓ currents ✓
Nov 22 15:52:31 solaranzeige evcc[663]: [lp-1  ] INFO 2021/11/22 15:52:31   vehicles:  ✓
Nov 22 15:52:31 solaranzeige evcc[663]: [lp-1  ] INFO 2021/11/22 15:52:31     car 0:   range ✓ finish ✓ status ✗ climate ✗
Nov 22 15:52:31 solaranzeige evcc[663]: [lp-1  ] INFO 2021/11/22 15:52:31     car 1:   range ✓ finish ✓ status ✓ climate ✓
Nov 22 15:52:31 solaranzeige evcc[663]: [lp-1  ] DEBUG 2021/11/22 15:52:31 charge power: 6W
Nov 22 15:52:31 solaranzeige evcc[663]: [lp-1  ] DEBUG 2021/11/22 15:52:31 charge currents: [0.033 0 0]A
Nov 22 15:52:31 solaranzeige evcc[663]: [lp-1  ] DEBUG 2021/11/22 15:52:31 charger status: B
Nov 22 15:52:32 solaranzeige evcc[663]: [lp-1  ] DEBUG 2021/11/22 15:52:32 vehicle status: A (Twingo)
Nov 22 15:52:32 solaranzeige evcc[663]: [lp-1  ] WARN 2021/11/22 15:52:32 target charging: not possible
Nov 22 15:52:32 solaranzeige evcc[663]: [lp-1  ] DEBUG 2021/11/22 15:52:32 max charge current: 0A = 0A + -2.7A (620W @ 1p)
Nov 22 15:52:32 solaranzeige evcc[663]: [lp-1  ] DEBUG 2021/11/22 15:52:32 reset pv enable timer: 1m0s
Nov 22 15:52:32 solaranzeige evcc[663]: [lp-1  ] DEBUG 2021/11/22 15:52:32 pv enable timer: keep disabled
Nov 22 15:52:32 solaranzeige evcc[663]: [lp-1  ] DEBUG 2021/11/22 15:52:32 pv max charge current: 0A
Nov 22 15:52:33 solaranzeige evcc[663]: [lp-1  ] DEBUG 2021/11/22 15:52:33 charge power: 6W
Nov 22 15:52:33 solaranzeige evcc[663]: [lp-1  ] DEBUG 2021/11/22 15:52:33 charge currents: [0.033 0 0]A
Nov 22 15:53:14 solaranzeige evcc[663]: [lp-1  ] DEBUG 2021/11/22 15:53:14 charge power: 6W
Nov 22 15:53:14 solaranzeige evcc[663]: [lp-1  ] DEBUG 2021/11/22 15:53:14 charge currents: [0.033 0 0]A
Nov 22 15:53:14 solaranzeige evcc[663]: [lp-1  ] DEBUG 2021/11/22 15:53:14 charger status: B
Nov 22 15:53:14 solaranzeige evcc[663]: [lp-1  ] DEBUG 2021/11/22 15:53:14 vehicle status: A (Twingo)
```

Nach starten des Ladevorgangs:

```
pi@solaranzeige:~ $ evcc -l debug vehicle
[main  ] INFO 2021/11/22 16:01:19 evcc 0.69 (2c1d727)
[main  ] INFO 2021/11/22 16:01:19 using config file /etc/evcc.yaml
[hyundai] DEBUG 2021/11/22 16:01:22 found vehicle:
hyundai
-------
SoC:         68%
Range:       190km
Finish time: 2021-11-22 20:41:00 +0100 CET
Capacity:    38kWh

renault
-------
SoC:            85%
Charge status:  A
Range:          153km
Odometer:       31km
Finish time:    2021-11-24 06:23:00 +0000 UTC
Climate active: false
Outside temp:   0.0°C
Target temp:    20.0°C
Capacity:       22kWh
```

Und den Inhalt des angehängten Logs
[log.txt](https://github.com/evcc-io/evcc/files/7582122/log.txt)

Letztendlich war wohl doch irgendwo ein Fehler in er Config, den scheinbar beheben konnte.
Trotzdem hab ich noch das eingangs beschriebene Verhalten, dass der Ioniq nicht erkannt wird, wenn er geladen wird wodurch kein SoC angezeigt wird und Target Charging nicht möglich ist.

Soweit zur 0.69 - die 0.70 will ich gleich nochmal probieren.

premultiply, 2021-11-22:
Der Ioniq liefert keinen `charge status` welcher aber für eine Erkennung obligatorisch ist.
Somit bliebe ja nur das Ausschlussverfahren: Wenn der Twingo nicht lädt müsste es ja der Ioniq (als Fallback) sein.
Und was macht man dann wenn es ein fremdes Fahrzeug ist?
Ich wundere mich dass das überhaupt schon mal funktioniert haben soll - mit mehreren Fahrzeugen.

SwaY88LA, 2021-11-22:
So die 0.70 will nun auch wie sie soll
Hier zeigt sich das selbe Verhalten, debug vehicle liefert die Daten, Log siehe Anhang
[log.txt](https://github.com/evcc-io/evcc/files/7582196/log.txt)

SwaY88LA, 2021-11-22:
> Der Ioniq liefert keinen `charge status` welcher aber für eine Erkennung obligatorisch ist. Somit bliebe ja nur das Ausschlussverfahren: Wenn der Twingo nicht lädt müsste es ja der Ioniq (als [...]

Ich denke der Ioniq selber liefert es schon, in der Bluelink App wird mir angezeigt ob 
not connected
connected oder 
connected + charging

Im Prinzip wäre für meinen Zweck der Ioniq als Rückfallebene ok und als Workaround nutzbar. Wenn es wirklich mal vorkommt, dass vielleicht mal ein anderes Fahrzeug an der Wallbox (in der Doppelgarage, in der die beiden Fzg. stehen) lädt, könnte ich die falsche Anzeige im UI für mich ignorieren.

premultiply, 2021-11-22:
Ja, das habe ich auch gerade nachgeschaut. Offensichtlich ist der Ladezustand in der API-Response doch drin, wird aber momentan von EVCC nicht verarbeitet. Da gibt es im `evStatus` `batteryCharge` und `batteryStatus`.

Nach https://github.com/evcc-io/evcc/issues/618#issuecomment-767790776 sieht die komplette Antwort ungefähr so aus:
```json
{
  "retCode": "S",
  "resCode": "0000",
  "resMsg": {
    "airCtrlOn": false,
    "engine": false,
    "doorLock": true,
    "doorOpen": {
      "frontLeft": 0,
      "frontRight": 0,
      "backLeft": 0,
      "backRight": 0
    },
    "trunkOpen": false,
    "airTemp": {
      "value": "00H",
      "unit": 0,
      "hvacTempType": 1
    },
    "defrost": false,
    "acc": false,
    "evStatus": {
      "batteryCharge": false,
      "batteryStatus": 93,
      "batteryPlugin": 2,
      "remainTime2": {
        "etc1": {
          "value": 55,
          "unit": 1
        },
        "etc2": {
          "value": 90,
          "unit": 1
        },
        "etc3": {
          "value": 55,
          "unit": 1
        },
        "atc": {
          "value": 75,
          "unit": 1
        }
      },
      "drvDistance": [
        {
          "rangeByFuel": {
            "evModeRange": {
              "value": 270,
              "unit": 1
            },
            "totalAvailableRange": {
              "value": 270,
              "unit": 1
            }
          },
          "type": 2
        }
      ],
      "reservChargeInfos": {
        "reservChargeInfo": {
          "reservChargeInfoDetail": {
            "reservInfo": {
              "day": [
                9
              ],
              "time": {
                "time": "1200",
                "timeSection": 0
              }
            },
            "reservChargeSet": false,
            "reservFatcSet": {
              "defrost": false,
              "airTemp": {
                "value": "00H",
                "unit": 0,
                "hvacTempType": 0
              },
              "airCtrl": 0,
              "heating1": 0
            }
          }
        },
        "offpeakPowerInfo": {
          "offPeakPowerTime1": {
            "starttime": {
              "time": "1200",
              "timeSection": 0
            },
            "endtime": {
              "time": "1200",
              "timeSection": 0
            }
          },
          "offPeakPowerFlag": 0
        },
        "reserveChargeInfo2": {
          "reservChargeInfoDetail": {
            "reservInfo": {
              "day": [
                9
              ],
              "time": {
                "time": "1200",
                "timeSection": 0
              }
            },
            "reservChargeSet": false,
            "reservFatcSet": {
              "defrost": false,
              "airTemp": {
                "value": "00H",
                "unit": 0,
                "hvacTempType": 0
              },
              "airCtrl": 0,
              "heating1": 0
            }
          }
        },
        "reservFlag": 0,
        "ect": {
          "start": {
            "day": 9,
            "time": {
              "time": "1200",
              "timeSection": 0
            }
          },
          "end": {
            "day": 9,
            "time": {
              "time": "1200",
              "timeSection": 0
            }
          }
        },
        "targetSOClist": [
          {
            "targetSOClevel": 100,
            "dte": {
              "rangeByFuel": {
                "evModeRange": {
                  "value": 290,
                  "unit": 1
                },
                "totalAvailableRange": {
                  "value": 290,
                  "unit": 1
                }
              },
              "type": 2
            },
            "plugType": 0
          },
          {
            "targetSOClevel": 100,
            "dte": {
              "rangeByFuel": {
                "evModeRange": {
                  "value": 290,
                  "unit": 1
                },
                "totalAvailableRange": {
                  "value": 290,
                  "unit": 1
                }
              },
              "type": 2
            },
            "plugType": 1
          }
        ]
      }
    },
    "ign3": true,
    "hoodOpen": false,
    "transCond": true,
    "steerWheelHeat": 0,
    "sideBackWindowHeat": 0,
    "tirePressureLamp": {
      "tirePressureLampAll": 0,
      "tirePressureLampFL": 0,
      "tirePressureLampFR": 0,
      "tirePressureLampRL": 0,
      "tirePressureLampRR": 0
    },
    "battery": {
      "batSoc": 83,
      "batState": 0
    },
    "time": "20210126201754"
  },
  "msgId": "11d03105-e28b-4327-888c-1186858e4b7e"
}
```

SwaY88LA, 2021-11-22:
>     * Ging das tatsächlich früher mal? Das wäre mit Log zu belegen @SwaY88LA

So gesehen nein. Ich hatte zuerst den Ioniq und dieser war als einzelnes Fahrzeug konfiguriert.
Der Twingo ist seit einer Woche da und das "Problem" zeigt sich nun, wenn ich ihn in die Config mit aufnehme.

Aber dann ist das aktuelle Verhalten ja Stand der Technik.
Und wenn der Charge Status künftig vielleicht mit ausgelsen werden kann wärs natürlich super.

Kann ich hier irgendwie unterstützen?
Der Ioniq hängt eh gerade am Kabel - wie könnte ich für meinen denn diese Abfrage wie premultiply machen?

Dann kann ich gern die verschiedenen Daten für die 3 Zustände auslesen.

andig, 2021-11-22:
> Dann kann ich gern die verschiedenen Daten für die 3 Zustände auslesen.

@SwaY88LA schau mal https://github.com/evcc-io/evcc/pull/1923, da ist eine Frage drin

premultiply, 2021-11-22:
Ich denke um das sauber zu halten sollten "non-API-Fahrzeuge" eben wie Fremdfahrzeuge behandelt werden wenn über `vehicles` definiert wird. Alternativ wird dann über RFID manuell zugeordnet.

Bei Definition über `vehicle` gibt es nur eine einzige feste Zuordnung die immer das angegebene Fahrzeug fest identifiziert.

andig, 2021-11-22:
> Ich denke um das sauber zu halten sollten "non-API-Fahrzeuge" eben wie Fremdfahrzeuge behandelt werden wenn über vehicles definiert wird. Alternativ wird dann über RFID manuell zugeordnet.

Finde ich gut. Wir haben ja noch das Thema "Parkplatz" um dafür eine Lösung zu finden (@naltatis).

>> Ging das tatsächlich früher mal? Das wäre mit Log zu belegen @SwaY88LA

> So gesehen nein. Ich hatte zuerst den Ioniq und dieser war als einzelnes Fahrzeug konfiguriert.

Dann wäre hier alles erledigt.

closed 2021-11-22 by andig