evcc-io/evcc #1926: NRGkick bluetooth: Operation failed with ATT error: 0x0e

noradtux, 2021-11-22:
**Describe the bug**
I get the message "Operation failed with ATT error: 0x0e" more or less regularly

**To Reproduce**
Steps to reproduce the behavior:
- Use NRGKick BT

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.70 (9cfbe16)
```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
config
======

grid: grid
----------
Power:          812W
Current L1..L3: 2.74A 1.39A 0.12A

pv 0: pv
--------
Power: 0W

battery: battery
----------------
Power: 0W
SoC:   5%

loadpoint 1
===========

charger: nrgkick
----------------
Power:          0W
Energy:         196.2kWh
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        false
</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
uri: 0.0.0.0:7070 # uri for ui
interval: 1s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
sponsortoken: XXX

# log settings
#log: debug
log: error
levels:
  #nrg-bt: trace
  #core: debug
  #lp-1: debug
  #lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters
meters:
- name: grid
  type: custom
  power:
    type: mqtt
    topic: Pwr/meter/MeterPgrid_Tot
    scale: -1
  currents:
[... 560 lines cut ...]
[cache ] DEBUG 2021/11/22 19:56:00 chargePower: 0
[lp-1  ] DEBUG 2021/11/22 19:56:02 charge currents: [0 0 0]A
[cache ] DEBUG 2021/11/22 19:56:02 chargeCurrents: [0 0 0]
[cache ] DEBUG 2021/11/22 19:56:02 chargeCurrent: 0
[cache ] DEBUG 2021/11/22 19:56:02 chargedEnergy: 0
[cache ] DEBUG 2021/11/22 19:56:02 chargeDuration: 0s
[lp-1  ] DEBUG 2021/11/22 19:56:05 charger status: B
[lp-1  ] DEBUG 2021/11/22 19:56:05 soc estimated: 80.00% (vehicle: 80.00%)
[lp-1  ] DEBUG 2021/11/22 19:56:05 vehicle soc: 80%
[cache ] DEBUG 2021/11/22 19:56:05 connected: true
[cache ] DEBUG 2021/11/22 19:56:05 charging: false
[cache ] DEBUG 2021/11/22 19:56:05 enabled: false
[cache ] DEBUG 2021/11/22 19:56:05 vehicleSoC: 80
[lp-1  ] DEBUG 2021/11/22 19:56:07 max charge current: 0A = 0A + -1.25A (865W @ 3p)
[lp-1  ] DEBUG 2021/11/22 19:56:07 reset pv enable timer: 1m0s
[lp-1  ] DEBUG 2021/11/22 19:56:07 pv enable timer: keep disabled
[lp-1  ] DEBUG 2021/11/22 19:56:07 pv max charge current: 0A
[cache ] DEBUG 2021/11/22 19:56:07 timerSet: false
[cache ] DEBUG 2021/11/22 19:56:07 timerActive: false
[cache ] DEBUG 2021/11/22 19:56:07 timerProjectedEnd: 2021-11-22 19:56:00 +0100 CET
[lp-1  ] DEBUG 2021/11/22 19:56:09 charge power: 0W
[cache ] DEBUG 2021/11/22 19:56:09 chargePower: 0
[lp-1  ] DEBUG 2021/11/22 19:56:11 charge currents: [0 0 0]A
[site  ] DEBUG 2021/11/22 19:56:11 ----
</pre></details>

premultiply, 2021-11-22:
Stell mal bitte dein `interval` auf mindestens 18 Sekunden und probiere ob das Problem dann weg ist.

noradtux, 2021-11-23:
Hab das Interval auf 20s gesetzt. Keine Änderung.

andig, 2021-11-30:
Ich fürchte da können wir nicht helfen. Die Steuerung über Bluetooth ist nicht der glücklichste Ansatz.

closed 2021-11-30 by andig