evcc-io/evcc #1957: Was directed by the comand to open an Issue - why is that

lmai95, 2021-11-29:
This was the Output from the detect function.
`sudo evcc detect

Auto detection will now start to scan the network for available devices.
Scanning focuses on devices that are commonly used that are detectable with reasonable efforts.
On successful detection, suggestions for EVCC configuration can be made. The suggestions should simplify
configuring EVCC but are probably not sufficient for fully automatic configuration.

+--------------+----------+----------+-------------------------------------------------------------------------------------------------------+
|      IP      | HOSTNAME |   TASK   |                                                DETAILS                                                |
+--------------+----------+----------+-------------------------------------------------------------------------------------------------------+
| 192.168.1.33 |          | sunspec  | {"IP":"192.168.1.33","Port":502,"ModbusResult":{"SlaveID":1,"Model":1,"Point":"Mn","Value":"SolarEdge |
|              |          |          | "}}                                                                                                   |
+              +----------+----------+-------------------------------------------------------------------------------------------------------+
|              |          | inverter | {"IP":"192.168.1.33","Port":502,"ModbusResult":{"SlaveID":1,"Model":103,"Point":"W","Value":0}}       |
+--------------+----------+----------+-------------------------------------------------------------------------------------------------------+

Please open https://github.com/evcc-io/evcc/issues/new in your browser and copy the
results above into a new issue. Please tell us:

	1. Is the scan result correct?
	2. If not correct: please describe your hardware setup.
`
The IP address and the port is the right one. The system consist of a Solar Edge Inverter (SE7K-RWS48BNN4) and a solar Edge Meter (SE-MTR-3Y-400V-A). The battery hasn't been delivered yet, but will be soon, I think it will connect via CAN. Is it possible to connect an ABL eMH1 with the RS485 of the Inverter or do need a converter for this task.

andig, 2021-11-29:
Please format. Your post is unreadable.

andig, 2021-11-29:
Are you saying that the output is incorrect? If yes, what is incorrect? For all other questions, pelase use discussison.

lmai95, 2021-11-29:
As far a I can tell these are correct but the command commanded  me to open to open a new issue. 
`Please open https://github.com/evcc-io/evcc/issues/new in your browser and copy the results above into a new issue. Please tell us`

andig, 2021-11-29:
Das ist super, danke. Sieht aus, als könnten wir diesen check langsam deaktivieren…

/cc @DerAndereAndi

closed 2021-11-29 by andig