evcc-io/evcc #1961: BMW: add timeout param

andig, 2021-11-30:
Should fix api slow in some cases

Hofyyy, 2021-11-30:
Also erstmal mein Log zu der Geschichte.
Was mir auffällt ist, das er recht häufig das API anfragt
Siehe:
[log.txt](https://github.com/evcc-io/evcc/files/7628637/log.txt)
e.

Config:
soc:
    poll:
      mode: connected
      interval: 5m

Hofyyy, 2021-11-30:
Ein "sudo systemctl stop evcc" und "sudo systemctl start evcc" hat das Problem behoben, das spricht wie ich finde gegen den Timeout Wert.

closed 2021-11-30 by andig

Hofyyy, 2021-11-30:
Warte mal :-) Kann man sicherstellen, dass das API nur alle 5 min abgefragt wird auch im Fehlerfall und das er sich wieder fängt? Ich lese im text das man aufpassen muss das man nicht gebannt wird, wenn man zu oft anfragen sendet. Oder kommt nur das logging jede Sekunde und die Anfrage ans API nur alle 5 min?

Hofyyy, 2021-11-30:
des Weiteren stellt sich ja schon die Frage warum es zu der Fehlermeldung kommt. Gibt es detailliertes Socket logging was ich aktivieren könnte?

andig, 2021-11-30:
schau erstmal was bei Dir los war. Auch bei Fehler sollte es nicht exzessiv werden.

Hofyyy, 2021-11-30:
Ich stelle mal zwei Thesen auf, mit dem Wagnis das ich noch nicht so viel vom Sourcecode verstehe.
- Das API wird beim Laden immer gepollt. Alle 10 Sek. da dann socPollAllowed() immer True liefert.
Frage: Soll das so, sind 10 Sek sicher, oder lieber den eingestellten intervall?

Mein API Token könnte abgelaufen sein:
- https://customer.bmwgroup.com/gcdm/oauth/authenticate": context deadline exceeded

Hier wäre interessant wie oft und wann "configureVehicles()" aufgerufen wird.
Wird das nur einmal aufgerufen? Und was passiert wenn es da einen Fehler gab?

Das würde erklären warum es nach einem neuen "Start" "Stop" wieder ging.

Hofyyy, 2021-11-30:
Ok die erste Aussage ziehe ich schon mal zurück nachdem ich den Cache: interval, mit 15 min entdeckt habe. 

ich werde mich die Tage mal einlesen und dem auf den Grund gehen.

andig, 2021-11-30:
...und bei vermeintlichen Fehlern Deine Config angeben wie angefragt. Meist liegt das Problem da...