evcc-io/evcc #1970: Tesla / vehicle unavailable in dashboard

Brutalok, 2021-12-01:
Tesla Model Y bzw. vehicle wird im evcc dashboard nicht erkannt und als "unavailabe" angezeigt, connection Status "verbunden". 
Im debug wird folgender Fehler ausgegeben (Auszug):

[lp-2  ] INFO 2021/12/01 15:37:41 vehicle updated: unknown -> unavailable
[lp-2  ] ERROR 2021/12/01 15:37:51 vehicle soc: vehicle not available: cannot create vehicle 'Tesla': 1 error(s) decoding:

* '' has invalid keys: password, user

Ich habe bestimmt einen Fehler in der Konfig, es wäre nett wenn jemand einen Blick drauf werfen könnte.

[image]

Hardware setup:

- Tesla Powerwall 2 (battery, grid, pv)
- Tesla Model Y (vehicle)
- 2x Tinkerforge warp smart (chargers)
- evcc (Docker)
- mqtt (Docker)

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.70 (5188ce5)

```

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>

uri: 192.168.xxx.xx:7070 # uri for ui
interval: 10s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
# sponsortoken:

# log settings
[image]

log: debug
levels:
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters
meters:
- name: battery
  type: tesla
  uri: https://192.168.xxx.xx/
  usage: battery
  password: Password123
- name: grid
  type: tesla
  uri: https://192.168.xxx.xx/
  usage: grid
  password: Password123
- name: pv
  type: tesla
  uri: https://192.168.xxx.xx/
  usage: pv
  password: Password123

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/evcc-io/config#chargers
chargers:
- name: warp2-rechts        # Konfigurationsname der Wallbox
  type: warp        # Typ der Wallbox
[... 267 lines cut ...]
[lp-2  ] ERROR 2021/12/01 15:37:51 vehicle soc: vehicle not available: cannot create vehicle 'Tesla': 1 error(s) decoding:

* '' has invalid keys: password, user
[cache ] DEBUG 2021/12/01 15:37:51 charging: false
[cache ] DEBUG 2021/12/01 15:37:51 enabled: false
[lp-2  ] DEBUG 2021/12/01 15:37:51 target charging active for 0001-01-01 00:00:00 +0000 UTC: projected 2021-12-01 15:38:00 +0100 CET (0s remaining)
[cache ] DEBUG 2021/12/01 15:37:51 timerSet: false
[cache ] DEBUG 2021/12/01 15:37:51 timerActive: true
[cache ] DEBUG 2021/12/01 15:37:51 timerProjectedEnd: 2021-12-01 15:38:00 +0100 CET
[lp-2  ] DEBUG 2021/12/01 15:37:51 max charge current: 0A = 0A + -1.29A (893W @ 3p)
[lp-2  ] DEBUG 2021/12/01 15:37:51 reset pv enable timer: 1m0s
[lp-2  ] DEBUG 2021/12/01 15:37:51 pv enable timer: keep disabled
[lp-2  ] DEBUG 2021/12/01 15:37:51 pv max charge current: 0A
[lp-2  ] DEBUG 2021/12/01 15:37:52 charge power: 0W
[cache ] DEBUG 2021/12/01 15:37:52 homePower: 959
[cache ] DEBUG 2021/12/01 15:37:52 chargePower: 0

</pre></details>

**If using Docker:**
Show output of <code>docker run andig/evcc -v</code>:
```
evcc version 0.70 (5188ce5)

```

andig, 2021-12-01:
> has invalid keys: password, user

Da steht alles. Bitte Tokens verwenden.

closed 2021-12-01 by andig

Brutalok, 2021-12-01:
> > has invalid keys: password, user
> 
> Da steht alles. Bitte Tokens verwenden.

Auf meinem Account ist kein MFA aktiv, Token hatte ich schon erfolglos versucht:

Ist die Vorlage korrekt?

 tokens:
    access: # generated by running "evcc tesla-token" once
    refresh: # generated by running "evcc tesla-token" once

token wurde über die Android App generiert.