evcc-io/evcc #1976: Add PC Electric / GARO Wallbox

opensprinklershop, 2021-12-01:
**Is your feature request related to a problem? Please describe.**
Need charger support for evcc for this Modbus 485 RTU PCE/GARO Wallbox

**Describe the solution you'd like**
charger available in evcc

**Describe alternatives you've considered**
-

**Additional context**
Add any other context or screenshots about the feature request here.

The GARO Digital Multi-Meter GNM3D (RS485) can control the GARO wallbox. I found this documentation:
[108740_3_Protocol_GNM3T_RS485.PDF](https://github.com/evcc-io/evcc/files/7637631/108740_3_Protocol_GNM3T_RS485.PDF)

I can test! And I can help to develop the plugin. But...how can I start?

andig, 2021-12-02:
> Need charger support for evcc for this Modbus 485 RTU PCE/GARO Wallbox

Gibt es eine Doku zur Wallbox?

> But...how can I start?

Bei Modbus ziemlich einfach- Du kannst mit https://github.com/evcc-io/evcc/blob/master/charger/simpleevse.go anfangen. Falls der Zähler nicht über die Box API auslesbar ist sondern eine eigene Slave ID hätte wäre es sinnvoll, den bei https://github.com/volkszaehler/mbmd mit zu hinterlegen- die können wir alle einbinden.

andig, 2021-12-05:
Ping @opensprinklershop

andig, 2021-12-12:
ping @opensprinklershop

opensprinklershop, 2021-12-12:
Hi, ich habe die Wallbox da, aber wird erst nächste Woche installiert. Erst
wenn die läuft, fange ich an.

Mit freundlichen
Stefan Schmaltz
OpenSprinklerShop

opensprinklershop, 2021-12-17:
Hi, es gibt ein undokumentiertes Webinterface, das ist natürlich 1000mal besser als die serielle Modbus-Schnittstelle! Wichtig ist, Firmware >= 1.3.1 !
Habe das hier gefunden:
https://github.com/sockless-coding/garo_wallbox
https://www.domoticz.com/forum/viewtopic.php?t=26327

Folgende Funktionen konnte ich schon ermitteln:
1. STATUS: http://garo2216247:8080/servlet/rest/chargebox/status
2. SLAVE BOXEN: http://garo2216247:8080/servlet/rest/chargebox/slaves/false
3. ZÄHLER: http://garo2216247:8080/servlet/rest/chargebox/meterinfo/CENTRAL100
4. CONFIG: http://garo2216247:8080/servlet/rest/chargebox/config

To enable/disable charger I’m posting JSON-string 'ALWAYS_ON' or 'ALWAYS_OFF' to 
http://<glb-ip>:2222/rest/chargebox/mode with header {'Content-Type': 'application/json; charset=UTF-8'}

To change charge current I’m posting JSON-string {"reducedIntervalsEnabled": true,"reducedCurrentIntervals": [{"schemaId": 1,"start": "00:00:00","stop": "24:00:00","weekday": 8,"chargeLimit": "<NEW_CHARGE_CURRENT>}]} to http://<glb-ip>:2222/rest/chargebox/currentlimit with header {'Content-Type': 'application/json; charset=UTF-8'}

@andig falls du Live Daten brauchst, ich kann dir eine Verbindung zu meinen Boxen herstellen

[garo.txt](https://github.com/evcc-io/evcc/files/7738315/garo.txt)

andig, 2021-12-18:
Danke für die Doku, das sieht gut aus. Ich frage mich aber ehrlich gesagt, ob es sich lohnt die Box in evcc aufzunehmen- das scheint mir doch recht spezifisch und in DACH wenig verbreitet?

opensprinklershop, 2021-12-18:
Hi,
die Garo Boxen werden in DACH als PCE Wallboxen verkauft. PC Electric ist
ein Unternehmen aus Österreich. Die Wallboxen werden auch in vielen
Baumärkten verkauft.

Hofyyy, 2021-12-18:
Hi,

ich habe mal den Hersteller Garo angeschrieben und eine offizielle REST Dokumentation angefragt. Auch in wie weit sie die Schnittstelle kompatibel halten etc.

Mal gucken wie sie sich positionieren.

andig, 2021-12-19:
> @andig falls du Live Daten brauchst, ich kann dir eine Verbindung zu meinen Boxen herstellen

@opensprinklershop bitte per Mail an cpuidle@gmx.de

andig, 2022-01-06:
> Hi,
> 
> ich habe mal den Hersteller Garo angeschrieben und eine offizielle REST Dokumentation angefragt. Auch in wie weit sie die Schnittstelle kompatibel halten etc.
> 
> Mal gucken wie sie sich positionieren.

Kam da was?

Hofyyy, 2022-01-07:
> > Hi,
> 
> > 
> 
> > ich habe mal den Hersteller Garo angeschrieben und eine offizielle REST Dokumentation angefragt. Auch in wie weit sie die Schnittstelle kompatibel halten etc.
> 
> > 
> 
> > Mal gucken wie sie sich positionieren.
> 
> 
> 
> Kam da was?

Nein die haben nicht geantwortet. Ich kann da nächste Woche mal nachhaken. Vielleicht haben die auch noch Urlaub.

Hofyyy, 2022-01-07:
Ich hab jetzt noch mal direk an die Emailadresse von der Firma geschrieben und nicht über das Kontaktformular.

Hofyyy, 2022-01-10:
Anbei die offizielle Antwort :-/

Hi,
 
Our chargers work only with closed API. Therefore we won´t be able to provide you with any API documentation.
We recommend our chargers GLB+, TWIN+ or LS4 for that, they work with the closed API program OCPP 1.6.
 
Best Regards

Hofyyy, 2022-01-11:
ich komme bei dem Hersteller leider nicht weiter. Sorry

Letzte email:

Hi xxx,
 
We appreciate the time you took to contact us about this.
However, we do not have such documentation whether official nor unofficial and that is nothing we are planning to develop either.
 
Many thanks for your understanding and have a nice day!

PanicMan, 2022-01-15:
Hallo,
ich habe zwei von den Wallboxen von PCE/GARO in meiner Garage, ganz ehrlich, würde ich nicht wieder kaufen, aber nun sind die schon mal da. Die Steuerung der Box geht NUR wenn ein WLAN Modul (ein RasPi 3b) und ein Zähler in der Box ist und die Box auf Single-Betrieb ist. Bei verbundenen Boxen geht da gar nix und ihr braucht die nicht anzuschreiben, die werden nichts antworten, habe ich selber mehrfach versucht. Übler Sauhaufen. Die einfachste Art herauszufinden wie was funktioniert ist lokal deren Konfiguration aufzurufen und mit Wireshark zu überwachen. Bei den verlinkten sockless-coding/garo_wallbox ist im prinzip schon alles vorhanden.
Würde mich auch freuen wenn die pull-requests dafür akzeptiert würde, dann würde ich das hier auch ausprobieren.
Übrigens, ein POST mit: http://ip:8080/servlet/rest/chargebox/energy und folgendem JSON als payload: 
{chargeboxSerial: "12345678", fromDate: "2022-01-01T00:00:00.000Z", toDate: "2022-01-31T23:59:00.000Z", meterSerial: "DEFAULT", resolution: "DAY"} liefert alle Verbrauchsdaten für den jeweiligen Zeitraum, das hat @sockless-coding noch nicht.
Grüße Jack

opensprinklershop, 2022-01-15:
Hi,
ich habe gerade die GARO Firmware gepatcht, damit diese mit evcc läuft.
Du hast jetzt einen internen Zähler, ich habe einen Externen. Funktioniert
bei dir http://ip:8080/servlet/rest/chargebox/meterinfo/CENTRAL100
oder ist das
http://garo2216247:8080/servlet/rest/chargebox/meterinfo/INTERNAL   ?

Grüße
Stefan

closed 2022-01-15 by andig

PanicMan, 2022-01-15:
Hi @opensprinklershop, du hast wohl den externen, auf Adresse 100, den hatte ich vorher auch und da hatte ich keine Verbrauchsinfos und auch kene Steuerung der Ladestärke. 
Ich habe den dann aus dem Zählerschrank in die Wallbox umgebaut und auf Adresse 1 gestellt. Jetzt geht beides.
Somit bekomme ich bei CENTRAL100 (das ist für das Loadbalancing) keine Werte, nur bei CENTRAL oder INTERNAL, dann aber die Werte des internen Zählers.
Grüße Jack

DerAndereAndi, 2022-01-16:
Was müsste ein Anwender denn da Eintragen und woher bekommt er die Infos?

DerAndereAndi, 2022-01-18:
D.h. wenn man im Konfigurationsprozess nach den Informationen fragt, muss der Anwender auf welche der beiden Seite genau gehen und wo sieht er dann was? Ich muss das ja beschreiben so dass es ein Unbedarfter auch versteht. Hast du mir da einen Screenshot?

opensprinklershop, 2022-01-18:
Die evcc Parameter für die GARO Wallbox sind:
URI        string
SlaveIndex int
Meter      string

Meter ist default=CENTRAL100, kann aber auch INTERNAL oder EXTERNAL sein, je nachdem ob der Zähler eingebaut oder extern verbunden ist. Die Modbus-ID des Zählers ist entscheidend, 100=CENTRAL100, 1=INTERNAL, 2=EXTERNAL, 101=CENTRAL101
URI=<ip>:8080
SlaveIndex nur bei Slave-Boxen eintragen (1..n)

Achtung: Bei der Verwendung von Slaveboxen gibt es das Problem, dass immer alle Boxen aktiviert oder deaktiviert werden. Eine vernüftige evcc Steuerung ist damit nicht möglich!! 
Ich habe daher die Firmware des Wifi-Moduls gepatcht, damit eine Einzelsteuerung möglich ist. Der Patch ist derzeit nur auf Nachfrage bei mir erhältlich. Außerdem braucht man die neueste Version der pcelectric implementierung

DerAndereAndi, 2022-01-18:
Wir ignorieren Slaves und werden diesen Kommentar einfügen:

> Es können momentan nur als Master konfigurierte Geräte verwendet werden!

Aber nochmals, woher bekommt der Anwender die Info was in `Meter` einzutragen ist? Wie formuliert man das, damit es auch Einsteiger verstehen?

opensprinklershop, 2022-01-18:
Na irgendjemand muss ja den Zähler verbaut haben, wenn er extern ist. Dabei muss ja die Modbus-ID eingetragen worden sein.

Und nur Typ=GLB wird unterstützt!

Laut Prospekt
https://oxomi.com/service/json/catalog/pdf?portal=3000237&user=&roles=&accessToken=&catalog=10163042
hat nur der 353411P und 353409P einen internen Stromzähler.
Externer ist ja ein Extra Produkt, die Modbus ID kann hier ja nur 2, 100 oder 101 sein

PanicMan, 2022-01-21:
> D.h. wenn man im Konfigurationsprozess nach den Informationen fragt, muss der Anwender auf welche der beiden Seite genau gehen und wo sieht er dann was? Ich muss das ja beschreiben so dass es ein Unbedarfter auch versteht. Hast du mir da einen Screenshot?

Ja, hier sieht man es, das ist die Oberfläche wenn man mit einem Browser auf die Wallbox navigiert:
[image]
Man muss auf den 2. Tab gehen, dann steht ganz oben die Seriennummer der Wallbox und etwas drunter die Seriennummer des Zählers...