evcc-io/evcc #1980: Deactivate gokrazy build

andig, 2021-12-03:
@DerAndereAndi aktuell bauen wir das relativ mühsam bei jedem Release, verwenden es aber nicht. Würdest Du das evtl. deaktivieren wollen?

DerAndereAndi, 2021-12-03:
Das habe ich mir noch gar nicht angeschaut. Das sind diese Komplettimages, oder?

andig, 2021-12-03:
Genau. In build.yaml erzeugt, beim goreleaser mit hochgeladen.

DerAndereAndi, 2021-12-03:
Wir wissen halt nicht ob das verwendet wird. Aber aufgrund nicht ausführlicher Doku dazu, wohl eher niemand. Aber können wir ja für die Zukunft in der Hinterhand behalten. Auskommentieren?

andig, 2021-12-03:
Yep. Bei Bedarf wieder rein, erstmal Ressourcen sparen.

DerAndereAndi, 2021-12-03:
In dem gleichen PR wie deb?

DerAndereAndi, 2021-12-03:
Habs mal im PR gemacht: https://github.com/evcc-io/evcc/pull/1977/commits/37c0507e3d9675a7faaeac11a90129933c5cd474

closed 2021-12-03 by andig