evcc-io/evcc #1998: Mennekes Amtron Xtra/Premium wallbox support

stiajoe, 2021-12-08:
**Is your feature request related to a problem? Please describe.**
Latest firmware update of Mennekes Amtron Xtra/Premium (with HCC3 controller) enables modbus support, I can't find any application offering support for this yet. 

**Describe the solution you'd like**
Please add support for this wallbox charger. 

**Describe alternatives you've considered**
Complete modbus library description for the wallbox is found here: https://update.mennekes.de/hcc3/1.13/Description%20Modbus_AMTRON%20HCC3_v01_2021-06-25_en.pdf

Linked from this site: 
https://www.chargeupyourday.de/services/software-updates/

Also added the PDF-file here
[Description Modbus_AMTRON HCC3_v01_2021-06-25_en.pdf](https://github.com/evcc-io/evcc/files/7674436/Description.Modbus_AMTRON.HCC3_v01_2021-06-25_en.pdf)
.

andig, 2021-12-08:
Gute Idee, können wir gerne mit aufnehmen!

andig, 2021-12-08:
@stiajoe can you contact me at cpuidle@gmx.de for remote access?

andig, 2021-12-08:
Note: phase switching is not supported over Modbus. Only reading the configured phases is documented.

andig, 2021-12-31:
@stiajoe any update?

stiajoe, 2021-12-31:
I’m afraid not. Going to pick up where I left during next week. Hopefully people are back to work and on site after the Christmas vacation so they can do some testing.

Happy New Years - Stian

matspi, 2022-01-02:
There is also a variant of the Mennekes boxes called _Amtron Professional_, which uses a different set of modbus registers (cf. [PDF](https://www.chargeupyourday.com/fileadmin/07_wissen/01_anwendungsfaelle/05_kompatible_energie_management_systeme/ECU_modbus_tcp_server_spec_rev_1.04.pdf)).

I have started an implementation here: https://github.com/matspi/evcc/blob/amtron-professional/charger/amtron_prof.go

Would it make sense to incorporate these changes into this module/PR or would you rather create a separate charger implementation, i.e. should I open a new issue?

andig, 2022-01-02:
Registers look completely different- I'd suggest separate implementation.

closed 2022-01-28 by github-actions[bot]