evcc-io/evcc #2018: onIdentify läuft nicht über type:template

Chris591, 2021-12-11:
**Describe the bug**
Die Identifizierung über RFID läuft nur wenn ich über type: ford das Auto Konfiguriere.
`- type: ford `
Aber nicht über die config erzeugt vom Parameter configure:
```
- type: template
  template: ford 
```

**To Reproduce**
Steps to reproduce the behavior:
```
- type: template
  template: ford 
...
  identifiers:
    - 123456AB
  onIdentify:
    mode: "pv"
    minSoC: 20
    targetSoC: 80
    minCurrent: 6
    maxCurrent: 16
```

**Expected behavior**
Sollte genauso funktionieren wie type: ford

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.72.1639190112.1639190112 (55ecbba)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
[main  ] INFO 2021/12/11 12:02:29 evcc 0.72.1639190112.1639190112 (55ecbba)
[main  ] INFO 2021/12/11 12:02:29 using config file /etc/evcc.yaml
[mqtt  ] INFO 2021/12/11 12:02:29 connecting evcc-124083564 at 10.201.1.10:1883
config
======

grid: grid
----------
Power: 15W

pv 0: pv1
---------
Power: 3513W

Battery 0: battery1
-------------------
Power: -2667W
SoC:   27%

loadpoint 1
===========

charger: Einfahrt
-----------------
Power:          0W
Energy:         1699.4kWh
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        false
Duration:       0s
Identifier:     0d03ef2b

vehicle 0
---------
[... 336 lines cut ...]
[cache ] DEBUG 2021/12/11 12:04:57 connectedDuration: 0s
[cache ] DEBUG 2021/12/11 12:04:57 chargeCurrent: 0
[cache ] DEBUG 2021/12/11 12:04:57 connected: true
[cache ] DEBUG 2021/12/11 12:04:57 charging: false
[cache ] DEBUG 2021/12/11 12:04:57 enabled: false
[lp-2  ] WARN 2021/12/11 12:04:57 target charging: not possible
[lp-2  ] DEBUG 2021/12/11 12:04:57 charge power: 0W
[cache ] DEBUG 2021/12/11 12:04:57 chargePower: 0
[lp-2  ] DEBUG 2021/12/11 12:04:58 charge currents: [0 0 0]A
[cache ] DEBUG 2021/12/11 12:04:58 homePower: 888
[cache ] DEBUG 2021/12/11 12:04:58 chargeCurrents: [0 0 0]

</pre></details>

**If using Docker:**
Show output of <code>docker run andig/evcc -v</code>:
```
<- paste here

```

<details><summary>Show evcc log output with <code>docker logs <container id></code>:</summary><br><pre>
<- paste here

</pre></details>

DerAndereAndi, 2021-12-12:
Das kommt mit https://github.com/evcc-io/evcc/pull/2019

Chris591, 2021-12-13:
Immer noch die gleichen Problem.
evcc version 0.73.1639362857.1639362857 (cd43f09)

evcc[17625]: [lp-1  ] WARN 2021/12/13 10:20:56 target charging: not possible
evcc[17625]: [lp-2  ] WARN 2021/12/13 10:21:06 target charging: not possible

vehicle 0
---------
.SoC:     vehicle not available: cannot create vehicle 'template': template: yaml:2:2: executing "yaml" at <include "vehicle-common" .>: error calling include: template: yaml:39:12: executing "vehicle-common" at <.identifiers>: range can't iterate over '[123456AB]'
Capacity: 0kWh

vehicle 1
---------
.SoC:     vehicle not available: cannot create vehicle 'template': template: yaml:2:2: executing "yaml" at <include "vehicle-common" .>: error calling include: template: yaml:39:12: executing "vehicle-common" at <.identifiers>: range can't iterate over '[234567ABC]'
Capacity: 0kWh

DerAndereAndi, 2021-12-13:
@Chris591 wie sieht deine vehicle Konfiguration aus? Beachte bitte dass es bei den Templates kein Einrücken und `onIdentify` gibt: https://docs.evcc.io/docs/reference/configuration/vehicles#onidentify

DerAndereAndi, 2021-12-13:
ich kann die Fehlermeldung reproduzieren, hängt aber nicht an `onIdentifier` Elementen, sondern an `ìdentifiers`. Ich schau mal.
@andig hätte demnach nichts mit dem ursprünglichen Ticket hier zu tun

Chris591, 2021-12-13:
Wenn man den Fehler vom config dump richtig liest sieht man es ja auch aber hab ich auch nicht. Sorry - hast recht sind die identifiers und nicht das onIdentifier was hier nicht geht.

DerAndereAndi, 2021-12-13:
@Chris591 alles gut, hab ich am Anfang auch übersehen 😁

DerAndereAndi, 2021-12-13:
@Chris591 könntest bitte ein neues ticket aufmachen, damit wir und andere hier nicht durcheinander kommen? Bin aber schon dran.

closed 2021-12-13 by DerAndereAndi

Chris591, 2021-12-13:
> @Chris591 könntest bitte ein neues ticket aufmachen, damit wir und andere hier nicht durcheinander kommen? Bin aber schon dran.

identifiers not working with templates #2036

PR #2019 (2021-12-12): Template improvements
Fix #2018

- Update go-e firmware requirements
- Change cfos devices not to use modbus config schema
- Add support for `onIdentify` params of vehicles
- Add support for `resetOnDisconnect` for loadpoints
- Add support for defining device specific modbus port, baudrate and comset defaults
- Handle fatal error when `evcc.yaml` exists and the process doesn't have write permissions
- Some bug fixes
commit 96a04753 (2021-12-12): Template improvements (#2019)

Fix #2018

- Update go-e firmware requirements
- Change cfos devices not to use modbus config schema
- Add support for `onIdentify` params of vehicles
- Add support for `resetOnDisconnect` for loadpoints
- Add support for defining device specific modbus port, baudrate and comset defaults
- Handle fatal error when `evcc.yaml` exists and the process doesn't have write permissions
- Some bug fixes