evcc-io/evcc #2020: BMW: connect: no route to host issue

Hofyyy, 2021-12-11:
**Describe the bug**
After connection the car via the Wallbox, EVCC is not able to connect to the BMW API, see error log.
This happens randomly.

**To Reproduce**
Connect the car to the Wallbox.
I am not able to force the problem, but I saw it now multiple times.

**Expected behavior**
I think the error will have a reason, but it would be good if it could recover out of this state.
If I connect via SSH to my linux server, I am able to ping and connect the BMW API.
And if I to "sudo systemctl stop EVCC / start EVCC, it works immediately.

So I think their is a short connection problem which might be real,
but afterwards EVCC is not able to retry or restore in a correct way.
Instead of this I see the same error message every 10 sec.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
0.71
```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
<- paste here
</pre></details>

[evcc.yaml.txt](https://github.com/evcc-io/evcc/files/7696909/evcc.yaml.txt)

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>
<- paste here

</pre></details>

[syslog.txt](https://github.com/evcc-io/evcc/files/7696908/syslog.txt)

andig, 2021-12-11:
Geht bei mir. Temporäres Problem...

closed 2021-12-11 by andig

Hofyyy, 2021-12-11:
> Geht bei mir. Temporäres Problem...

Kannst du mir noch ein wenig erklären wie das funktioniert?

Wir das Objekt BMW nur einmal erstellt und der Fehler bleibt dann stehen? Oder probiert er alle 10 sek das Objekt neu zu erstellen?

Ich stimme übrigens zu das es temporär ist.  die frage ist an welcher stelle im source kann es dazu kommen und warum fängt er sich nicht?

andig, 2021-12-11:
Es fängt sich nicht weil das Fahrzeug initial nicht angelegt werden kann. Da gibts keinen retry.

Hofyyy, 2021-12-11:
Ok das erklärt es schon mal.

Was hältst du denn von folgender Idee?

ich verstehe den Sourcecode so, dass während einer Ladung alle 15min die APIs real abgefragt werden und dazwischen nur mit Cache Values gearbeitet wird.

Könnte man nicht in dem Fall probieren das Auto zu "initialisieren" wenn es davor noch nicht geklappt hat?
In dem Fall würde es sich dann "ggf." nach 15 Minuten selber reparieren.

premultiply, 2021-12-11:
`no route to host` klingt eher nach fundamentalm Routing-Problem des Systems.
Zwei Netzwerkschnittstellen aktiv (z. B. Ethernet und Wifi parallel verbunden?
Fehlerhafte manuelle IP- bzw. DNS-Konfiguration?
Kann man eher mit traceroute etc. analysieren. Hat jedenfalls nichts mit evcc zu tun.

Hofyyy, 2021-12-12:
> `no route to host` klingt eher nach fundamentalem Routing-Problem des Systems.
> 
> Zwei Netzwerkschnittstellen aktiv (z. B. Ethernet und Wifi parallel verbunden)?
> 
> Fehlerhafte manuelle IP- bzw. DNS-Konfiguration?
> 
> Kann man eher mit traceroute etc. analysieren. Hat jedenfalls nichts mit evcc zu tun.

Erstmal hat es nichts mit EVCC zu tun, allerdings wäre es doch klasse, wenn EVCc sich fangen könnte. 

Das Problem besteht nicht statisch. Als ich es untersuchen wollte ging alles tatellos und ich konnte einfach evcc neustarten und alles ging. 

ich werd mir trotzdem die Netzwerk settings mal anschauen. vielleicht liegt es auch am pihole dns cache. 

Aber wie gesagt. ja das der fehler auftritt ist infrastruktur. wie evcc damit umgeht ist evcc :-)

Hofyyy, 2021-12-12:
> Zwei Netzwerkschnittstellen aktiv (z. B. Ethernet und Wifi parallel verbunden)?
Der Pi war noch in einem Wifi mit VLAN ohne internet :-) Warum es dann zu 99% funktionierte und nur manchmal nicht ist mir schleierhaft. Scheint Zufall zu sein welche route er dann nimmt.

premultiply, 2021-12-12:
Grundlagen Netzwerktechnik ;)
Die ersten Schritte legt die lokale Routingtabelle des Systems fest.