evcc-io/evcc #2036: identifiers not working with templates

Chris591, 2021-12-13:
**Describe the bug**
Autos können nicht identifiziert werden mit identifiers wenn über templetes konfiguriert

**To Reproduce**
Steps to reproduce the behavior:
vehicles:
- type: template
  template: ford
...
  identifiers:
    - 123456AB

**Expected behavior**
Sollte genauso funktionieren wie mit type: ford

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.73.1639362857.1639362857 (cd43f09)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
[main  ] INFO 2021/12/13 12:35:05 evcc 0.73.1639362857.1639362857 (cd43f09)
[main  ] INFO 2021/12/13 12:35:05 using config file /etc/evcc.yaml
[mqtt  ] INFO 2021/12/13 12:35:05 connecting evcc-1225036323 at tcp://10.201.1.10:1883
config
======

grid: grid
----------
Power: -51W

pv 0: pv1
---------
Power: 830W

Battery 0: battery1
-------------------
Power: -0W
SoC:   3%

loadpoint 1
===========

charger: Einfahrt
-----------------
Power:          0W
Energy:         1699.4kWh
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        false
Duration:       34s
Identifier:     123456AB

vehicle 0
---------
.SoC:     vehicle not available: cannot create vehicle 'template': template: yaml:2:2: executing "yaml" at <include "vehicle-common" .>: error calling include: template: yaml:39:12: executing "vehicle-common" at <.identifiers>: range can't iterate over '[123456AB]'
Capacity: 0kWh

vehicle 1
---------
.SoC:     vehicle not available: cannot create vehicle 'template': template: yaml:2:2: executing "yaml" at <include "vehicle-common" .>: error calling include: template: yaml:39:12: executing "vehicle-common" at <.identifiers>: range can't iterate over '[234567AB]'
Capacity: 0kWh

loadpoint 2
===========

charger: Garage
---------------
[... 257 lines cut ...]
[cache ] DEBUG 2021/12/13 12:41:46 connectedDuration: 0s
[cache ] DEBUG 2021/12/13 12:41:46 chargeCurrent: 0
[cache ] DEBUG 2021/12/13 12:41:46 connected: true
[cache ] DEBUG 2021/12/13 12:41:46 charging: false
[cache ] DEBUG 2021/12/13 12:41:46 enabled: false
[lp-2  ] WARN 2021/12/13 12:41:46 target charging: not possible
[lp-2  ] DEBUG 2021/12/13 12:41:47 charge power: 0W
[cache ] DEBUG 2021/12/13 12:41:47 chargePower: 0
[lp-2  ] DEBUG 2021/12/13 12:41:47 charge currents: [0 0 0]A
[cache ] DEBUG 2021/12/13 12:41:47 homePower: 1519
[cache ] DEBUG 2021/12/13 12:41:47 chargeCurrents: [0 0 0]

</pre></details>

**If using Docker:**
Show output of <code>docker run andig/evcc -v</code>:
```
<- paste here

```

<details><summary>Show evcc log output with <code>docker logs <container id></code>:</summary><br><pre>
<- paste here

</pre></details>

DerAndereAndi, 2021-12-13:
Danke nochmal für den Hinweis! Der fix ist morgen im nightly build drin.

closed 2021-12-13 by DerAndereAndi

Chris591, 2021-12-13:
Ihr seit einfach Klasse vielen Danke das ihr euch so schnell auf alle Probleme stürzt und sie löst egal ob diese vor dem Bildschirm sind oder doch manchmal auf in der Software!!  - Hoffe jetzt kommt auch mal die Sonne raus das ich es auch mal wirklich nutzen kann.

Chris591, 2021-12-14:
Wollte nur bescheid geben es läuft jetzt so wie erwartet!! Danke

PR #2037 (2021-12-13): Fix identifiers in templates not working
Fixes #2036
commit 0c4e338c (2021-12-13): Fix identifiers in templates not working (#2037)

Fixes #2036