evcc-io/evcc #2054: mA current steps for EVSE DIN

premultiply, 2021-12-16:
### Discussed in https://github.com/evcc-io/evcc/discussions/2053

<div type='discussions-op-text'>

<sup>Originally posted by **RTTTC** December 16, 2021</sup>
So I have asked EvRacing.cz if smaller than 1A steps are planned for the future, and got this response:

> Yes you can actually change this step to 0.01A. You will need to enable bit 7 from register 2005. Then you will be able to change current in register 1000 in smaller steps (16A -->1600)

So great news for those that have firmware revision 17 and later!   http://evracing.cz/evse/evse-wallbox/evse-wb-din_latest.pdf
I have changed bit7 using "Modbus pool" and charge controller can now be controlled in small steps using direct modbus commands :)
Will EVCC be able to detect and handle this change automatically? (don't have a car to test..)</div>

schenlap, 2022-01-17:
For evse wifi (smartwb), which uses evse din internally, high resolution mode is already in evcc.
The question is for https://docs.evcc.io/docs/devices/chargers#simple-evse ?

andig, 2022-01-17:
Ja, so was das gemeint

premultiply, 2022-01-17:
Neben der Premium-Variante mit Fallunterscheidung nach Firmware-Version usw. könnte man die einfache Version machen und die mA-Regelung einfach fest umsetzen.

Nach dem Motto: Wer die EVSE-DIN hat wird wohl auch in der Lage sein die Firmware zu aktualisieren und die entsprechenden Registereinstellungen vorzunehmen.

Das wäre mit minimalem Aufwand umzusetzen.

andig, 2022-01-17:
Stimmt. Aber damit machen wir es für alle die das nicht können kaputt. Wenn niemand Kapa dafür hat würde ich hier zu machen, es läuft ja nicht weg…

RTTTC, 2022-01-17:
> For evse wifi (smartwb), which uses evse din internally, high resolution mode is already in evcc.
> The question is for https://docs.evcc.io/docs/devices/chargers#simple-evse ?

BTW controller in this link is named incorrectly, (Simple EVSE has no RS485)
Appropriate name is EVSE DIN
http://evracing.cz/simple-evse-wallbox

premultiply, 2022-01-17:
@RTTTC Looks like you are one of the very few users using this controller under direct control of evcc.
Wouldn't you like to try an appropriate PR for this controller to close this issue?

andig, 2022-01-17:
Closing until someone feels like picking this up.

RTTTC, 2022-01-18:
> @RTTTC Looks like you are one of the very few users using this controller under direct control of evcc.
> Wouldn't you like to try an appropriate PR for this controller to close this issue?

I don't have the necessary skills.

RTTTC, 2022-01-18:
> Stimmt. Aber damit machen wir es für alle die das nicht können kaputt. Wenn niemand Kapa dafür hat würde ich hier zu machen, es läuft ja nicht weg…

What if there are simply two templates : `EVSE DIN`  and  `EVSE DIN mA` ?

andig, 2022-01-18:
We‘re not going to change it right now.

RTTTC, 2022-05-01:
> @RTTTC Looks like you are one of the very few users using this controller under direct control of evcc.
> Wouldn't you like to try an appropriate PR for this controller to close this issue?
> 

Am I _the only one_ ? :DD
Could someone point me in the right direction - where would I look into, what files are needed to be changed / added to create PR for this?

benesolar, 2025-06-27:
Danke für die Umsetzung.
Sieht man ob der mA Modus aktiv ist?

Ich kenne die Firmwareversion meiner EVSE nicht und weiß auch nicht ob es das Auto ebenfalls unterstützt.

premultiply, 2025-06-28:
Derzeit nur im Tracelog durch Interpretation der Modbus-Kommunikation.

Wäre aber kein Problem das als Diagnoseausgabe zu ergänzen.

PR welcome.

benesolar, 2025-06-29:
Funktioniert aufgrund der Ladeleistung offensichtlich bei mir.
Programmieren kann ich leider nicht.