evcc-io/evcc #2056: PV indicator signals charge start when no pv production

andig, 2021-12-17:
[image]

/cc @naltatis probably backend issue, I will take a look

rivengh, 2021-12-17:
Das ist mir auch aufgefallen. Der Timer scheint tatsächlich auch zu laufen. Hatte aber leider noch keine Zeit tiefer reinzuschauen.

andig, 2021-12-17:
Hatte heute morgen einen Fall mit Logfile, muss ich aber noch analysieren…

andig, 2021-12-17:
Da klemmt definitiv die Logik:

    [lp-1  ] DEBUG 2021/12/17 07:59:50 max charge current: 0A = 0A + -12.6A (2888W @ 1p)
    [lp-1  ] DEBUG 2021/12/17 07:59:50 reset pv enable timer: 30s
    [lp-1  ] DEBUG 2021/12/17 07:59:50 pv action: enable in 29.999999235s
    [lp-1  ] DEBUG 2021/12/17 07:59:50 pv enable timer: keep disabled
    [lp-1  ] DEBUG 2021/12/17 07:59:50 pv max charge current: 0A

closed 2021-12-17 by andig