evcc-io/evcc #2064: Zielladen: Timer kann über GUI nicht gelöscht werden

schenlap, 2021-12-18:
**Describe the bug**
Wenn man den Timer löscht wird er intern auf 1.1.1970 gesetzt. Da dies vor der aktuellen Zeit ist beginnt damit das Laden sofort bis zum ZielSoC

**To Reproduce**
Steps to reproduce the behavior:
1. Setzt Timer und Soc in der GUI
2. Klicke: Target Time remove
3. Auto beginnt sofort zu lasen

**Expected behavior**
Auto lädt nicht mehr nach Timer

schenlap, 2021-12-18:
Ich kann natürlich einen Workaround im timer einbauen. Besser wär es aber wenn es die GUI richtig macht und reset aufgerufen wird. Ich weiß aber nicht wo/was durch die GUI aufgerufen wird.

schenlap, 2021-12-18:
Einen Workaround hätte ich vorbereitet, https://github.com/schenlap/evcc/commit/45d8d07775106943a429d40dfd8eef233e037a62

Damit wird aber der Wert in der GUI dennoch nicht gelöst. Es sollte die GUI auch Reset selbst auslösen (oder time auf Zero setzen) im timer, dann würde es diesen Workaround nicht brauchen.

DerAndereAndi, 2021-12-18:
@naltatis Könntest du bitte hier mal schauen?

andig, 2021-12-18:
Mache ich

andig, 2021-12-18:
> @naltatis Könntest du bitte hier mal schauen?

Siehe https://github.com/evcc-io/evcc/pull/2066

closed 2021-12-18 by andig