evcc-io/evcc #2071: Targetcharging - not using all three phases

bacanol, 2021-12-19:
**Describe the bug**
When using target charging, not all possible phases are used. The system only uses one phase on a charger that is capable of switching phases from 1 to 3:

[image]

The current implementation is working precisely with one phase as far as I tried it.

**To Reproduce**
Steps to reproduce the behavior:
1. Set target charging active
2. Wait for target charging to start
3. See the screen above.

**Expected behavior**
Expected would be that target charging initiates the charge to determine the available phases and use them all. If needed it could also stop again if at the calculated start time for one phase charging, a three phase possibility is found and re-starts when the calculated time for three phase is reached. 

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.73 (52852275)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
```
config
======

grid: grid
----------
Power: 4175W

pv: pv
------
Power: 151W

loadpoint 1
===========

charger: wallboxlinks
---------------------
Power:          3700W
Energy:         312.1kWh
Current L1..L3: 15.9A 0A 0A
Charge status:  C
Enabled:        true
Charged:        4.2kWh

vehicle 0
---------
SoC:           58%
Charge status: C
Charged:       3.8kWh
Range:         247km
Odometer:      1976km
Finish time:   2021-12-19 15:44:00 +0100 CET
Position:      53.303505,10.569959
Capacity:      78kWh
Identifiers:   [04347E7A 0434-7E7A]
OnIdentified:  TargetSoC:80, Mode:pv

loadpoint 2
===========

charger: wallboxrechts
----------------------
Power:          0W
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        false
Charged:        2.0kWh

vehicle 0
---------
SoC:            65%
Charge status:  B
Range:          199km
Odometer:       5370km
Finish time:    2021-12-19 03:41:00 +0000 UTC
Climate active: false
Outside temp:   0.0°C
Target temp:    20.0°C
Capacity:       52kWh
Identifiers:    [04B5FF7A 04B5-FF7A]
OnIdentified:   TargetSoC:80, Mode:pv
```

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
*please ask if really needed*

</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>
```
Dez 19 10:47:17 evcc evcc[1460149]: [lp-1  ] DEBUG 2021/12/19 10:47:17 charge power: 3730W
Dez 19 10:47:17 evcc evcc[1460149]: [lp-1  ] DEBUG 2021/12/19 10:47:17 charge currents: [16.2 0 0]A
Dez 19 10:47:17 evcc evcc[1460149]: [lp-1  ] DEBUG 2021/12/19 10:47:17 detected phases: 1p [16.2 0 0]A
Dez 19 10:47:17 evcc evcc[1460149]: [lp-1  ] DEBUG 2021/12/19 10:47:17 charger status: C
Dez 19 10:47:17 evcc evcc[1460149]: [lp-1  ] DEBUG 2021/12/19 10:47:17 next soc poll remaining time: 59m40s
Dez 19 10:47:17 evcc evcc[1460149]: [lp-1  ] DEBUG 2021/12/19 10:47:17 soc estimated: 57.81% (vehicle: 57.00%)
Dez 19 10:47:17 evcc evcc[1460149]: [lp-1  ] DEBUG 2021/12/19 10:47:17 vehicle soc: 57%
Dez 19 10:47:17 evcc evcc[1460149]: [lp-1  ] DEBUG 2021/12/19 10:47:17 vehicle range: 244km
Dez 19 10:47:17 evcc evcc[1460149]: [lp-1  ] DEBUG 2021/12/19 10:47:17 vehicle odometer: 1976km
Dez 19 10:47:17 evcc evcc[1460149]: [lp-1  ] DEBUG 2021/12/19 10:47:17 estimated charge duration: 1h42m0s to 65% at 3680W
Dez 19 10:47:17 evcc evcc[1460149]: [lp-1  ] DEBUG 2021/12/19 10:47:17 projected end: 2021-12-19 12:29:00 +0100 CET
Dez 19 10:47:17 evcc evcc[1460149]: [lp-1  ] DEBUG 2021/12/19 10:47:17 desired finish time: 2021-12-19 12:00:00 +0100 CET
Dez 19 10:47:17 evcc evcc[1460149]: [lp-1  ] DEBUG 2021/12/19 10:47:17 target charging: speedup (16A)
Dez 19 10:47:17 evcc evcc[1460149]: [lp-1  ] DEBUG 2021/12/19 10:47:17 charge power: 3730W
Dez 19 10:47:17 evcc evcc[1460149]: [lp-1  ] DEBUG 2021/12/19 10:47:17 charge currents: [16.2 0 0]A
Dez 19 10:47:17 evcc evcc[1460149]: [lp-1  ] DEBUG 2021/12/19 10:47:17 detected phases: 1p [16.2 0 0]A
```

andig, 2021-12-19:
Probier bitte mal aktuellen Master. Es gab noch ein Problem mit dem hochschalten auf 3p, sollte seit 2 Tagen behoben sein. Falls nein bitte volles Log ab Start.

bacanol, 2021-12-19:
Moin @andig,

ist der aktuelle Master. Log ist gerade in Arbeit. Bitte wieder aufmachen.

andig, 2021-12-19:
Mist :(

andig, 2021-12-19:
Fixed in https://github.com/evcc-io/evcc/commit/2a320d1379a382acc69d9c5c508f40a8a07f040c

closed 2021-12-19 by andig