evcc-io/evcc #2085: Current master does not start charging despite pv has enough power

bacanol, 2021-12-20:
**Describe the bug**
The current master (today 12:00) does not start charging in pv mode with enough power available:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to portal
2. Wait for sun
3. Even with enough power available the system does not start loading

**Expected behavior**
Start loading if enough power is available (like with the older 0.6x versions)

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.74 (c641f267)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>

```
config
======

grid: grid
----------
Power: -1783W

pv: pv
------
Power: 2360W

loadpoint 1
===========

charger: wallboxlinks
---------------------
Power:          0W
Energy:         319.8kWh
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        true
Charged:        0.2kWh

vehicle 0
---------
.SoC:          408 Request Timeout
Charge status: 408 Request Timeout
Charged:       408 Request Timeout
Range:         408 Request Timeout
Odometer:      408 Request Timeout
Finish time:   408 Request Timeout
Position:      408 Request Timeout
Capacity:      78kWh
Identifiers:   [04347E7A 0434-7E7A]
OnIdentified:  TargetSoC:80, Mode:pv

loadpoint 2
===========

charger: wallboxrechts
----------------------
Power:          0W
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        true
Charged:        0.0kWh

[... 205 lines cut ...]
  #   recipients:
  #   - # list of recipient ids
  #- type: telegram
  #  token: <redacted> # bot id
  #  chats:
  #  - 574871920
    - type: email
      uri: <redacted>

# influx database
influx:
  url: http://localhost:8086
  database: evcc
  # user:
  # password:

```

</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>

[evcc-1.log.zip](https://github.com/evcc-io/evcc/files/7744894/evcc-1.log.zip)

</pre></details>

andig, 2021-12-20:
Das sieht komisch aus:

    Dez 20 10:43:00 evcc evcc[1602798]: [lp-1  ] DEBUG 2021/12/20 10:43:00 charger status: B
    Dez 20 10:43:00 evcc evcc[1602798]: [lp-1  ] DEBUG 2021/12/20 10:43:00 next soc poll remaining time: 20m59s
    Dez 20 10:43:00 evcc evcc[1602798]: [lp-1  ] DEBUG 2021/12/20 10:43:00 max charge current: 9.76A = 0A + 9.76A (-2245W @ 1p)
    Dez 20 10:43:00 evcc evcc[1602798]: [lp-1  ] DEBUG 2021/12/20 10:43:00 site power -2245W < enable threshold 0W
    Dez 20 10:43:00 evcc evcc[1602798]: [lp-1  ] DEBUG 2021/12/20 10:43:00 pv action: enable in 39.897925917s
    Dez 20 10:43:00 evcc evcc[1602798]: [lp-1  ] DEBUG 2021/12/20 10:43:00 pv enable timer remaining: 40s
    Dez 20 10:43:00 evcc evcc[1602798]: [lp-1  ] DEBUG 2021/12/20 10:43:00 pv enable timer: keep disabled
    Dez 20 10:43:00 evcc evcc[1602798]: [lp-1  ] DEBUG 2021/12/20 10:43:00 pv max charge current: 0A
    Dez 20 10:43:00 evcc evcc[1602798]: [lp-1  ] DEBUG 2021/12/20 10:43:00 charge power: 0W
    Dez 20 10:43:00 evcc evcc[1602798]: [lp-1  ] DEBUG 2021/12/20 10:43:00 charge currents: [0 0 0]A
    Dez 20 10:43:09 evcc evcc[1602798]: [lp-2  ] DEBUG 2021/12/20 10:43:09 charge power: 0W
    Dez 20 10:43:09 evcc evcc[1602798]: [lp-2  ] DEBUG 2021/12/20 10:43:09 charge currents: [0 0 0]A
    Dez 20 10:43:10 evcc evcc[1602798]: [lp-2  ] DEBUG 2021/12/20 10:43:10 charger status: B
    Dez 20 10:43:10 evcc evcc[1602798]: [lp-2  ] INFO 2021/12/20 10:43:10 stop charging <-

Eigentlich müsste es da los gehen. Ich schaue.

closed 2021-12-20 by andig

bacanol, 2021-12-20:
Hallo @andig, das mit dem Auto kann ich ausschließen, war vorher in gleicher Konstellation nicht so. Ich werde es Dir wahrscheinlich erst morgen mitloggen können, da heute hier keine Sonne mehr zu erwarten ist.

bacanol, 2021-12-20:
@andig , eine Zwischenvermutung: 
Es scheint einen Zusammenhang zu den go-e chartern zu geben. Habe einen V2 und einen V3 und beide starten nicht mehr über EVCC nachdem sie einmal an waren. Wenn ich dann ohne EVCC zu beenden eine Box neu starte, fängt er anschließend sofort an, mit dem was man möchte.

andig, 2021-12-20:
Merkwürdig. Aber dass is eine tiefe Abneigung gegen die Qualität der Go-E SW habe kann ich nicht verhehlen...

bacanol, 2021-12-20:
> Merkwürdig. Aber dass is eine tiefe Abneigung gegen die Qualität der Go-E SW habe kann ich nicht verhehlen...
> Nur irgendwas müsste sich ja trotzdem geändert haben wenn es vorher ging???

Das hattest Du schonmal geschrieben. Was könnte ich neben dem Beobachten morgen (wenn die Sonne scheint) machen? Es lief ja mit der 0.6 eigentlich problemfrei. Oder könnte es am Lastmanagement liegen, welches die Boxen machen untereinander? Das hatte ich an der Front zumindest zu letzt geändert (im Nov). Könnte das ein Ansatz sein?

andig, 2021-12-20:
> Könnte das ein Ansatz sein?

Ehrlich keine Ahnung. Dann hätte der Fehler vorher kommen müssten. Du kannst natürlich schauen ob eine ältere Version wieder tut, so als A/B Vergleich.

bacanol, 2021-12-20:
@andig , glaub es oder nicht. Ohne das Lastmanagement der go-e aktiviert, geht es sofort, wie früher. ;-) Da ich vom Anschluss her aktuell nicht mehr als 20A nutzen darf, wäre mein Interesse am "einfachen" Lastmanagement (Limit der Gesamtleistung auf 20A für mehrere Loadpoints sehr interessiert. Dann braucht man sich nicht auf die "Qualität der Go-E SW" verlassen.

bacanol, 2021-12-20:
Workaround für alle betroffenen:
Im Lastmanagement der Go-E unterschiedliche Prioritäten vergeben, dann scheint es zu laufen. Gleiche Prioritäten führen zu dem oben beschriebenen Problem.

premultiply, 2021-12-20:
Ich höre noch Andi sagen: go-E Software... :D