evcc-io/evcc #2125: Better sponsoring options

KlausHans, 2021-12-27:
**Is your feature request related to a problem? Please describe.**
I've installed evcc for my parents. They use a go.e with pv and the 1p/3p switching is important. 
Unfortunately i learned, that this feature is behind a paywall. Not only that, it is a Github only subscription via a token, that must be renewed every year (if i understand correctly). I don't want to go every year in the config and change the token. Also it is most likely that i simply would forget it. 
And i don't really want to give github my payment information.

**Describe the solution you'd like**
It would be very kind if you could offer other options. A one time lifetime purchase, for eg 30€ or so, would be my favourite. 
Of course freeing the phase switching would be an even better alternative.

**Describe alternatives you've considered**

**Additional context**
Phase switching is in my opinion a crucial feature for better pv usage. This is kind of important for the Energiewende. ;)

Caibuk, 2021-12-27:
Kostenlos ist immer irgendwie "besser".
Nur, wer bezahlt den Aufwand mit zickigen Wallbox-APIs dann, wenn die jeweiligen Hersteller sich nicht beteiligen wollen?!
SCNR.

RTTTC, 2021-12-27:
I don't really need the token, would still like to donate to you guys, to show my support and appreciation. This is an amazing, and fast evolving project! Preferably with a simple bank transfer, current sponsorship option is not acceptable for me.

KlausHans, 2021-12-27:
War klar dass so ein unnötiger Kommentar kommt...