evcc-io/evcc #2133: Smappe

josbmw, 2021-12-28:
Would it be possible to connect evcc to a Smappee energy monitor?
Specific info over the measured and produced electricity is available through an API.
Details here: https://smappee.atlassian.net/wiki/spaces/DEVAPI/pages/526581813/Get+Electricity+Consumption
Unfortunately I'm not a programmer, so I have no idea how to implement this.

goebelmeier, 2021-12-28:
Most detailed data is already aggregated over 5 minutes. This is much too inaccurate for evcc. The product itself looks nice, but as long as it doesn't return more granular data, I don't think this will work satisfactorily with evcc.

josbmw, 2021-12-28:
Smappee has it's own wall charger what works in combination with the energy monitor, but the wall charger is (to) expensive.
I also have a production meter on my solar installation, a Carlo Gavazzi EM23-DIN, http://www.productselection.net/PDF/UK/em23dinds.pdf

Can this one be used?

premultiply, 2021-12-28:
If it has an unused Modbus RTU port it should be usable.