evcc-io/evcc #2142: Daheimladen: charging session aborts

andig, 2021-12-29:
### Discussed in https://github.com/evcc-io/evcc/discussions/2139

<div type='discussions-op-text'>

<sup>Originally posted by **yeehman** December 29, 2021</sup>
Hallo zusammen,
ich bin ganz neu in der YAML Welt und habe Probleme das mein Ladevorgang abgebrochen wird.
Ausgangsituation:

Kostal Plenticore
Kostal Smart Energie Meter
BYD Speicher
Wallbox: Daheimladen.de
Auto: Fiat 500E
EVCC läuft auf einem Raspberry Pi 4

Die Config habe ich hinbekommen und das EVCC Webportal startet auch ich sehe Hausverbrauch, PV Energie, Ladezustand Akku, etc.

Wenn ich jetzt meinen Fiat 500E an die Wallbox Stecke wird Ladezustand , Reichweite , etc. erkannt und er startet den Ladevorgang. Für ca. 1 Minute läuft alles gut. Dann bricht er den Ladevorgang ab. Mit der Info: 
Charger out of sync: expected diasbled, got enabled. 
gefolgt von: Charger enable: charging station refused to start transaction
gefolgt von: Charger: invalid status: Finishing.

Screenshot ist angehängt. Denke das irgendwas an meiner EVCC.YAML noch nicht passt aber VS zeigt keinen Fehler an.

[image]

Anbei auch meine YAML Datei. Könnt Ihr dort einen Fehler/Problem erkennen? bzw. kann ich das irgendwie "loggen" das ich vielleicht selbst auf das Problem komme?

Was mir noch aufgefallen ist das in der Übersicht bei Ladepunkt keine Leistung der Wallbox gezeigt wird. Der eigentliche Hausverbrauch geht aber die 11kW zusätzlich nach oben. Vielleicht liegt da ja auch der Fehler?

Vielen Dank für die Hilfe:

YAML Datei

```yaml
uri: 192.168.0.76:7070 # uri for ui
interval: 10s # control cycle interval

log: info

meters:
- name: grid
  type: modbus
  model: sunspec
  uri: 192.168.0.210:502
  id: 71 # kostal default sunspec modbus id

- name: pv
  type: custom
  power:
    source: calc
    add:
    - source: modbus # use ModBus plugin
      uri: 192.168.0.220:1502 # Plenticore IP
      id: 71
      register: # manual non-sunspec register configuration
        address: 260 # String 1 DC Power
        type: holding
        decode: float32s # may be float32 on specific firmware/devices
    - source: modbus # use ModBus plugin
      uri: 192.168.0.220:1502 # Plenticore IP
      id: 71
      register: # manual non-sunspec register configuration
        address: 270 # String 2 DC Power
        type: holding
        decode: float32s # may be float32 on specific firmware/devices

- name: battery
  type: modbus
  model: sunspec
  uri: 192.168.0.220:1502 # Plenticore IP
  id: 71 # kostal default sunspec modbus id
  power: 802:W # sunspec model 802 battery
  soc: 802:SoC # sunspec model 802 battery

chargers:
- type: template
  template: daheimladen
  token: TOKEN # Sie bekommen dieses Access Token vom Daheimladen Support: info@daheimladen.de
  stationid: 84702913276859  # Die Software Seriennummer, welche auf dem Bildschirm der Wallbox angezeigt wird
  name: wallbox5

vehicles:
- type: template
  template: fiat
  title: Fiat 500E 
  user: USER
  password: PASSWORD
  vin: VIN
  capacity: 42 
  name: ev4
  pin: PIN

loadpoints:
- title: Garage
  charger: wallbox5
  vehicles:
  - ev4
  mode: now
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  soc:
    poll:
      mode: charging
      interval: 60m
    min: 0
    target: 100
    estimate: false

site:
  title: Zu Hause
  meters:
    grid: grid
    pvs:
    - pv
    batteries:
    - battery
  prioritySoC: 50 # Hausbatterie bekommt bis zum SoC 50% Priorität beim laden
  bufferSoC: 80 # Hausbatterie wird oberhalb SoC 80% als Puffer genutzt

```</div>

andig, 2021-12-29:
@utsavanand2 could you take a look at the error messages?

yeehman, 2021-12-29:
Hallo Andi,
wie mache ich Das Logfile? Muss ich da in der evcc.yaml etwas anpassen?
Danke für die Mühe.

andig, 2021-12-29:
> Hallo Andi,
> wie mache ich Das Logfile? Muss ich da in der evcc.yaml etwas anpassen?
> Danke für die Mühe.

Entweder `log: trace` oder `--log trace`.

yeehman, 2021-12-30:
Moin,

 

hoffe das ist die richtige Logdatei... Da sind noch ein paar Errors drin da ich nochmals versucht habe an der YAML rumzubasteln. Hat aber alles nichts geholfen. Habe heute 30.12 erneut versucht zu laden mit gleichem Ergebnis wie gestern. Das ist auch im Logfile sichtbar

 

Vielen Dank

yeehman, 2021-12-30:
[evcc.log](https://github.com/evcc-io/evcc/files/7792049/evcc.log)

andig, 2021-12-30:
@utsavanand2 the problem is that `ChargePointStatus=FINISHING` is not mapped in the `Status()` function. IMHO it should be either `B` or `C`. Could you please confirm which?

utsavanand2, 2021-12-30:
> @utsavanand2 the problem is that `ChargePointStatus=FINISHING` is not mapped in the `Status()` function. IMHO it should be either `B` or `C`. Could you please confirm which?
> 
> The logfile also has quite a number of
> 
> ```
> ERROR 2021/12/24 12:15:34 charger: Get "https://api.daheimladen.com/v1/cs/xxx/status": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
> ERROR 2021/12/24 12:15:44 charger: Get "https://api.daheimladen.com/v1/cs/xxx/status": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
> ERROR 2021/12/24 12:15:55 charger: Get "https://api.daheimladen.com/v1/cs/xxx/status": context deadline exceeded 
> ```
> 
> were there any problems with the backend API or do we need a longer timeout?

Hi @andig. Yes you're correct, the FINISHING state is not mapped in the `Status` function, the problem is that the charging stations do not start new charging sessions when they are stuck in Finishing state for long until the connector is reconnected. So your addition at #2156 returning Status `B` for `Finishing` makes a lot of sense but I would first like to test it out before merging it.

As for the context deadline exceeded errors, its interesting to see only calls to `/status` that are timing out the most, even when the server doesn't have any internal errors (It is the same behavior we were getting when we set a timeout on the client), as to the best of my knowledge and looking up the code, I do not see any relevant changes to the code itself on the backend that might be affecting this, currently waiting up for Stephan to run these tests with him and try to reproduce this issue.

yeehman, 2021-12-30:
Hey Utsav,

this is Thorsten... you provided the Token for me in the past. If you want and if it makes sense I can test something.

Let me know I`m ready.

Thanks

andig, 2021-12-30:
@yeehman Du könntest einfach den PR auschecken und schauen ob es damit funktioniert?

yeehman, 2021-12-30:
@andig sorry alles noch ein wenig Neuland für mich... Was ist PR? und wie checke ich den aus?

Vielen Dank

utsavanand2, 2021-12-30:
@yeehman Don't worry I'll create an executable for you. Andreas just created another branch with the necessary changes, he meant if you could clone, checkout, build from source and then try running it.
Give me a few minutes and I'll paste a link to an arm executable

utsavanand2, 2021-12-30:
@yeehman Can you try running this binary on your raspberry pi and see if this works?
https://storage.googleapis.com/evcc-test/evcc

utsavanand2, 2021-12-30:
@yeehman This binary includes the fix that @andig made, with me trying a bigger timeout value to see if it plays well with the server.
https://storage.googleapis.com/evcc-test/evcc-timeout-fix

yeehman, 2021-12-30:
Hello Utsav,

 

sorry all is really new for me… 

 

Can you explain that a little bit more… “could clone, checkout, build from source and then try running it”

 

Thanks and Regards, Thorsten

 

https://storage.googleapis.com/evcc-test/evcc

utsavanand2, 2021-12-30:
@yeehman Don't worry, you don't have to do that now that I have compiled the binaries for you that you can download from here
https://storage.googleapis.com/evcc-test/evcc

Just download it on your Raspberry Pi with
```sh
# To download evcc to your current directory
curl -L https://storage.googleapis.com/evcc-test/evcc -o evcc

# Give executable permission to your downloaded evcc binary
sudo chmod +x ./evcc

# Running evcc with the path to your yaml configuration specified
./evcc -c /path/to/your/evcc.yaml
```

yeehman, 2021-12-31:
Good Morning,
tried it with the new  binaries. 
This one is working: curl -L https://storage.googleapis.com/evcc-test/evcc -o evcc
This one is also working: sudo chmod +x ./evcc

But the last one: ./evcc -c /path/to/your/evcc.yaml --> changed it to ./evcc -c /etc/evcc.yaml is not working.
I restarted the EVCC then with --> sudo systemctl restart evcc and enable the function again with -->sudo systemctl enable evcc

Checked the status of EVCC and this looks fine.

I connected the Fiat 500E but same situation. Charging process stopped 30 seconds after it was connected with the same failure message:

New log file is here:
[evcc.log](https://github.com/evcc-io/evcc/files/7795647/evcc.log)

Not sure if this is important. If I change in the webview the mode setting (OFF, NOW, minPV and PV) you can not see any difference. Also if I click on OFF the Wallbox starts loading directly when the plug is connected to the car. Same for minPV, PV and Now.

Thanks
Thorsten

utsavanand2, 2021-12-31:
Hi @yeehman :)
It looks like you're still using the previously installed evcc binary not the new one with `sudo systemctl restart evcc`
What error did you see when running `./evcc -c /etc/evcc.yaml`?

yeehman, 2021-12-31:
Hey Utsav, 
this is the error message:
pi@raspberrypi:~ $ ./evcc -c /etc/evcc.yaml
-bash: ./evcc: Kann die Binärdatei nicht ausführen: Fehler im Format der Programmdatei

English --> Unable to run the binary: Error in the format of the program file
Thanks

utsavanand2, 2021-12-31:
Hi @yeehman 
Okay got it, this is because the compiled binary is not compatible with the architecture of the OS on the Pi.
Can you please run
`uname -a`
And tell me what does the output say?
Also which Raspberry Pi you're running?

yeehman, 2021-12-31:
Outcome of uname -a

 

Linux raspberrypi 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux

 

And that is the Raspberry Pi

 

Raspberry Pi 4 Model B 4 GB Super Kit mit 64GB Micro SD-Karte, originaler Raspberry Pi 4 Cortex A72 1.5GHz unterstützt Dual Display 4K/1000Mbps/Bluetooth 5.0

premultiply, 2021-12-31:
armv6 bzw. armhf

yeehman, 2022-01-02:
Hey Utsav,
I run again this binary file and checked afterwards the status of the system. Status looks good but in the trace/debug I have a  error to remove the evcc in the temp folder. The EVCC interface is not possible to open (ERR_CONNECTION_REFUSED). Not sure if this helps or if you need something more.
Thanks
Thorsten

{"charging_station_id":"84702913276859","status":"Available","active_power_import":0,"total_meter_value":34900,"voltage_phase_l1l2":228.3,"voltage_phase_l2l3":229,"voltage_phase_l3l1":230.8,"current_import_phase_l1":0,"current_import_phase_l2":0,"current_import_phase_l3":0,"charging_station_temperature":28,"current_time":"2022-01-02T09:29:09+01:00","transaction_start_at":"0","transaction_power_used":0}
[site  ] INFO 2022/01/02 09:29:09 site config:
[site  ] INFO 2022/01/02 09:29:09   meters:    grid ✓ pv ✓ battery ✓
[cache ] DEBUG 2022/01/02 09:29:09 siteTitle: Zu Hause
[cache ] DEBUG 2022/01/02 09:29:09 gridConfigured: true
[site  ] INFO 2022/01/02 09:29:09     grid:    power ✓ energy ✗ currents ✗
[site  ] INFO 2022/01/02 09:29:09     pv 0:    power ✓ energy ✗ currents ✗
[site  ] INFO 2022/01/02 09:29:09     battery 0: power ✓ energy ✗ currents ✗ soc ✓
[cache ] DEBUG 2022/01/02 09:29:09 pvConfigured: true
[cache ] DEBUG 2022/01/02 09:29:09 batteryConfigured: true
[lp-1  ] INFO 2022/01/02 09:29:09 loadpoint 1:
[lp-1  ] INFO 2022/01/02 09:29:09   mode:      now
[lp-1  ] INFO 2022/01/02 09:29:09   charger:   power ✓ energy ✓ currents ✓ timer ✗
[lp-1  ] INFO 2022/01/02 09:29:09   meters:    charge ✓
[lp-1  ] INFO 2022/01/02 09:29:09     charge:  power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/01/02 09:29:09   vehicles:  ✓
[lp-1  ] INFO 2022/01/02 09:29:09     car 0:   range ✓ finish ✗ status ✓ climate ✗
[cache ] DEBUG 2022/01/02 09:29:09 prioritySoC: 50
[server] FATAL 2022/01/02 09:29:09 remove /tmp/evcc: operation not permitted
[cache ] DEBUG 2022/01/02 09:29:09 chargeConfigured: true
[main  ] FATAL 2022/01/02 09:29:09 listen tcp 192.168.0.76:7070: bind: address already in use
[site  ] DEBUG 2022/01/02 09:29:09 ----
[cache ] DEBUG 2022/01/02 09:29:09 error: remove /tmp/evcc: operation not permitted
pi@raspberrypi:~ $ sudo systemctl status evcc
● evcc.service - evcc
     Loaded: loaded (/etc/systemd/system/evcc.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-01-02 09:29:01 CET; 13s ago
   Main PID: 1932 (evcc)
      Tasks: 9 (limit: 4915)
        CPU: 522ms
     CGroup: /system.slice/evcc.service
             └─1932 /usr/local/bin/evcc --log error

Jan 02 09:29:01 raspberrypi systemd[1]: Started evcc.

utsavanand2, 2022-01-02:
Hi Thorsten

I believe you have an instance of evcc running already hence the error.

Can you please check the status of your systemd evcc process with `sudo
systemctl status evcc`?

yeehman, 2022-01-02:
Hey Utsav,
this is the outcome of the status check: Could it be possible that the old and the new version running simultaneous? How can I stop the old version?

pi@raspberrypi:~ $ sudo systemctl status evcc
● evcc.service - evcc
     Loaded: loaded (/etc/systemd/system/evcc.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-01-02 09:29:01 CET; 13s ago
   Main PID: 1932 (evcc)
      Tasks: 9 (limit: 4915)
        CPU: 522ms
     CGroup: /system.slice/evcc.service
             └─1932 /usr/local/bin/evcc --log error

Jan 02 09:29:01 raspberrypi systemd[1]: Started evcc.
pi@raspberrypi:~ $ evcc -l error
[server] FATAL 2022/01/02 09:29:38 remove /tmp/evcc: operation not permitted
pi@raspberrypi:~ $ sudo systemctl status evcc
● evcc.service - evcc
     Loaded: loaded (/etc/systemd/system/evcc.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-01-02 09:45:40 CET; 20min ago
   Main PID: 2150 (evcc)
      Tasks: 10 (limit: 4915)
        CPU: 6.328s
     CGroup: /system.slice/evcc.service
             └─2150 /usr/local/bin/evcc --log error

Jan 02 09:45:40 raspberrypi systemd[1]: Started evcc.
pi@raspberrypi:~ $

utsavanand2, 2022-01-02:
@yeehman The command would be 
`sudo systemctl stop evcc`

utsavanand2, 2022-01-02:
@andig I tried a test with @yeehman with the compiled binary of the pull request #2156 . The status error messages are gone.

@yeehman still gets the error messages `charger logic error: disabled but charging` and `charger out of sync: expected disabled, got enabled`.
Does the error mean that evcc didn't start the charging but the charging started on it's own? Hence out of sync?
I believe these errors are caused not by any errors not because of any issues but rather the setting of the charging station, if I'm correct @yeehman 's station is in plug and charge mode and not in the RFID mode, which means the charging starts automatically as soon as you plug the connector into your car which is what plug & charge mode is supposed to do.

@yeehman also reports that after the charging is started (while the charger is in plug and charge mode), the charging gets stopped after 20 seconds. @andig I am wondering if the charging gets stopped by evcc itself as it sees the charger out of sync and tries to get it back into sync by stopping it? Please correct me if I'm wrong.

andig, 2022-01-02:
> Does the error mean that evcc didn't start the charging but the charging started on it's own? Hence out of sync?

Possible.

> @yeehman also reports that after the charging is started (while the charger is in plug and charge mode), the charging gets stopped after 20 seconds. @andig I am wondering if the charging gets [...]

Yes, evcc would try to establish the expected (disabled) state.

> if I'm correct @yeehman 's station is in plug and charge mode and not in the RFID mode, which means the charging starts automatically as soon as you plug the connector into your car which is what plug & charge mode is supposed to do.

That would mean we would always need RFID mode? Should be documented or ensured by the plugin?

Can we merge the current PR?

utsavanand2, 2022-01-02:
Okay so this makes a lot of sense.

I think we need to document this, I don't think that at this point we can ensure it with the plugin as RFID mode can be enabled only by the user. Although we can add a field in the config yaml, where a user needs to specify if the plug and charge mode is enabled or not, if it is can we modify evcc to not try to sync with its own state, and just try to change the current settings and not start and stop charging sessions to fix the out of sync errors and making the charging stop?

But as it sits today, I think it's good to merge and ready for a new release to fix the issues on the server and evcc side.

utsavanand2, 2022-01-02:
@andig Is this where the documentation for the chargers live?
I can add a PR as a comment in the configuration yaml itself.

https://github.com/evcc-io/config#charger-daheimladen-cloud-api

andig, 2022-01-02:
@DerAndereAndi we beed to doxument the mode

utsavanand2, 2022-01-02:
Also can we remove the comment
"# an evcc sponsortoken is required for using this charger" ?
This is causing some users to get confused making them add a sponsor token
in the configuration which causes errors.

andig, 2022-01-02:
> Also can we remove the comment

Which location are you referring to?

closed 2022-01-02 by andig

utsavanand2, 2022-01-02:
> For sure. Which location are you referring to?
This comment? https://github.com/evcc-io/config/blob/f72415e57f509af89d049e1bbb51064f03494c52/README.md?plain=1#L1217

andig, 2022-01-02:
Never mind, we will deprecate the config repo in the near future 👍🏻

utsavanand2, 2022-01-02:
Thank you so much. Let me know if I can help you with a PR for the
documentation :)