evcc-io/evcc #2146: Warp2 Pro: MQTT timeouts and wrong phase count in web interface

bboehmke, 2021-12-29:
**Describe the bug**

`2P` instead of `3P` shown
[image]

also some MQTT timeouts in log

**To Reproduce**
Start manual charging with a Warp2 Pro

**Expected behavior**
`3p` instead of `2p`

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.74 (f6d9d5a)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
[main  ] INFO 2021/12/29 15:19:58 evcc 0.74 (f6d9d5a)
[main  ] INFO 2021/12/29 15:19:58 using config file /data/iot-pi/evcc/evcc.yml
[mqtt  ] INFO 2021/12/29 15:19:59 connecting evcc-1205594314 at tcp://192.168.4.13:1883
config
======

grid: grid
----------
Power:          481W
Energy:         7904.1kWh
Current L1..L3: 1.39A 0.42A 1.45A
Diagnostic dump:
  IP:          192.168.4.50:9522
  Serial:      ***
  EnergyMeter: true

  ActiveEnergyMinus:     5992121520 Ws
  ActiveEnergyMinusL1:   9617228280 Ws
  ActiveEnergyMinusL2:   11717222040 Ws
  ActiveEnergyMinusL3:   8850379320 Ws
  ActiveEnergyPlus:      28454586120 Ws
  ActiveEnergyPlusL1:    27848492640 Ws
  ActiveEnergyPlusL2:    7381860840 Ws
  ActiveEnergyPlusL3:    17416940400 Ws
  ActivePowerMinus:      0.000000 W
  ActivePowerMinusL1:    0.000000 W
  ActivePowerMinusL2:    0.000000 W
  ActivePowerMinusL3:    0.000000 W
  ActivePowerPlus:       481.100000 W
  ActivePowerPlusL1:     183.200000 W
  ActivePowerPlusL2:     51.700000 W
  ActivePowerPlusL3:     246.100000 W
  ApparentEnergyMinus:   12778698600 VAs
  ApparentEnergyMinusL1: 10409590080 VAs
  ApparentEnergyMinusL2: 12045916440 VAs
  ApparentEnergyMinusL3: 9719375040 VAs
  ApparentEnergyPlus:    39286109520 VAs
  ApparentEnergyPlusL1:  29453882040 VAs
  ApparentEnergyPlusL2:  9276667200 VAs
  ApparentEnergyPlusL3:  22615202520 VAs
  ApparentPowerMinus:    0.000000 VA
  ApparentPowerMinusL1:  0.000000 VA
  ApparentPowerMinusL2:  0.000000 VA
  ApparentPowerMinusL3:  0.000000 VA
  ApparentPowerPlus:     637.700000 VA
  ApparentPowerPlusL1:   239.200000 VA
  ApparentPowerPlusL2:   89.800000 VA
  ApparentPowerPlusL3:   311.800000 VA
  CurrentL1:             1.394000 A
[... 571 lines cut ...]
[cache ] DEBUG 2021/12/29 13:24:09 chargeRemainingEnergy: 4299.999999999999
[cache ] DEBUG 2021/12/29 13:24:09 vehicleRange: 92
[lp-1  ] DEBUG 2021/12/29 13:24:09 vehicle odometer: 21716km
[cache ] DEBUG 2021/12/29 13:24:09 vehicleOdometer: 21715.5
[lp-1  ] DEBUG 2021/12/29 13:24:09 charger disable
[cache ] DEBUG 2021/12/29 13:24:09 chargeCurrent: 0
[lp-1  ] DEBUG 2021/12/29 13:24:09 charge power: 9097W
[lp-1  ] DEBUG 2021/12/29 13:24:09 detected phases: 2p (16.0A @ 9097W)
[cache ] DEBUG 2021/12/29 13:24:09 chargePower: 9097.265
[cache ] DEBUG 2021/12/29 13:24:09 activePhases: 2
[cache ] DEBUG 2021/12/29 13:24:09 homePower: 529.335000000001

</pre></details>

**If using Docker:**
Show output of <code>docker run andig/evcc -v</code>:
```
<- paste here

```

<details><summary>Show evcc log output with <code>docker logs <container id></code>:</summary><br><pre>
<- paste here

</pre></details>

andig, 2021-12-30:
@bboehmke der 2p Fehler sollte behoben sein. Für den `timeout` müsstest Du mir die Box bitte mal auf `broker.hivemq.com` legen und das Topic per PM teilen.

bboehmke, 2021-12-31:
mh ich hab jetzt einen anderen MQTT broker verwendet und nun sind die timeouts weg.
Ich würde vorschlagen ich beobachte das mal weiter und melde mich falls ich das reproduzieren kann bzw es Probleme macht

andig, 2022-01-02:
Hier haben wir einen Konflikt der Phasenlösung mit https://github.com/evcc-io/evcc/issues/2177

andig, 2022-01-02:
Hier ist das Problem wiefolgt:

W | A | phases
-- | -- | --
9114 | 16 | 2,47663043

Wir sind offensichtlich am Ladeende und die Leistung geht langsam runter. An der Stelle muss die Phasenschätzung versagen!

Der Fix mittels `Ceil` ist m.E. falsch. Richtig wäre stattdessen, sofern über Leistung abgeschätzt wird, die ermittelte geschätzte Phasenzahl _während_ eines Ladevorgangs nicht mehr zu reduzieren.

/cc @premultiply

bboehmke, 2022-01-02:
In meinem Fall war das aber der Ladebeginn bei etwa 60% soc. 
Ein einmaliges ermitteln der Phasen klingt da auch nicht ganz richtig und würde in diesem Fall wohl auch nicht helfen.

premultiply, 2022-01-02:
Ceil ist korrekt, denn es gibt nur ganze Phasen.
Deine sonstige Erklärung ist aber auch korrekt. Bei der abfallenden Ladeleistung wird es uneindeutig. Ebenso bei kalten Akkus etc.
Ist halt nur eine Schätzung. Wer es korrekt haben will braucht eine Phasenstrommessung.

bboehmke, 2022-01-02:
Achso und mein ursprüngliches Problem wurde ja mit #2141 behoben.

andig, 2022-01-02:
> Ein einmaliges ermitteln der Phasen klingt da auch nicht ganz richtig und würde in diesem Fall wohl auch nicht helfen.

Davon hab ich auch nicht geschrieben. Ich sagte: keine Reduktion der Phasen während des Ladevorgangs ;)

Evtl. könnte man zusätzlich noch etwas tricksen und sowas in der Art machen:

    phases := int(math.Ceil(lp.chargePower / Voltage / lp.chargeCurrent - 0.2))

Dadurch würden wir kleine Übertretungen weiter nach unten runden. Es wird aber beide Komponenten für einen richtigen Fix brauchen.

andig, 2022-01-02:
Mit https://github.com/evcc-io/evcc/commit/94b889707c8c1f20b9396e2cb04e2a8b9f4ab921 nach Slackdiskussion erledigt. Zum Ladeende/ bei kaltem Akku kann/wird es weiter tw. falsch sein.

closed 2022-01-02 by andig