evcc-io/evcc #2150: Kostal Piko 15, Fehler: add[1]: strconv.ParseFloat: parsing "<nil>": invalid syntax

Tuffiglig, 2021-12-29:
**EVCC details:**
```
evcc version 0.74 (f6d9d5a)
```

I tried to play around with evcc in Virtualbox with Ubuntu 20.04 as OS.
I tried to generate the first config file with
`evcc configure`
But i got an error (hope german output is OK?):
```
Wähle ein PV System: Kostal Piko + Piko BA
IP-Adresse oder Hostname 192.168.156.91
Teste die grid Konfiguration von Kostal Piko + Piko BA ...
   Fehler: add[1]: strconv.ParseFloat: parsing "<nil>": invalid syntax
```

PV inverter is a Kostal Piko 15 with Firmware version 06.16.

what could be the problem?

DerAndereAndi, 2021-12-29:
We just released version 0.75, could you please try it with that again?

Tuffiglig, 2021-12-29:
With evcc version 0.75 it is not crashing anymore, if i select my inverter.
Now i only need to add a supported Wallbox, to get the config healthcheck passing.
Trying now to figure out how this has to be done.

DerAndereAndi, 2021-12-29:
Then we can close this?

The configuration process will ask you for your wallbox when creating the loadpoint. Please use the forum if you have further questions to his.

closed 2021-12-29 by Tuffiglig