evcc-io/evcc #2153: charge power:  4294967295

Hofyyy, 2021-12-29:
**Describe the bug**
Charge power toggled ohne Verbindung zum Auto
selten aber regelmäßig von 0 auf 4294967295.

**To Reproduce**
Ist bei mir einfach vorhanden, siehe log.

[syslog.txt](https://github.com/evcc-io/evcc/files/7790708/syslog.txt)
[evcc_clear.yaml.txt](https://github.com/evcc-io/evcc/files/7790713/evcc_clear.yaml.txt)

Hofyyy, 2021-12-30:
Ich hab mal den Taschenrechner bemüht. 0xFFFFFFFF.
Jetzt ist die Frage ob das ein gültiger Nullwert ist, oder wo das herkommt.

premultiply, 2021-12-30:
Interessant. Ich gehe aber erstmal davon aus, dass das so schon von der Wallbe kommt.
Genaueres dürfte ein Tracelog der Modbus-Kommunikation liefern.

Hofyyy, 2021-12-30:
Hast du da mal

> Interessant. Ich gehe aber erstmal davon aus, dass das so schon von der Wallbe kommt. Genaueres dürfte ein Tracelog der Modbus-Kommunikation liefern.

Ich kann mal gucken ob ich den gleichen Wert auch im WEB in der Wallbe sehen kann. Also die Frage ist, sollen wir das mit der Wallbe mal ausforschen, oder ist bei Modbus 0xFFFFFFFF auch einfach als Null zu mappen und gut ist?

andig, 2021-12-30:
Nein weiles keine Null ist. Bitte —log trace.

Hofyyy, 2021-12-30:
> Nein weiles keine Null ist. Bitte —log trace.

@andig 
Hallo, ich habe mal in der config alles auf trace gesetzt und den Service neugestartet.
Leider bekomme ich bisher nur debug Ausgaben. Müsste ich in /var/log/syslog sofort
Tracelogs sehen?

... log settings
log: trace
levels:
  core: trace
  lp-1: trace
  lp-2: trace

Meine bisherige Analyse:
- Die Werte der Stromrichtungen sind die ganze Zeit ganz leicht am schwanken. Siehe Fotos.
- Wenn ich es richtig sehe ist der Modbus offset im Sourcecode 100 und das benutzte Register 20, was Adresse 120 macht.
- Wie man im Fotos sieht verändert es sich die ganze Zeit leicht.

Ich würde vermuten, das hier in dem Fehlerfall eine -1 kommt was dann als 0xFFFFFFFF codiert ist?
[image]
[image]

premultiply, 2021-12-30:
Ausgewertet wir Active Power (W).
Wenn hier seitens evcc UINT als externer Modbus-Datentyp erwartet wird ist es erklärbar was passiert wenn der Zählerwert minimalst negativ wird.

Hofyyy, 2021-12-30:
Kannst du mir die Sourcecodestelle verraten wo du Active Power siehst? Ich finde nur 120 ...

Aber davon mal abgesehen ist der Fehler dann ja gefunden, da das Register in der Wallbe als "int" deklariert ist. Ich denke man muss ja signed an der stelle nicht unterstützen, aber man könnte alles kleiner 0 auf Null mappen oder?

premultiply, 2021-12-30:
Du verwechselst da was. Das Wallbe-WebIF konfiguriert die interne Modbus-Kommunikation mit dem Zähler.
Auf der externen Schnittstelle fragt evcc derzeit auf Register ein UINT32-Wert ab. Dort trifft dann das Problem auf wenn der vom Controller gerundete Wert kurz negativ sein sollte.

Hofyyy, 2021-12-30:
1.
Also die Logleven in der evcc.yaml scheinen nicht zu funktionieren. Kann das sein? Wenn ich es manuell mit evcc --log trace mache hat es funktioniert?

2.
Anbei mal die modbus kommunikation. Einfach nach 4294967295W suchen.
[log.txt](https://github.com/evcc-io/evcc/files/7794228/log.txt)

Hofyyy, 2021-12-30:
> Du verwechselst da was. Das Wallbe-WebIF konfiguriert die interne Modbus-Kommunikation mit dem Zähler. Auf der externen Schnittstelle fragt evcc derzeit auf Register ein UINT32-Wert ab. Dort trifft dann das Problem auf wenn der vom Controller gerundete Wert kurz negativ sein sollte.

Ok da kenne ich mich im modbus noch nicht aus. Ich hätte erwartet da stehen hexwerte in RAW und man muss wissen wie diese zu interpretieren sind und kann sich das nicht aussuchen. ...

premultiply, 2021-12-30:
Hier liegt das Problem:
https://github.com/evcc-io/evcc/blob/e017ce058600d8367dab2ac2c7382fd997f02bc2/charger/wallbe.go#L194

Hofyyy, 2021-12-30:
Erstmal danke für die Zeile.

Da wird doch aber folgendes Register gelesen oder?
`	wbRegPower          = 120 // power reading
`

Welches nach alles was ich sehe signed Daten hält. Was ja auch den Fehler erklären würde.

premultiply, 2021-12-30:
Ja, Register 120 des Controllers! Das hat aber nix mit dem Register 20 des Zählers zu tun. Die Ähnlichkeit in rein zufällig.

Hofyyy, 2021-12-30:
Ahh ok.
Vielen Dank für die Erklärung.

Gibt es dazu vom Hersteller eine feste Map.
Oder kann man sich ein Externes Register programmieren und dann immer abfragen?

premultiply, 2021-12-30:
Das machst du genau mit der oben abgebildeten Tabelle im Web-Interfaces des Controllers.

andig, 2021-12-30:
Für Power macht ja int Sinn- sollten wir ändern @premultiply ?

premultiply, 2021-12-30:
PR ist doch schon da... ;)

Hofyyy, 2021-12-31:
@premultiply 
Vielleicht kannst du dir das hier noch mal angucken.

Ich bekomme jetzt nachvollziehbarerweise folgende Meldung:
`	lp.log.WARN.Printf("charge power must not be negative: %.0f", lp.chargePower)
`

Jetzt stelle ich mir zwei Fragen:
- Nimmt man die Warning komplett raus?
- Oder wrappt man Sie in ein !StateA && !StateB mit dem Hinblick auf V2H?

Also wenn es aktiv am Laden ist finde ich die Meldung sinnig, sonst scheint das ja zu Wandern.

Was mich auch noch wundert ist, wie er Rundet. Ich hätte erwartet das er -1 Watt im Register rausgibt und das dann mit 0.001 multipliziert wird. Runden wir das dann auf 1 auf?

premultiply, 2021-12-31:
Nee andersrum. Der Controller multipliziert den Zählerwert intern mit dem Faktor, rundet das irgendwie (vielleicht ab?) und gibt dann ganzahlige Werte in Watt aus. Wahrscheinlich rauscht der Messwert des Zählers einfach etwas und das kommt dabei heraus wenn er ganz minimal ins Negative rutscht.
Vielleicht kann man dem Zähler durch Konfiguration es ja irgendwie beibringen nur in eine Richtung zu messen?
Sonst kann man eigentlich nur sagen die Warnung vielleicht erst ab weniger als -1 auszugeben? /cc @andig
Vielmehr fällt mir gerade nicht ein.

Hofyyy, 2021-12-31:
Ahh ok. 

Ich beobachte und debugge das die Tage mal. Ich habe eben im GUI auch kurz mal das 3 Phasen logo gesehen. (Laden mit 3 Phasen), ohne das ein Auto angeschlossen war. Da gucke ich auch mal nach wann das angezeigt wird.

Ich denke auch das der Zähler rauscht, ist ja auch mal nicht weiter schlimmt. Die Frage ist ob wir wenn nichts angeschlossen ist, diese Werte verwenden wollen. Daher die Frage ob man die Warnung nicht an den State koppelt. Gleiches ginge auch bei der 1. oder 3 Phasen logic. Das wäre dann ja auch Wallbox unabhängig.

premultiply, 2022-01-01:
Die Warning sollte IMHO nur angezeigt werden wenn Werte < -1 um solche Rauscheffekte zu eleminieren.

andig, 2022-01-01:
> Die Warning sollte IMHO nur angezeigt werden wenn Leistungswerte < -1 annehmen um solche Rauscheffekte zu eleminieren.
> Auch die Phasenerkennung muss diesbezüglich nochmal validiert werden ob nur Betrag(current) > 1 Watt als aktiv detektiert wird.

Ich kenne ehrlich gesagt bisher keinen Zähler, der Einspeisung anstatt Null anzeigt. Sieht aus wie ein Fehler in der Wallbe Firmware- ich würde da egtl. keinen Workaround machen zumal es außer der Warnung auch kein wirkliches Problem darstellt.

Aus meiner Sicht zumachen.

Hofyyy, 2022-01-01:
> Die Warning sollte IMHO nur angezeigt werden wenn Leistungswerte < -1 annehmen um solche Rauscheffekte zu eleminieren.
> 
> Auch die Phasenerkennung muss diesbezüglich nochmal validiert werden ob nur Betrag(current) > 1 Watt als aktiv detektiert wird.

Also wenn ich das mit den Registern besser verstanden habe, kann ich da gerne auf Wallbe zugehen und das melden. Es spricht ja nichts dagegen das die Firmware das sauber glättet. In zukunft mag das aber auch bei anderen auftauchen, da es ja nun mal eine A/D wandlung ist.

Ich finde des Vorschlag gegen <-1 und >1 zu validieren daher auch sinnvoll, da es EVCC robuster und besser macht. Es stellt ja keinen komplexen Workaround da der die Software verschlechtert, sondern erhöht den Nutzen der User. Auch wenn die Ursache in der Implemetation der Wallbox hängt.

andig, 2022-01-01:
In https://github.com/evcc-io/evcc/commit/3b7f3824a586ca4f4abd78db26372f265719f85e behoben

PR #2161 (2021-12-30): Fix Wallbe int32 decoding
fixes #2153
closed 2022-01-01 by andig