evcc-io/evcc #2165: vehicleRange and vehicleOdometer is published initialy with zero after evcc start

F-L-0, 2021-12-30:
**Describe the bug**
After starting evcc vehicleRange and vehicleOdometer is published via mqtt and sent to influx with value "0", and afterwards in the same second with the correct value. This leads to strange graphs when plotting those values.
For vehicleSoc this works in my setup correctly, but I guess thats just a timing issue since the http-Request to VW API is just a few ms before the first values are published, but filling all variables with the correct values from API needs more than one api call.
I assume after retrieving batterySoc from vehicle API the two other metrics vehicleRange and vehicleOdometer are sent to evcc cache with default value 0 and soon after updated with the correct values once subsequent api calls in another thread are finished.

**To Reproduce**
Steps to reproduce the behavior:
1. 
/usr/local/bin/evcc --log trace --metrics 2>&1 | tee -a /var/tmp/evcc_trace.log

2. 
root@rpi4-2:~# egrep -e "(\[cache|\[influx)" /var/tmp/evcc_trace.log
[cache ] DEBUG 2021/12/30 21:52:33 title: Garage
[influx] TRACE 2021/12/30 21:52:33 write minCurrent=6 (map[loadpoint:Garage vehicle:])
[cache ] DEBUG 2021/12/30 21:52:33 minCurrent: 6
[influx] TRACE 2021/12/30 21:52:34 write maxCurrent=16 (map[loadpoint:Garage vehicle:])
[cache ] DEBUG 2021/12/30 21:52:34 maxCurrent: 16
[cache ] DEBUG 2021/12/30 21:52:34 phases: 2
[influx] TRACE 2021/12/30 21:52:34 write phases=2 (map[loadpoint:Garage vehicle:])
[cache ] DEBUG 2021/12/30 21:52:34 activePhases: 2
[influx] TRACE 2021/12/30 21:52:34 write activePhases=2 (map[loadpoint:Garage vehicle:])
[cache ] DEBUG 2021/12/30 21:52:34 hasVehicle: true
[cache ] DEBUG 2021/12/30 21:52:34 mode: pv
[influx] TRACE 2021/12/30 21:52:34 write targetSoC=85 (map[loadpoint:Garage vehicle:])
[cache ] DEBUG 2021/12/30 21:52:34 targetSoC: 85
[influx] TRACE 2021/12/30 21:52:34 write minSoC=80 (map[loadpoint:Garage vehicle:])
[cache ] DEBUG 2021/12/30 21:52:34 minSoC: 80
[cache ] DEBUG 2021/12/30 21:52:34 vehiclePresent: true
[cache ] DEBUG 2021/12/30 21:52:34 vehicleTitle: eGolf
[cache ] DEBUG 2021/12/30 21:52:34 vehicleCapacity: 36
[influx] TRACE 2021/12/30 21:52:34 write vehicleCapacity=36 (map[loadpoint:Garage vehicle:eGolf])
**[cache ] DEBUG 2021/12/30 21:52:34 vehicleRange: 0
[cache ] DEBUG 2021/12/30 21:52:34 vehicleOdometer: 0
[influx] TRACE 2021/12/30 21:52:34 write vehicleRange=0 (map[loadpoint:Garage vehicle:eGolf])
[influx] TRACE 2021/12/30 21:52:34 write vehicleOdometer=0 (map[loadpoint:Garage vehicle:eGolf])**
[cache ] DEBUG 2021/12/30 21:52:34 siteTitle: Daheim
[cache ] DEBUG 2021/12/30 21:52:34 gridConfigured: true
[cache ] DEBUG 2021/12/30 21:52:34 pvConfigured: true
[cache ] DEBUG 2021/12/30 21:52:34 batteryConfigured: true
[influx] TRACE 2021/12/30 21:52:34 write prioritySoC=70 (map[])
[cache ] DEBUG 2021/12/30 21:52:34 prioritySoC: 70
[influx] TRACE 2021/12/30 21:52:34 write pvPower=0 (map[])
[cache ] DEBUG 2021/12/30 21:52:34 pvPower: 0
[cache ] DEBUG 2021/12/30 21:52:34 chargeConfigured: false
[influx] TRACE 2021/12/30 21:52:34 write gridPower=1242 (map[])
[cache ] DEBUG 2021/12/30 21:52:34 gridPower: 1242
[cache ] DEBUG 2021/12/30 21:52:34 batteryPower: 0
[influx] TRACE 2021/12/30 21:52:34 write batteryPower=0 (map[])
[cache ] DEBUG 2021/12/30 21:52:34 gridCurrents: [2.214 1.829 2.833]
[cache ] DEBUG 2021/12/30 21:52:34 gridEnergy: 19828.3621
[influx] TRACE 2021/12/30 21:52:34 write gridCurrents=[2.214 1.829 2.833] (map[])
[influx] TRACE 2021/12/30 21:52:34 write gridEnergy=19828.3621 (map[])
[influx] TRACE 2021/12/30 21:52:34 write batterySoC=13 (map[])
[cache ] DEBUG 2021/12/30 21:52:34 batterySoC: 13
[cache ] DEBUG 2021/12/30 21:52:34 mode: pv
[cache ] DEBUG 2021/12/30 21:52:34 chargePower: 0
[cache ] DEBUG 2021/12/30 21:52:34 chargeCurrent: 0
[influx] TRACE 2021/12/30 21:52:34 write chargePower=0 (map[loadpoint:Garage vehicle:eGolf])
[influx] TRACE 2021/12/30 21:52:34 write chargeCurrent=0 (map[loadpoint:Garage vehicle:eGolf])
[cache ] DEBUG 2021/12/30 21:52:34 chargedEnergy: 0
[cache ] DEBUG 2021/12/30 21:52:34 chargeDuration: 6m17s
[influx] TRACE 2021/12/30 21:52:34 write chargedEnergy=0 (map[loadpoint:Garage vehicle:eGolf])
[cache ] DEBUG 2021/12/30 21:52:34 connectedDuration: 0s
[cache ] DEBUG 2021/12/30 21:52:34 chargeCurrent: 0
[cache ] DEBUG 2021/12/30 21:52:34 connected: false
[influx] TRACE 2021/12/30 21:52:34 write chargeCurrent=0 (map[loadpoint:Garage vehicle:eGolf])
[cache ] DEBUG 2021/12/30 21:52:34 charging: false
[cache ] DEBUG 2021/12/30 21:52:34 enabled: false
[cache ] DEBUG 2021/12/30 21:52:34 vehicleSoC: 48
[cache ] DEBUG 2021/12/30 21:52:34 chargeRemainingDuration: -1ns
[influx] TRACE 2021/12/30 21:52:34 write vehicleSoC=48 (map[loadpoint:Garage vehicle:eGolf])
[cache ] DEBUG 2021/12/30 21:52:34 chargeRemainingEnergy: 14800
[cache ] DEBUG 2021/12/30 21:52:34 vehicleRange: 81
[influx] TRACE 2021/12/30 21:52:34 write chargeRemainingEnergy=14800 (map[loadpoint:Garage vehicle:eGolf])
**[influx] TRACE 2021/12/30 21:52:34 write vehicleRange=81 (map[loadpoint:Garage vehicle:eGolf])
[influx] TRACE 2021/12/30 21:52:34 write vehicleOdometer=27634 (map[loadpoint:Garage vehicle:eGolf])**
[cache ] DEBUG 2021/12/30 21:52:34 vehicleOdometer: 27634
[cache ] DEBUG 2021/12/30 21:52:34 homePower: 1242
[cache ] DEBUG 2021/12/30 21:52:34 chargePower: 0
[influx] TRACE 2021/12/30 21:52:34 write homePower=1242 (map[])
[influx] TRACE 2021/12/30 21:52:34 write chargePower=0 (map[loadpoint:Garage vehicle:eGolf])

**Expected behavior**
vehicle metrics are published/written to influx/mqtt after start of evcc only when successfuly retreived from vehicle API. In case of API errors no values are written to influx/mqtt.

F-L-0, 2022-01-29:
Hallo @andig ,
vielleicht noch eine Info warum ich es für sinnvoll halte das keine 0 Werte in die Influx + MQTT geschrieben werden.
Aus Sicht von nachfolgenden Automationen (die von den mqtt-Werten abhängig sind) ist es besser wenn kein Wert kommt als wenn ein falscher Wert geschickt wird. Neben den "kaputten" Graphen im Grafana wären das dann zB auch SoC-abhängig Automatisierungen in der Hausautomation. OK, 0 Werte kann man in den Plot-Definitionen im Grafana und auch in irgendwelchen DoIf's in der Hausautomation wieder rausfiltern, aber es wäre schön wenn man das an der Quelle beheben kann.

Gibt es Gründe warum 0-Werte überhaupt in die Influx oder an den Broker geschickt werden sollten? Wenn zB die API vom BEV das nicht liefert könnte man das doch komplett raus lassen oder?

andig, 2022-01-29:
> Gibt es Gründe warum 0-Werte überhaupt in die Influx oder an den Broker geschickt werden sollten?

Alle Datensenken hängen an der gleichen Versorgung. Auch das UI!

Mir kam grad die Idee, statt 0 (oder -1) mal `nil` zu schreiben. Das wäre dann eindeutig als "unbekannt" zu erkennen. Können wir nach dem Release mal in nightly ausprobieren.

andig, 2022-01-29:
@F-L-0 könntest Du das mal probieren?

F-L-0, 2022-01-29:
Hi,
gerne, ich scheitere aber gerade am bauen. Das mag auch an mir liegen...

andig, 2022-01-30:
Probier nochmal, sollte jetzt bauen

F-L-0, 2022-01-30:
Yup, baut, allerdings wird jetzt NIL als string sowohl via mqtt als auch an die Influx geschickt.

evcc:
`Jan 30 10:57:19 rpi4-2 evcc[7510]: [influx] ERROR 2022/01/30 10:57:19 write failed (attempts 0): 400 Bad Request: partial write: field type conflict: input field "value" on measurement "vehicleRange" is type string, already exists as type integer dropped=2`

fhem (als Empfänger via MQTT):
`2022.01.30 10:57:18 1: PERL WARNING: Argument "<nil>" isn't numeric in sprintf at (eval 424934) line 1.`

Die Auswirkungen in der UI kann ich nicht beurteilen, da NIL ja nach ein paar ms wenn der Abruf via VW API fertig ist wieder mit dem korrekten Wert überschrieben wird.

andig, 2022-01-30:
> Request: partial write: field type conflict: input field "value" on measurement "vehicleRange" is type string, already exists as type integer dropped=2

Kann ich nicht nachvollziehen:

    [influx] TRACE 2022/01/30 12:27:09 write foo=map[value:<nil>] (map[])

Ich sehe nicht, wo da `string` her kommen sollte. Das müsstest Du leider debuggen :( Mqtt ist ein anderes Thema.

andig, 2022-01-30:
Mqtt nil sollte gefixt sein. Zu string hab ich keine Idee am Besten in influx.go.Run() debuggen :(

andig, 2022-01-30:
> PERL WARNING: Argument "<nil>" isn't numeric in sprintf at (eval 424934) line 1.

`""` wäre natürlich auch nicht numeric, aber das wäre dann das Problem von fhem ;)

F-L-0, 2022-01-30:
Jetzt kommt sowohl im mqtt als auch "Richtung" Influx NULL / nil an. Sowohl der influx-writer in evcc als auch dort wo weiterverarbeitet wird (mqtt -> fhem -> doif bzw mysql persistierung) laufen diese Werte auf einen Fehler und werden nicht in der Datenbank gespeichert. Der Restart von evcc führt damit nicht mehr zu Fehlern im Plot von Grafana.
Ich probier mich jetzt mal daran das zu debuggen - und fang damit an mal zu googlen was du mit "influx.go.Run()" meinst :-)

andig, 2022-01-30:
Bei mqtt sollte leerer String ankommen. In evcc sollte das kein Fehler sein. Alles Andere ist kein evcc Problem ;)

F-L-0, 2022-01-30:
Stimmt, im mqtt ist ein leerer String.
Ich hab mal etwas "probiert" - und hätte ne Frage. Wenn ich die beiden Zeilen

810 lp.publish("vehicleRange", nil)
811 lp.publish("vehicleOdometer", nil)

in loadpoint.go auskommentiere werden keine initialen nil/"" Werte mehr geschrieben, nur noch die die von der VW API ein paar ms später kommen. Habe zum gegenprüfen auch mal in der evcc.yaml die config für die VW API rausgenommen um zu schauen ob der initale publish evtl fürs Frontend gebraucht wird, aber auch da scheint es auf den ersten Blick zu passen (Reichweite wird in der UI durch Dauer ersetzt, SoC durch "getrennt").
Jetzt hat der lp.publish für evvc intern bestimmt einen Zweck - oder wäre das vielleicht sogar eine Möglichkeit das Problem gleich zu umgehen?

andig, 2022-01-31:
> Wenn ich die beiden Zeilen auskommentiere werden keine initialen nil/"" Werte mehr geschrieben,

Das nutzt nichts da auch bei Abmelden des Fahrzeugs die Werte wieder entfernt werden- es gibt dann ja genau keine gültigen Werte. 

Was ich mir noch vorstellen könnte, wäre zumindest ggü. Influx die Werte mit den jeweils identifizierten Fahrzeug zu taggen und v.a. nicht-getaggte Werte (das sind dann die Null/Nil) auch nicht nach Influx zu schreiben. 
An der MQTT-Problematik ändert das allerdings erstmal nichts. Auch da wäre denkbar, eine tiefere Struktur aufzubauen.

Ist alles relativ aufwändig und hat keine Prio...

F-L-0, 2022-02-02:
> Ist alles relativ aufwändig und hat keine Prio...
kann ich verstehen, dann belassen wir den Zustand erstmal wie er ist, oder kann zumindest der "Fix" von hier mit rein?

Tendenziell würden von der Struktur her diese Metriken auch besser ans Vehicle passen, und nicht an den LP, oder? Mit einem BEV ist das ja kein Problem/egal, aber dein Kommentar hat mich erst darauf gebracht, ab 2 Fahrzeugen kann da eigentlich nix sinnvolles mehr rauskommen wenn im selben Influx-Feld die Werte verschiedener BEV landen :-)

andig, 2022-02-02:
Isso... ändern wir auch sobald der Parkplatz kommt. Unter dem Aspekt würde ich hier schließen. Bleibt die Frage ob der PR nil statt anderer Werte zu schreiben nicht trotzdem sinnvoll wäre. Würde ich gerne so umsetzen wenns keine unüberwindbaren Hürden gäbe.

closed 2022-02-04 by andig