evcc-io/evcc #2170: evcc does not honor prioritySoC in PV mode when loading to minSoC is finished

F-L-0, 2021-12-31:
**Describe the bug**
Setup: Wallbe Eco 2.0s, eGolf, SMA Tripower STP10000, Sunny Island SI 6.0H Battery

Normal behaviour: evcc starts loading the vehicle in PV mode only after prioritySoC is loaded to battery. This is working fine. 

When connecting the car to the wallbox with a SoC less than minSoC evcc running in mode "PV" starts loading the vehicle with 16A@2P immediately even when prioritySoC is not fulfilled (correct behaviour). However, after minSoC is reached evcc continues to load the vehicle with all available PV power and does not give the power to the home battery system.
Home battery system is using all available power that would be sent to grid (adapting each 1000ms, measurement done by SMA Energy Meter). SMA features like prognostic charging are disabled.

**To Reproduce**
Steps to reproduce the behavior:
1. set minSoC 2% above current SoC of your vehicle
2. set prioritySoC well above current SoC of your battery system
3. /usr/local/bin/evcc --log debug --metrics 2>&1 | tee -a /var/tmp/evcc_trace.log
4. evcc starts charging with maximum power (in my case 16A@2P)
5. once minSoC is reached all available power should go to home battery system until prioritySoC is reached, but evcc keeps loading vehicle with all available power
6. switching evcc to "stop" mode immediately leads to home battery system charging with all available power

**Expected behavior**
see above

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.76 (07c559b)

```

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>

```
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
# sponsortoken:

# log settings
log: error
levels:
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
- name: grid
  type: sma
  uri: 192.168.1.13
- name: STP10000
  type: sma
  uri: 192.168.1.11
- name: SI60H
  type: sma
  uri: 192.168.1.12

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/andig/evcc-config#chargers
chargers:
- name: wallbe
  type: wallbe # Wallbe charger
  uri: 192.168.1.15:502 # ModBus address

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/andig/evcc-config#vehicles
vehicles:
- name: eGolf
[... 3375 lines cut ...]
[site  ] DEBUG 2021/12/31 11:48:30 battery power: -60W
[cache ] DEBUG 2021/12/31 11:48:30 batteryPower: -60
[site  ] DEBUG 2021/12/31 11:48:30 grid currents: [2.36 2.63 4.16]A
[site  ] DEBUG 2021/12/31 11:48:30 battery soc 0: 38%
[cache ] DEBUG 2021/12/31 11:48:30 gridCurrents: [2.357 2.626 4.156]
[cache ] DEBUG 2021/12/31 11:48:30 gridEnergy: 19835.5688
[cache ] DEBUG 2021/12/31 11:48:30 batterySoC: 38
[site  ] DEBUG 2021/12/31 11:48:30 giving priority to battery charging at soc: 38
[site  ] DEBUG 2021/12/31 11:48:30 site power: -210W
[lp-1  ] DEBUG 2021/12/31 11:48:30 charge power: 5603W
[cache ] DEBUG 2021/12/31 11:48:30 mode: pv
[cache ] DEBUG 2021/12/31 11:48:30 chargePower: 5603.4
[cache ] DEBUG 2021/12/31 11:48:30 chargeCurrent: 8.12086956521739
[cache ] DEBUG 2021/12/31 11:48:30 chargedEnergy: 1209.7720088309552
[cache ] DEBUG 2021/12/31 11:48:30 chargeDuration: 11m46s
[lp-1  ] DEBUG 2021/12/31 11:48:30 charger status: C
[lp-1  ] DEBUG 2021/12/31 11:48:30 next soc poll remaining time: 59m50s
[lp-1  ] DEBUG 2021/12/31 11:48:30 soc estimated: 50.35% (vehicle: 50.00%)
[lp-1  ] DEBUG 2021/12/31 11:48:30 vehicle soc: 50%
[cache ] DEBUG 2021/12/31 11:48:30 connected: true
[cache ] DEBUG 2021/12/31 11:48:30 charging: true
[cache ] DEBUG 2021/12/31 11:48:30 enabled: true
[cache ] DEBUG 2021/12/31 11:48:30 vehicleSoC: 50
[lp-1  ] DEBUG 2021/12/31 11:48:30 vehicle range: 81km
[cache ] DEBUG 2021/12/31 11:48:30 chargeRema

andig, 2021-12-31:
> When connecting the car to the wallbox with a SoC less than minSoC evcc running in mode "PV" starts loading the vehicle with 16A@2P immediately even when prioritySoC is not fulfilled (correct behaviour). 

Not sure we ever defined this but it makes sense ;)

> However, after minSoC is reached evcc continues to load the vehicle with all available PV power and does not give the power to the home battery system.

Das Problem dürfte hier sein:

		// if battery is charging below prioritySoC give it priority
		if socs < site.PrioritySoC && batteryPower < 0 {
			site.log.DEBUG.Printf("giving priority to battery charging at soc: %.0f", socs)
			batteryPower = 0
		}

/cc @premultiply

premultiply, 2021-12-31:
Sicher? Das Merkwürdige ist ja, dass das Problem wohl nur Auftritt wenn es aus dem "minSoC-Bereich" heraus kommt.

andig, 2021-12-31:
> Sicher? Das Merkwürdige ist ja, dass das Problem wohl nur Auftritt wenn es aus dem "minSoC-Bereich" heraus kommt.

Vorher wirds vom Ladepunkt ignoriert da der einfach drauf los lädt. Danach wird schon geladen, so dass die Batterie vmtl. keine Leistung mehr aufnimmt- daher glaube ich dass die `if` Abfrage das Problem enthält.

premultiply, 2021-12-31:
Glaube ich irgendwie noch nicht, da der BWR normalerweise ganz schnell jedes Watt aufsaugt - es sei denn er würde zwischenzeitlich richtig schlafen und die Batterie trennen. Dann dauert das Wecken mit Resync wieder etwas länger. Glaube ich aber auch noch nicht dran.

F-L-0, 2021-12-31:
Die Batterie hat nicht geschlafen und auch sauber mitgeregelt, jedes Watt an Einspeisung wurde sofort (Regelintervall 1s) in die Batterie gepackt bzw wenn evcc mehr gezogen hat wurde vom Hausakku der Netzbezug ausgeglichen. Sieht man auch schön im Übergang zu vor und nach der Ladephase.
Zur verdeutlichung noch ein Plot der evcc-Daten aus Grafana im Anhang - die Timestamps passen zum Logfile und der Auflistung der relevanten Stellen von oben:

<details><pre>
Start of evcc
[main ] INFO 2021/12/31 11:36:15 evcc 0.76 (07c559b)

All available power goes to home battery
[site ] DEBUG 2021/12/31 11:36:20 battery power: -3350W

evcc knows that prioritySoC is not reached
[site ] DEBUG 2021/12/31 11:36:20 giving priority to battery charging at soc: 40

vehicle is connected
[lp-1 ] INFO 2021/12/31 11:36:40 car connected

vehicle starts charging, SoC of vehicle (48%) is below minSoC (50%)
[lp-1 ] INFO 2021/12/31 11:36:50 start charging ->

minSoC is reached, chargePower reduced
[lp-1 ] DEBUG 2021/12/31 11:43:40 charge power: 3910W
[lp-1 ] DEBUG 2021/12/31 11:43:40 vehicle soc: 50%

evcc detects 3p charging (not correct, perhaps another issue?)
[lp-1 ] DEBUG 2021/12/31 11:45:10 max charge current: 7.33A = 7.26A + 0.0672A (-31W @ 2p)
[lp-1 ] DEBUG 2021/12/31 11:45:10 detected phases: 3p (7.3A @ 3370W)

manually stopped charging via GUI
[lp-1 ] INFO 2021/12/31 11:51:40 stop charging <-

home battery immediately takes over all available power for charging
[site ] DEBUG 2021/12/31 11:51:50 battery power: -3470W

Switching back to PV mode after that will not start loading the vehicle again, prioritySoC is respected in that case
</pre></details>

Man sieht auch schön die  falschdetektierung von 3p um 11:45:10 wo die Werte in der influx dann nicht mehr der Realität entsprechen - am Plot des vorgeschalteten Zählers vor der Wallbox den ich auch erfasse (S0 alle 60s) sieht man das die Ladeleistung mehr oder weniger konstant bleibt.
[image]

premultiply, 2022-01-01:
Nach einmal drüber schlafen bin ich zu der Erkenntnis gelangt dass die "automatische Phasenschätzung" ohne Zähler überhaupt gar keinen Sinn ergibt.
Da gaukeln sich zwei freie Interpolationen gegenseitig irgendwelche Sachen vor. Mindestens in dem Fall können wir uns nur auf die Angabe des Nutzers mittels `phases:` verlassen. Diese war hier auch korrekt und hat auch wie erwartbar zur langsamen Übernahme des Speichers geführt bei >minSoc.
Beschleunigen und sicherstellen könnte man dies mittels residualPower und einer kleinen Mindesteinspeisung.

/cc @andig

andig, 2022-01-01:
Die Phasen haben damit nix zu tun- das ist ein anderes Thema. Das Problem hier hatten wir schon mehrfach in unterschiedlichen Facetten.

Die Lösung ist ganz einfach: wir stellen (site) nur die aktuelle Ladeleistung der Batterie zur Verfügung und auch nur falls die lädt. Da der Ladevorgang im Auto aber schon läuft (loadpoint) wird die Leistung nicht für die Batterie verfügbar gemacht. 

Lösungsansätze:
- Ladevorgang nach minSoC immer unterbrechen- dann kommt die Batterie wieder zum Vorrang (aber auch nur 1x)
- Laden bei Batterie unter prioritySoC immer unterbinden- dann gelangt aber auch Restleistung nicht ins Auto falls es welche gibt

Da wir die Ladeleistung der Batterie(n) nicht kennen wird es in jedem Fall bei einer Heuristik bleiben.

premultiply, 2022-01-01:
> Die Lösung ist ganz einfach: wir stellen (site) nur die aktuelle Ladeleistung der Batterie zur Verfügung und auch nur falls die lädt.

Müssten wir nicht dann nicht auch umgekehrt Batterieentladung unterhalb prioritySoc wie Netzbezug behandeln?

andig, 2022-01-01:
Spielt keine Rolle. Falls Auto genau die Batterie auf Null ausbalancieren würde gäbe es auch keine Batterieladung. Das Problem ist die Autoladung die dazu führt, dass die Batterie gar nicht erst lädt.

premultiply, 2022-01-01:
So genau kann das doch eigentlich gar nicht sein, oder? Selbst 0,1A sind ja immer noch ein paar Watt... 🧐
Vor allem ist der BWR ja viel schneller.
Irgendwie ist das merkwürdig.

Die Batterieentladung explizit als Netzbezug zu behandeln würde zumindest unmittelbar eine Reduktion des Ladestroms nach sich ziehen wenn sie auftritt.

F-L-0, 2022-01-01:
> Lösungsansätze:

>    Ladevorgang nach minSoC immer unterbrechen- dann kommt die Batterie wieder zum Vorrang (aber auch nur 1x)
>   Laden bei Batterie unter prioritySoC immer unterbinden- dann gelangt aber auch Restleistung nicht ins Auto falls es welche gibt

Beides würde in meinem Fall das Problem lösen, wobei ich das kurze Unterbrechen nach minSoC besser finde - es stellt den Normalzustand her der funktioniert. Das generelle Unterbinden ist für andere Anwender mit größerer PV vermutlich nicht wünschenswert.

> Da wir die Ladeleistung der Batterie(n) nicht kennen wird es in jedem Fall bei einer Heuristik bleiben.

Könnte man nicht genau da ansetzen? Also einen Parameter dafür einbauen:
prioritySoC: 70 # give home battery priority up to this soc (empty to disable)
prioritySoCreservedPower:  20 # reserved current in Ampere for home battery when SoC is below prioritySoC. When not consumed by home battery this goes to grid

Das würde auch andere Szenarien ermöglichen, wie zB paralleles laden des Autos und eine Mindestladung an die home battery oder laden der HomeBattery im verlustoptimierten Bereich des Wechselrichters (beim SI6.0H wären das glaub ich 1.1kw).

andig, 2022-01-01:
> So genau kann das doch eigentlich gar nicht sein, oder? Selbst 0,1A sind ja immer noch ein paar Watt... 🧐 Vor allem ist der BWR ja viel schneller. Irgendwie ist das merkwürdig.
>
> Die [...]

Du hast das Problem nicht verstanden. Der BWR bekommt schon die Leistung. Aber gegenüber dem Ladepunkt ziehen wir auch genau nur diese Leistung ab. Der Rest, also 1,4kW beim MinPV 1p, bleibt beim Ladepunkt da der BWR die gar nicht erst sieht!

premultiply, 2022-01-01:
Die Sache mit der reservierten Leistung für den BWR wäre auch nicht ganz unproblematisch, da dies ja auch nur ein Maximalwert ist. Der BWR kann auch gar nix wollen weil Akku (fast) voll oder verordnetet Ladestop vom HEMS usw.

Aber besonders merkwürdig finde ich hier dass es wenn manuell ausgeführt (Sofort->PV bei < prioritySoc) wie erwartet funktionieren soll. Zumindest habe ich das so verstanden. @F-L-0

premultiply, 2022-01-01:
@andig Doch, ich habe das sehr gut verstanden aber ich wundere mich, dass die Ladestromvorgabe so präzise soll sein, dass der BWR kein Leistungsoffset sieht (ein paar Watt reichen) was er sich dann nach und nach und je nach Offsetgröße einverleiben und steigern kann. Das funktioniert bei mir ja (manuell) auch. Schon durch die 0,1A-Regelschritte hier und die Abrundung (uint16) bleibt eigentlich immer eine kleine Leistung als Regeloffset ungenutzt. Das kann alleine kaum so genau sein dass sich der schnelle BWR und der viel trägere Ladepunkt da exakt über längere Zeit die Waage halten.

F-L-0, 2022-01-01:
>  Das kann alleine kaum so genau sein dass sich der schnelle BWR und der viel trägere Ladepunkt da exakt über längere Zeit die Waage halten. Jedenfalls würde es mich SEHR wundern.

Genau so ist es aber, evcc regelt das so ein (siehe Verlauf chargePower.mean). Kannst du dir bitte oben im Grafana nochmal die Kurve von batteryPower.mean zwischen 11:43:40 (minSoC is reached, chargePower reduced) und 11:51:40 ([lp-1 ] INFO 2021/12/31 11:51:40 stop charging <-) anschauen?
Der Battery-WR regelt um den 0-Punkt herum.

Ich denke was Andi sagen will, wenn bereits ein Ladevorgang läuft wird das was der WR der Battery bekommt als Erzeugung mit eingerechnet und dann im nächsten Regelzyklus von evcc wieder dem lp zugeschlagen, oder?

> Aber besonders merkwürdig finde ich hier dass es wenn manuell ausgeführt (Sofort->PV bei < prioritySoc) wie erwartet funktionieren soll. Zumindest habe ich das so verstanden.

Das habe ich nicht getestet - evcc steht beim Szenario oben wie in der config angegeben im Modus "Nur PV".

Ich versuchs nochmal anders zu formulieren:
- Wenn ich ein Auto mit Soc > minSoC anschließe (also keine unmittelbare Ladung beginnt) und der SoC der Hausbatterie < prioritySoC ist dann funktioniert alles wie gedacht (Hausbatterie wird bis prioritySoC geladen, anschließend startet evcc den Ladevorgang mit der Leistung die vorher in die Hausbatterie geladen wurde, die Hausbatterie pendelt um +/- 100W)
- Wenn ich ein Auto mit Soc < minSoC anschließe (also eine unmittelbare Ladung mit maximaler Leistung beginnt) und der SoC der Hausbatterie < prioritySoC ist dann wird nachdem das Auto auf minSoC geladen ist der Ladevorgang mit der überschüssigen PV-Leistung fortgesetzt, die Hausbatterie geht aber leer aus und pendelt um +/- 100W).
Ich hätte erwartet das in diesem Fall von evcc der Ladevorgang nach erreichen von minSoC angehalten wird und ab diesem Zeitpunkt alles was übrig ist in die Hausbatterie geht.
Es müsste damit zu tun haben das bereits ein Ladevorgang lief und dieser nicht gestoppt wird.

premultiply, 2022-01-01:
Spätestens auf längere Sicht schalten ja auch Verbraucher im Haus an und aus. Auf die reagiert auch zuerst der BWR weil er schneller ist. Mindestens mit jedem abgeschalteten Verbraucher (z. B. Kühlschrank) müsste also der Anteil des BWR größer werden.

Wenn evcc aber nicht explizit auf Batterieentladung sondern nur auf die weiterhin ausgeglichene Netzbilanz reagiert geht es auch in die andere Richtung. Der BWR gibt auch als Erster zu seinen Lasten nach und speichert aus. Das trägt dann somit unbeabsichtigt auch zum Verharren der Verhältnisse bei.
Das mag jetzt vielleicht in der Masse der Messwertsamples vernachlässigbar klingen sorgt aber wohl letztlich für diesen Verharrungszustand.
-> Entladung muss daher zum Netzbezug addiert werden wenn der Speicher Priorität hat.
Das gucke ich mir heute Abend nochmal im Code an.

Davon abgesehen folgende Optimierungen bitte mal schrittweise testen:
SHM-Messintervall auf 200ms stellen (via Sunny Portal).
residualPower auf z. B. 50 Watt Export stellen.

Ändert das jeweils etwas am Verhalten?

F-L-0, 2022-01-01:
> Davon abgesehen folgende Optimierungen bitte mal schrittweise testen: SHM-Messintervall auf 200ms stellen (via Sunny Portal). residualPower auf z. B. 50 Watt Export stellen.
> Ändert das jeweils etwas am Verhalten?

SHM Messintervall kenne ich nicht und finde ich im Portal auch nicht - ist das ggf. eine Option die beim SHM 1.0BT nicht existiert? Bei mir sind SHM und EM zwei getrennte Geräte, die Multicast-Werte die das EM 1/s schickt kann ich nicht beeinflussen, auch nicht an der Weboberfläche des EM selbst.

residualPower teste ich sobald es die Bedingungen (Sonne) zulassen.

andig, 2022-01-01:
> @andig Doch, ich habe das sehr gut verstanden aber ich wundere mich, dass die Ladestromvorgabe so präzise soll sein, dass der BWR kein Leistungsoffset sieht

Das tut er- aber eben nur das bisschen Rest! Die komplette Ladeleistung des Autos sieht er nicht und genau das ist das Problem. Du suchst den Fehler an der falschen Stelle ;)

premultiply, 2022-01-01:
Ich hab eben bemerkt dass ich bei mir `residualPower: 100` gesetzt hab. War mir gar nicht mehr bewusst.
Wahrscheinlich funktioniert das deshalb bei mir so problemlos.

andig, 2022-01-08:
Also was wollen wir hier machen: https://github.com/evcc-io/evcc/issues/2170#issuecomment-1003529080?

F-L-0, 2022-01-08:
Hallo Andi, ich würde gerne noch ein paar Szenarien durchtesten.
- tritt neben dem oben beschrieben Verhalten das selbe Problem auf wenn man aus dem Sofort in den Nur PV Modus wechselt
- hilft der Parameter residualPower um das langsame Umschwenken von Auto laden auf Hausakku laden zu forcieren - in beiden Fällen

Bis heute hatte ich jedoch keine Testbedingungen - ich brauch Sonne :-)

Bezüglich der konkreten Frage:
>Lösungsansätze:
Ladevorgang nach minSoC immer unterbrechen- dann kommt die Batterie wieder zum Vorrang (aber auch nur 1x)
Laden bei Batterie unter prioritySoC immer unterbinden- dann gelangt aber auch Restleistung nicht ins Auto falls es welche gibt

Bei der zweiten Lösung wären die mit großer PV Anlage sicherlich nicht glücklich. Bei der ersten Lösung, die hört sich gut an, das Unterbrechen muss allerdings so lange sein das im Zweifelsfall der Batteriewechselrichter genug Zeit hat aufzuwachen und mit dem Laden zu beginnen (bei meinem SI6.0H so um die 15s denke ich) - ansonsten kommt man aus dem Problem nicht raus.
Mein Favorit wäre aktuell residualPower auf 100W - wenn das klappt löst es das Problem und zusätzlich sehe ich gleich noch einen Vorteil:
Im PV Modus schwankt aktuell der Batteriewechselrichter bei +/- 100W - ständige wenn auch kleine Wechsel zwischen Laden und Entladen sind nicht gut für die Batterie, mit residualPower kann man das auf +0W bis +200W verschieben (Offset)

Können wir das Issue noch ein paar Tage "On Hold" setzen? Sobald ich was hab dokumentier ich es hier.

andig, 2022-01-08:
Kein Problem, warten wir 👍🏻

F-L-0, 2022-01-14:
Ich konnte heute einiges Testen - anbei mal die Zusammenfassung der Ergebnisse. Falls Logs oder Grafana Graphen notwendig kann ich die nachliefern. Als Basis für die Tests gilt die oben angehängte evcc.yaml mit evcc 0.81, geänderte Parameter stehen bei jedem Test.

Bei Test 1 und 2 war der Ladestand des Akkus < prioritySoC, der zur Verfügung stehende Überschuss aus PV Erzeugung war durchgehend bei allen Tests größer als die minimale Ladeleistung (6A@2P). Andere Verbraucher im Haus waren weitgehend "ruhiggestellt" um möglichst wenig Einfluss von "außen" zu haben.

1) evcc im Modus "Nur PV", SoC vom Auto ist kleiner als minSoC, Auto Laden bis minSoC, danach sollte der Ladevorgang anhalten und die Hausbatterie geladen werden
Parameter:
  residualPower: 100
  disable: # pv mode disable behavior
    delay: 2m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)

Ergebnis:
minSoC wird erreicht, ab dann ganz langsames Überschwenken (immer so um die 100W pro cycle) bis mit 6A Auto geladen wird, der freie Rest geht in den Hausakku, pv mode disable wird nicht getriggert (da residualPower kleiner 200W)

2) wie 1), aber geänderte Parameter:
  residualPower: 100
  disable: # pv mode disable behavior
    delay: 2m # threshold must be exceeded for this long
    threshold: 50 # maximum import power (W)

Ergebnis:
ganz langsames Überschwenken (immer so um die 100W pro cycle) bis mit 6A geladen wird, pv mode disable wird getriggert, danach läd der Hausakku mit Vollgas und evcc begibt sich in Wartestellung (solange prioritySoC nicht erreicht ist).
Aber: Die 50W Unterschied zwischen residualPower und threshold sind sehr knapp, bei kleineren Schwankungen im Hausverbrauch kann der Timer immer wieder kurz unterbrochen werden und läuft dann von vorne los.
Insbesondere Lasten wie ein Induktionsherd (schnelles schwanken +/- 400W) würde den Timer immer wieder unterbrechen.

3) evcc im Modus "Sofort", nach einer Weile umschalten auf der GUI auf "Nur PV". Es sollte der Ladevorgang anhalten und die Hausbatterie geladen werden
Parameter:
  residualPower: 0
  disable: # pv mode disable behavior
    delay: 2m # threshold must be exceeded for this long
    threshold: 100 # maximum import power (W)

Ergebnis:
Wie oben im Ticket beschrieben, kein Überschwenken auf die Hausbatterie, evcc läd weiter das Auto mit dem gesamten verfügbaren PV Überschuss.
--> Es ist also unabhängig ob man aus dem Sofort-Modus zum "Nur PV" Modus wechselt oder innerhalb des "Nur PV" Modus aus dem Zwangsladen bis minSoC in den eigentlichen PV Modus zurückkommt wenn minSoC erreicht wurde. Das Verhalten war bei meinen Tests identisch.

4) ähnlich 2), mit dem Unterschied nicht vom "minSoc" Modus in "Nur PV" zu wechseln sondern von "Sofort" in "Nur PV".
Parameter:
  residualPower: 200
  disable: # pv mode disable behavior
    delay: 2m # threshold must be exceeded for this long
    threshold: 100 # maximum import power (W)

Auch hier identisches Ergebnis:
ganz langsames Überschwenken (immer so um die 200W pro cycle) bis mit 6A geladen wird, pv mode disable wird getriggert, danach läd der Hausakku mit Vollgas und evcc begibt sich in Wartestellung (solange prioritySoC nicht erreicht ist).

Alles in allem würde ich dennoch für Lösungsvorschlag a) plädieren. Das Unterbrechen der Ladung bei einem Wechsel von "Sofort" nach "Nur PV" muss man nicht automatisiert implementieren - da bin ich eh schon an der GUI und kann das selber machen.
> Ladevorgang nach minSoC immer unterbrechen- dann kommt die Batterie wieder zum Vorrang (aber auch nur 1x)

Plan B wäre nur mit passenden residualPower und threshold Werten zu arbeiten, müsste ich mal schauen wie es sich im Lauf der Zeit so verhält.

Zusätzlich könnte man drüber nachdenken in der Doku die Wirkungsweise von residualPower im Zusammenhang mit einer Hausbatterie etwas ausführlicher zu beschreiben, da es ja unabhängig vom "umschwenken" ja auch noch andere Vorteile hat (vermindern von ständigem Wechsel zwischen Laden/Entladen). Könnte ich mich dran versuchen - bin schon auf euer Feedback gespannt.

andig, 2022-01-14:
D.h. Du plädierst für manuelle Lösung?

F-L-0, 2022-01-14:
Jein. Für den Fall das ich von Sofort auf Nur PV umschalte ist das kein Ding da dann in der GUI von Sofort -> Stop -> Nur PV zu machen, ich weiß ja jetzt wie es sich verhält.
Der Zustand das minSoC erreicht wird und zum "Nur PV" zurückgekehrt wird ist im Normalfall ja "unattended" - hier wäre eine automatische Lösung (das von dir vorgeschlagene einmalige Unterbrechen) die beste weil zuverlässigste Lösung. ResidualPower und Threshold allein und ohne Codeänderung können auch funktionieren, ist aber abhängig davon was grad im Haus passiert.

Der Idealzustand der glaub ich alles automatisch erschlägt wäre das bei jedem Zustandswechsel zwischen minSoC Laden, Nur PV, Sofort, Min + PV der Ladevorgang kurz unterbrochen wird - natürlich alles nur bei SoC Hausbatterie < prioritySoC. Dann hat die Hausbatterie in jedem Fall einen Cycle Zeit sich alles was frei ist zu "schnappen" - das regelt auch große Sprünge extrem schnell, beim SunnyIsland ca 2s.
Hört sich das sinnvoll an?

andig, 2022-01-14:
Das kommt vmtl drauf an, ob es diese universelle Zeitkonstante gibt. Schaffen das alle WR in 2s? Was ist der richtige Wert?

premultiply, 2022-01-14:
Ich halte nichts von einer erzwungenen Ladeunterbrechung bei Modusumschaltung.
Ein Alternativvorschlag wäre den internen Default (nicht gesetzt) für residualPower auf 115 zu setzen.

Abschaltende Verbraucher sorgen übrigens für einen schnelleren Ausgleich!

andig, 2022-01-14:
Wir könnten auch mit jedem Ausgleich dem Charger 5..50W wegnehmen und virtuell der Batterie zur Verfügung stellen. Wirkt analog `residualPower`, aber nur wenn ein Akku im Spiel ist.

F-L-0, 2022-01-14:
Hallo @premultiply 
der Alternativvorschlag bringt nur etwas wenn über den gesamten Zeitraum des PV Timers der threshold überschritten wird - ansonsten wird der Timer immer wieder zurückgesetzt. Der Ladevorgang wird dann effektiv in einer Art "Min +PV" fortgesetzt und kommt da nie raus.
Eine generelle erzwungene Ladeunterbrechung bei Modusumschaltung halte ich nicht für sinnvoll, nur und ausschließlich wenn der SoC der Battery kleiner prioritySoC ist.

premultiply, 2022-01-14:
Keine Ahnung was der interne Default von `threshold` ist aber ohne läuft das eigentlich prima.
Der Speicher holt sich langsam seine Anteile zurück und dann schaltet nach Timer ggf. die Ladung ganz ab.

F-L-0, 2022-01-14:
> Keine Ahnung was der interne Default von `threshold` ist aber ohne läuft das eigentlich prima.
> Der Speicher holt sich langsam seine Anteile zurück und dann schaltet nach Timer ggf. die Ladung ganz ab.

Ich hatte gehofft mit meinen Tests oben gezeigt zu haben das dies nicht zuverlässig der Fall ist. Hast du zu einem Fall bei dem es bei dir offensichtlich klappt und im PV Modus nach einer Ladung bis minSoC die Ladung abgeschalten wird weil prioritySoC nicht erreicht ist zufällig ein Log? Dann würde ich mich hinsetzen und versuchen den Unterschied zu finden - irgendwas muss ja anders sein wenn es bei dir geht.

andig, 2022-02-02:
Keine neuen Erkenntnisse- closing.

closed 2022-02-02 by andig