evcc-io/evcc #2177: Phases detection wrong

Hoelli4C, 2022-01-01:
**Describe the bug**
I have set phases to p1 and my WARP Charger is indeed connected via 1 Phase. For the first few minutes everything works, after some time EVCC autodetects 2 and later 3 phases (see output below). 

**Expected behavior**
I would expect EVCC to keep my 1p setting.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.77 (bac4e07)

```
<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
# sponsortoken:

# log settings
log: error
levels:
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
- name: SHM2            #SMA Home Manager 2.0
  type: sma
  uri: 192.168.30.50

- name: WR-PV           #SMA Sunny TriPower 20000TL
  type: sma
  uri: 192.168.30.51
  password: xxx

- name: WR-BAT          # SMA Sunny Boy Storage 3.7
  type: sma
  uri: 192.168.30.52
  password: xxx

# Home Energy Management System
hems:
  type: shm
  allowcontrol: false

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/andig/evcc-config#chargers
chargers:
- name: warp        # Konfigurationsname der Wallbox
  type: warp        # Typ der Wallbox
  broker: localhost # Hostname oder Adresse des MQTT-Brokers
  topic: warp/U7X   # Der Topic-Präfix
  usemeter: false    # true für WARP Charger Pro, false für Smart
  timeout: 30s

# Vehicles
vehicles:
- name: ID4
  type: id
  title: ID4 # display name for UI
  capacity: 77 # kWh
  user: thomas.henneberger@gmail.com # user
  password: xxx# password
  vin: xxx # optional

[... 103 lines cut ...]
Jan  1 12:56:15 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:15 vehicle soc: 71%
Jan  1 12:56:15 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:15 vehicle range: 314km
Jan  1 12:56:15 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:15 max charge current: 14.6A = 14.8A + -0.165A (76W @ 2p)
Jan  1 12:56:15 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:15 pv max charge current: 14.6A
Jan  1 12:56:15 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:15 max charge current: 14.6A
Jan  1 12:56:15 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:15 charge power: 6726W
Jan  1 12:56:15 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:15 detected phases: 3p (14.6A @ 6726W)
Jan  1 12:56:25 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:25 charge power: 6726W
Jan  1 12:56:25 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:25 detected phases: 3p (14.6A @ 6726W)
Jan  1 12:56:25 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:25 charger status: C
Jan  1 12:56:25 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:25 vehicle soc: 71%
Jan  1 12:56:25 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:25 vehicle range: 314km
Jan  1 12:56:25 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:25 max charge current: 14.6A = 14.6A + -0.00203A (1W @ 3p)
Jan  1 12:56:25 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:25 pv max charge current: 14.6A
Jan  1 12:56:25 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:25 max charge current: 14.6A
Jan  1 12:56:25 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:25 charge power: 10087W
Jan  1 12:56:25 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:25 detected phases: 3p (14.6A @ 10087W)
Jan  1 12:56:35 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:35 charge power: 10087W
Jan  1 12:56:35 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:35 detected phases: 3p (14.6A @ 10087W)
Jan  1 12:56:35 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:35 charger status: C
Jan  1 12:56:35 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:35 vehicle soc: 71%
Jan  1 12:56:35 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:35 vehicle range: 314km
Jan  1 12:56:35 raspi3 evcc[2363]: [lp-1  ] DEBUG 2022/01/01 12:56:35 max charge current: 14.6A = 14.6A + 0.00623A (-4W @ 3p)

</pre></details>

premultiply, 2022-01-01:
Auch in #2170 vorhanden.
-> Erkennungsautomatik funktioniert ohne Zähler am LP nicht.

Hoelli4C, 2022-01-01:
sieht so aus als würde zum Zeitpunkt des Fehler der Verbrauch im Haus auf die Ladeleistung aufgeschlagen und dadurch stimmen die Werte nicht mehr. Im Screenshot sieht man Verbrauch 0 (es laufen defintiv andere Verbraucher im Haus):

[image]

andig, 2022-01-01:
> Auch in #2170 vorhanden.
> -> Erkennungsautomatik funktioniert ohne Zähler am LP nicht.

Wenn ich bei 1p und 6A plötzlich 3,7kW sehe kann es nur 3p sein. Könntest Du konkreter werden, _was_ hier nicht funktioniert?

Hoelli4C, 2022-01-01:
Wo sind 3.7kw bei 6A? Die Ampere schwanken in meinem Log zwischen 14 und 16A.

Es ist defintiv so dass die WB nur per einer Phase (16A am DIP bzw. an der Sicherung) angebunden ist, da gehen nicht mehr als 3800W drüber. Es sieht so aus als würde der übrige Hausverbrauch über die Zeit dem Ladepunkt zugeschlagen werden und dann "sieht" EVCC mehr als 3800W per 16A und geht zuerrst auf 2P und dann später auf 3P.

`12:55:25 detected phases: 2p (15.1A @ 3466W)`
`12:56:15 detected phases: 3p (14.6A @ 6726W)`

premultiply, 2022-01-01:
@andig Ohne Zähler ist doch bereits die Ladeleistung fake. Da jetzt noch eine Phasennutzung draus abzuleiten ist doch ziemlich sinnfrei. Hier kann nur fest die `phases:` Angabe in der Konfig weiterhelfen.

andig, 2022-01-01:
Ich dachte eher an den Fall Leistungsmessung ohne Ströme. Das könnten wir natürlich noch differenzieren.

premultiply, 2022-01-01:
Ja das wäre wohl sehr sinnvoll weil sonst kommt da nur Quatsch raus.

Und auch den Erkennungsalgorithmus würde ich gerne noch verfeinern. Der muss auch den zuvor erkannten phases-Wert berücksichtigen können bzw. diesen nicht ändern für Leistungsbereiche wo es nicht eindeutig ist.

andig, 2022-01-01:
PR welcome

andig, 2022-01-01:
Also hier sind wir fertig?

premultiply, 2022-01-01:
Bis zum Fix nicht. Oder hab ich da was übersehen?

andig, 2022-01-01:
Der einfache Fix wäre das ohne Zähler nixht zu machen. Wär das besser? Ansonsten work as wxpected für mich!

premultiply, 2022-01-01:
Ja, das wäre besser. Wenn du das schnell zu fixen ist mach das bitte gerne und dann hier zu.

Die Optimierung der Phasenerkennung anhand der (gemessenen) Leistung kommt später und separat.
Da muss ich auch erst noch ein bisschen drüber sinnieren.

premultiply, 2022-01-02:
Genug sinniert: Die Phasennutzungsschätzung nur anhand der gemessenen(!) Summenwirkleistung kann man nicht wirklich viel besser machen.
@andig Das kann so bleiben wenn der Fall "Keine Wirkleistungsmessung und keine Phasenströme" vollständig ausgeklammert wird.

andig, 2022-01-02:
Ich verstehe die Schlussfolgerung nicht, deshalb nochmal zurück zum Problem. Die Ursache ist ganz einfach wenn man ins Log schaut:

W | A | phases
-- | -- | --
3497 | 15,2 | 1,00028604
3466 | 15,1 | 0,99798445

Das Problem ist schlicht die kürzlich eingeführte `ceil` Funktion statt des sauberen Rundens.

premultiply, 2022-01-02:
Nein, denn Achtung: Die Wirkleistung wird hier nicht gemessen sondern selbst geschätzt! Der Rest ist ein Float-Rundungsfehler.

Ceil ist korrekt. Es gibt keine 1,irgendwas Phasen sondern dann nur 2.

Bei gemessenen(!) Werten liegt die Wirkleistung immer unter der (Schein)Leistungsvorgange durch den max. Phasenstrom.

premultiply, 2022-01-02:
Hier muss einfach nur dafür gesorgt werden dann diese Funktion nur mit gemessenen(!) Leistungen gefüttert wird.

andig, 2022-01-02:
Mittels https://github.com/evcc-io/evcc/commit/13dac30def47a7666969fe090c2f4cf7a1a08133 behoben. 

@premultiply weiter in https://github.com/evcc-io/evcc/issues/2146

closed 2022-01-02 by andig