evcc-io/evcc #2180: go-e cloud v3 not working as expected

sebnaf, 2022-01-01:
Firmware 052.1
activated HTTP API on api.v3.go-e.io in app
entered an api key

config:
`pi@evcc:~ $ cat /etc/evcc.yaml

chargers:
- name: elektronenpumpe
  type: go-e
  # uri: http://10.10.20.22
  token: HdGAjWISCHNAPP0VBAQoA # go-e cloud API token
  cache: 10s # go-e cloud API cache duration
`

result: 
`pi@evcc:~ $ sudo evcc -l debug -c /etc/evcc.yaml charger
[main  ] INFO 2022/01/01 15:12:33 evcc 0.76 (07c559b)
[main  ] INFO 2022/01/01 15:12:33 using config file /etc/evcc.yaml
[mqtt  ] INFO 2022/01/01 15:12:33 connecting evcc-1478910102 at tcp://10.0.55.22:1883
[mqtt  ] DEBUG 2022/01/01 15:12:33 tcp://10.0.55.22:1883 connected
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x20 pc=0xa1c5cc]

goroutine 1 [running]:
github.com/evcc-io/evcc/charger.(*GoE).Identify(0x41eef48)
	/home/runner/work/evcc/evcc/charger/go-e.go:187 +0x40
github.com/evcc-io/evcc/cmd.(*dumper).Dump(0x455fbdc, {0x40b7860, 0xf}, {0x1083590, 0x41eef48})
	/home/runner/work/evcc/evcc/cmd/dumper.go:187 +0x1460
github.com/evcc-io/evcc/cmd.(*dumper).DumpWithHeader(0x4703bdc, {0x40b7860, 0xf}, {0x1083590, 0x41eef48})
	/home/runner/work/evcc/evcc/cmd/dumper.go:32 +0x74
github.com/evcc-io/evcc/cmd.runCharger(0x20a11b8, {0x44db280, 0x0, 0x4})
	/home/runner/work/evcc/evcc/cmd/charger.go:49 +0x414
github.com/spf13/cobra.(*Command).execute(0x20a11b8, {0x44db260, 0x4, 0x4})
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.3.0/command.go:860 +0x644
github.com/spf13/cobra.(*Command).ExecuteC(0x20a1590)
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.3.0/command.go:974 +0x414
github.com/spf13/cobra.(*Command).Execute(...)
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.3.0/command.go:902
github.com/evcc-io/evcc/cmd.Execute()
	/home/runner/work/evcc/evcc/cmd/root.go:133 +0x20
main.main()
	/home/runner/work/evcc/evcc/main.go:33 +0x30`

DerAndereAndi, 2022-01-01:
Schau dass du eine neues Go-E token bekommst, du hast das hier gerade geleaked!

sebnaf, 2022-01-01:
Da stand ein "SCHNAPP" drin - ziemlich sicher. Aber danke :)

DerAndereAndi, 2022-01-01:
Leider nein. Kannst ja die Versionen anschauen, habs editiert, schau deine Fassung an. Ganz raus bekomme ich es nicht, außer wir löschen das hier und starten von vorne

sebnaf, 2022-01-01:
Dann war es in meiner ersten Version. 
Ist bereits geändert. Danke Dir!

andig, 2022-01-01:
Panic in https://github.com/evcc-io/evcc/commit/9d61d9841454100bf03e4e26f6ac9262b5dbd4c6 behoben

sebnaf, 2022-01-01:
@andig 
Ich frage mich, obwohl die URL auskommentiert ist, weiter darauf zugegriffen wird:

```
 ~/Documents/repositories/evcc  on master  ./evcc -c ../evcc-config/evccdev.yaml -l debug charger                                1 ✘  took 8s  at 18:18:35
[main  ] INFO 2022/01/01 18:18:42 evcc 0.77 (fb1d6f55)
[main  ] INFO 2022/01/01 18:18:42 using config file ../evcc-config/evccdev.yaml
[mqtt  ] INFO 2022/01/01 18:18:42 connecting evcc-1896803973 at tcp://10.0.55.22:1883
[mqtt  ] DEBUG 2022/01/01 18:18:43 tcp://10.0.55.22:1883 connected
Power:          wrong token
Current L1..L3: wrong token
Charge status:  wrong token
Enabled:        wrong token
Charged:        wrong token
Identifier:     wrong token
```

sebnaf, 2022-01-01:
Ich glaube, das ist gar nicht die v2 API - `[go-e  ] TRACE 2022/01/01 18:21:53 GET https://api.go-e.co/api_status?token=***`

Doku: https://github.com/goecharger/go-eCharger-API-v2/blob/main/cloudapi-de.md

andig, 2022-01-01:
Was ist die Frage? Log trace? Ich bin gar nicht aicher, ob wir bei der Cloud Api überhaupt was geändert haben. Kann sein dass die immer noch v1 ist. Gibts da überhaupt eine v2?

sebnaf, 2022-01-01:
Die Frage ist: warum funktioniert die Cloud API nicht, wenn wie dokumentiert eingerichtet.

Wir ich nun inzwischen weiss: es gibt eine API v2, ohne poll limits, aktualisiert jede Sekunde. Link im comment weiter oben.

Bitte: Integration der neuen API v2 wenn möglich.

Danke

andig, 2022-01-01:
Aber warum sollte die v1 API nicht mehr tun?

    Power:          wrong token

sebnaf, 2022-01-01:
Deprecated. Man kann laut go-e keine Token mehr für die Api v1 erzeugen.
Einen Trace erzeuge ich morgen. Zusätzlich dazu kommt, dass wenn Du versuchst, die Cloud-API hier zu nutzen trotzdem der lokale Endpoint per IP angesprochen wird.

andig, 2022-01-01:
Warum hängst Du an der Cloud api?

sebnaf, 2022-01-02:
> Warum hängst Du an der Cloud api?

Hi, tue ich nicht. Ich hatte nur die Doku gesehen, dachte mir "sollte ich mal testen" und festgestellt, dass es aus diversen Gründen nicht funktioniert. Somit würde ich raten: weg damit aus Code und Doku ;)

andig, 2022-01-03:
Solange das Cloud api v1 noch tit würde ich es erstmal nicht ausbauen wollen.

closed 2022-01-03 by andig