evcc-io/evcc #2183: BMW API Issue, maybe JFI

Hofyyy, 2022-01-01:
**Describe the bug**
Bei mir hat heute der Ladevorgang nicht gestartet, weil es Ärger mit dem BMW API gab.
Vielleicht hatte wer anderes ja ähnliche Probleme. ich habe auf jeden Fall mal Logs gezogen.

1. Syslog vom Even -> First TIme Seen.txt
2. Dann ein Clean Reboot vom Linux Server, Fehler blieb bestehen -> Clean_reboot.txt
3. Dann EVCC neustart mit TRACE aktiv. -> TraceActiv.txt

Fehlermeldungen:
```
Jan  1 17:59:05 raspberrypi evcc[9646]: [lp-1  ] DEBUG 2022/01/01 17:59:05 charger status: B
Jan  1 17:59:05 raspberrypi evcc[9646]: [lp-1  ] DEBUG 2022/01/01 17:59:05 charge start energy: 745kWh
Jan  1 17:59:05 raspberrypi evcc[9646]: [lp-1  ] INFO 2022/01/01 17:59:05 car connected
Jan  1 17:59:05 raspberrypi evcc[9646]: [lp-1  ] DEBUG 2022/01/01 17:59:05 vehicle api refresh
Jan  1 17:59:05 raspberrypi evcc[9646]: [lp-1  ] DEBUG 2022/01/01 17:59:05 pv action: inactive
Jan  1 17:59:05 raspberrypi evcc[9646]: [lp-1  ] ERROR 2022/01/01 17:59:05 vehicle soc: Get "https://cocoapi.bmwgroup.com/eadrax-vcs/v1/vehicles?apptimezone=60&appDateTime=1641056345": invalid_client:Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method)
```

sudo reboot

```
Jan  1 18:05:13 raspberrypi evcc[515]: [lp-1  ] INFO 2022/01/01 18:05:13 vehicle updated: unknown -> unavailable
Jan  1 18:05:13 raspberrypi evcc[515]: [lp-1  ] INFO 2022/01/01 18:05:13 loadpoint 1:
Jan  1 18:05:13 raspberrypi evcc[515]: [lp-1  ] INFO 2022/01/01 18:05:13   mode:      now
Jan  1 18:05:13 raspberrypi evcc[515]: [lp-1  ] INFO 2022/01/01 18:05:13   charger:   power ✓ energy ✓ currents ✓ timer ✓
Jan  1 18:05:13 raspberrypi evcc[515]: [lp-1  ] INFO 2022/01/01 18:05:13   meters:    charge ✓
Jan  1 18:05:13 raspberrypi evcc[515]: [lp-1  ] INFO 2022/01/01 18:05:13     charge:  power ✓ energy ✓ currents ✓
Jan  1 18:05:13 raspberrypi evcc[515]: [lp-1  ] INFO 2022/01/01 18:05:13   vehicles:  ✓
Jan  1 18:05:13 raspberrypi evcc[515]: [lp-1  ] INFO 2022/01/01 18:05:13     car 0:   range ✗ finish ✗ status ✗ climate ✗
Jan  1 18:05:14 raspberrypi evcc[515]: [lp-1  ] DEBUG 2022/01/01 18:05:14 charge power: 0W
Jan  1 18:05:14 raspberrypi evcc[515]: [lp-1  ] DEBUG 2022/01/01 18:05:14 charge currents: [0 0 0]A
Jan  1 18:05:14 raspberrypi evcc[515]: [lp-1  ] DEBUG 2022/01/01 18:05:14 charger status: B
Jan  1 18:05:14 raspberrypi evcc[515]: [lp-1  ] ERROR 2022/01/01 18:05:14 vehicle soc: vehicle not available: cannot create vehicle 'bmw': invalid_client:Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method)
```

Trace on

```
[main  ] INFO 2022/01/01 18:09:40 evcc 0.77 (bac4e07)
[main  ] INFO 2022/01/01 18:09:40 using config file /etc/evcc.yaml
[main  ] INFO 2022/01/01 18:09:40 listening at 0.0.0.0:7070
[bmw   ] TRACE 2022/01/01 18:09:40 POST https://customer.bmwgroup.com/gcdm/oauth/authenticate
[bmw   ] TRACE 2022/01/01 18:09:41 
POST /gcdm/oauth/authenticate HTTP/1.1
Host: customer.bmwgroup.com
User-Agent: Go-http-client/1.1
Content-Length: 255
Content-Type: application/x-www-form-urlencoded
Accept-Encoding: gzip

client_id=31c357a0-7a1d-4590-aa99-33b97244d048&nonce=login_nonce&password=***&redirect_uri=com.bmw.connected%3A%2F%2Foauth&response_type=token&scope=authenticate_user+vehicle_data+remote_services&username=***

HTTP/1.1 401 Unauthorized
Content-Length: 190
Access-Control-Allow-Credentials: true
Access-Control-Allow-Headers: Authorization, Origin, X-c2b-Authorization, X-c2b-mTAN, X-Requested-With, X-c2b-Sender-Id, X-c2b-External-Id, Content-Type, Accept, Cache-Control, KeyId, x-dtc
Access-Control-Allow-Methods: POST, GET, OPTIONS, PUT, DELETE, HEAD
Access-Control-Allow-Origin: 
Access-Control-Max-Age: 3628800
Connection: keep-alive
Content-Type: application/json
Date: Sat, 01 Jan 2022 17:09:41 GMT
Referrer-Policy: no-referrer-when-downgrade

{"error": "invalid_client", "error_description": "Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method)"}
[cache ] DEBUG 2022/01/01 18:09:41 sponsor: Hofyyy
[cache ] DEBUG 2022/01/01 18:09:41 title: Auffahrt
[cache ] DEBUG 2022/01/01 18:09:41 minCurrent: 6
[cache ] DEBUG 2022/01/01 18:09:41 maxCurrent: 16
[cache ] DEBUG 2022/01/01 18:09:41 phases: 3
[cache ] DEBUG 2022/01/01 18:09:41 activePhases: 3
[cache ] DEBUG 2022/01/01 18:09:41 hasVehicle: true
[cache ] DEBUG 2022/01/01 18:09:41 mode: now
[cache ] DEBUG 2022/01/01 18:09:41 targetSoC: 90
[lp-1  ] INFO 2022/01/01 18:09:41 vehicle updated: unknown -> unavailable
```

[first_time_seen.txt](https://github.com/evcc-io/evcc/files/7798047/first_time_seen.txt)
[clearn_reboot.txt](https://github.com/evcc-io/evcc/files/7798048/clearn_reboot.txt)
[trace_active.txt](https://github.com/evcc-io/evcc/files/7798049/trace_active.txt)
[evcc_clear.yaml.txt](https://github.com/evcc-io/evcc/files/7798050/evcc_clear.yaml.txt)

andig, 2022-01-01:
Refs https://github.com/TA2k/ioBroker.bmw/issues/11

blenni, 2022-01-01:
Bei mir geht's aktuell zeitweise - mal wird aktualisiert, mal kommt obiger Fehler.
Betrifft bei mir bimmerconnected und evcc.

andig, 2022-01-02:
Wenns ab und an geht bastelt BMW wohl gerade.

blenni, 2022-01-03:
Seit ich's geschrieben habe, geht nichts mehr... evtl ist BMW fertig mit basteln.
Reopen?

Hofyyy, 2022-01-03:
Ja ist bei mir jetzt auch dauerhaft kaputt.
Siehe logs.
[evcc_clean.yaml.txt](https://github.com/evcc-io/evcc/files/7801426/evcc_clean.yaml.txt)
[log_trace.txt](https://github.com/evcc-io/evcc/files/7801427/log_trace.txt)

andig, 2022-01-03:
Refs https://github.com/bimmerconnected/bimmer_connected/issues/393

Clanchef, 2022-01-04:
Geht bei mir seit ein paar Minuten wieder :-)

Hofyyy, 2022-01-04:
> Geht bei mir seit ein paar Minuten wieder :-)

Hab ihn auch gerade dran gehängt und es scheint auch zu gehen.

closed 2022-01-04 by andig

rikroe, 2022-01-04:
Auch wenn das Issue schon geschlossen ist:
BMW scheint ursprünglich mal die Anforderungen für den Login erhöht zu haben (`state` und `code_challenge` wurden benötigt).
Anscheinend gab es dabei allerdings Probleme, sodass es nicht immer geklappt hat und sie das jetzt nur noch optional benötigen.

Ich teste gerade lokal ein paar Änderungen, die das Problem zwar nicht vollständig lösen, aber es minimieren: https://github.com/bimmerconnected/bimmer_connected/pull/394

andig, 2022-01-04:
@rikroe sag gerne Bescheid wenn Du eine Lösung für das initiale Login hast und weißt was hinzu zu fügen ist. Danke!

rikroe, 2022-01-07:
@andig
Ursprünglich war das Problem, dass BMW für den Login PKCE (Proof Key for Code Exchange) aktiviert hat (also `state` und `code_challenge`) benötigt. Mehr Infos hier: https://auth0.com/docs/get-started/authentication-and-authorization-flow/call-your-api-using-the-authorization-code-flow-with-pkce

Man konnte die ganze Zeit den Login auch ohne diese zusätzlichen Attribute nutzen, das ging irgendwann nicht mehr. Mit PKCE hat es für `bimmer_connected` wieder funktioniert.

Allerdings gabs es dann das Problem, dass ca. jeder dritte Aufruf auf `/authenticate` (der jeweils erste Aufruf) fehlgeschlagen ist. Das habe ich im Test jetzt gefixed, indem ich den ersten Aufruf einfach 3x mache. Auch in der MyBMW-App hatte ich das Problem.

Ggf. hat BMW aber wieder die Anforderungen etwas gelockert, sodass es nun auch ohne PKCE geht.

Hoffe das hilft Dir etwas weiter!

andig, 2022-01-07:
So einfach scheint das nicht zu sein. Ich hab hier z.B. einen i3 bei dem geht es immer, völlig ohne Änderungen/ PKCE.

Hofyyy, 2022-01-07:
Aktuell geht es bei mir auch "immer" mit meinem i3. Ich hatte nur in den 3 Tagen über Neujahr Probleme.

rikroe, 2022-01-09:
Ich glaube, dass BMW diese "härteren" Anforderungen zurück gedreht hat, weil sie selbst mit ihren eigenen Apps Probleme hatten. 

Kann durchaus sein, dass da irgendwann nochmal ein neuer Anlauf kommt, die MyBMW App nutzt PKCE jedenfalls unter Android bei jedem Login (was dann auch mit der offiziellen App zu Fehlern geführt hat).