evcc-io/evcc #2185: Powerfox grid Meter Status 401 und Status 500

Clanchef, 2022-01-02:
Seit den Updates vom 30.12./31.12. (bin mir nicht mehr 100% sicher seit wann genau) kann evcc die Grid Meter Daten von Powerfox nicht mehr auslesen. An der Konfiguration wurde nichts geändert.

Request failed with status code 500: API request failed http://192.168.178.250:7070/api/health 
updating grid meter: unexpected status: 401 

Das Powerfox Backend funktioniert, in anderen Anwendung wird der Powerfox Status korrekt ausgelesen.

Hofyyy, 2022-01-02:
Gleiches Verhalten bei mir. Bei logs oder einem remote zugang kann ich unterstützen.

andig, 2022-01-02:
Log wär super

Clanchef, 2022-01-02:
Hier das Log:

powerfox
--------
[http  ] TRACE 2022/01/02 13:56:07 GET https://backend.powerfox.energy/api/2.0/my/main/current
[http  ] TRACE 2022/01/02 13:56:07 
GET /api/2.0/my/main/current HTTP/1.1
Host: backend.powerfox.energy
User-Agent: Go-http-client/1.1
Authorization: Basic bWFydGluX3VuZ2VyQHdlYi5kZTpiYXN0aTA4NCM=
Accept-Encoding: gzip

HTTP/2.0 200 OK
Content-Length: 136
Access-Control-Allow-Headers: *
Access-Control-Allow-Methods: GET, POST, PUT, PATCH, DELETE, OPTIONS
Access-Control-Allow-Origin: *
Access-Control-Expose-Headers: Request-Context
Cache-Control: no-cache
Content-Type: application/json; charset=utf-8
Date: Sun, 02 Jan 2022 12:56:06 GMT
Expires: -1
Pragma: no-cache
Ratelimit-Limit: 7
Ratelimit-Remaining: 5
Request-Context: appId=cid-v1:9d7e501f-422d-4624-a8b7-5b542e8cb16f
Server: Microsoft-IIS/10.0
X-Aspnet-Version: 4.0.30319
X-Powered-By: ASP.NET

{"Outdated":false,"Watt":-1042.0,"Timestamp":1641128164,"A_Plus":8181766.0,"A_Plus_HT":588.0,"A_Plus_NT":8181178.0,"A_Minus":23816021.0}
Power: -1042W

andig, 2022-01-02:
@Clanchef ich sehe da kein http 401?

andig, 2022-01-02:
Ich mache mal zu bis es belastbare Informationen gibt

closed 2022-01-02 by andig

Clanchef, 2022-01-02:
Das ist das seltsame, im Container gehts nicht, auf Kommandozeile schon

andig, 2022-01-02:
Rätseln bringt nichts. Logfile bitte ;)

Clanchef, 2022-01-02:
Hier das Log aus dem Container, ich hoffe ich habe alles erwischt:

2022-01-02T13:08:24.665674618Z [http  ] TRACE 2022/01/02 14:08:24 GET https://backend.powerfox.energy/api/2.0/my/main/current
2022-01-02T13:08:24.665707079Z [cache ] DEBUG 2022/01/02 14:08:24 pvPower: 594
2022-01-02T13:08:24.697021020Z [http  ] TRACE 2022/01/02 14:08:24 
2022-01-02T13:08:24.697106404Z GET /api/2.0/my/main/current HTTP/1.1
2022-01-02T13:08:24.697138998Z Host: backend.powerfox.energy
2022-01-02T13:08:24.697170218Z User-Agent: Go-http-client/1.1
2022-01-02T13:08:24.697201909Z Accept-Encoding: gzip
2022-01-02T13:08:24.697233269Z 
2022-01-02T13:08:24.697263780Z HTTP/2.0 401 Unauthorized
2022-01-02T13:08:24.697294889Z Content-Length: 72
2022-01-02T13:08:24.697325071Z Access-Control-Allow-Headers: *
2022-01-02T13:08:24.697355596Z Access-Control-Allow-Methods: GET, POST, PUT, PATCH, DELETE, OPTIONS
2022-01-02T13:08:24.697387385Z Access-Control-Allow-Origin: *
2022-01-02T13:08:24.697418026Z Access-Control-Expose-Headers: Request-Context
2022-01-02T13:08:24.697451160Z Cache-Control: no-cache
2022-01-02T13:08:24.697482202Z Content-Type: application/json; charset=utf-8
2022-01-02T13:08:24.697515182Z Date: Sun, 02 Jan 2022 13:12:15 GMT
2022-01-02T13:08:24.697545666Z Expires: -1
2022-01-02T13:08:24.697576444Z Pragma: no-cache
2022-01-02T13:08:24.697606484Z Ratelimit-Limit: 700
2022-01-02T13:08:24.697636935Z Ratelimit-Remaining: 693
2022-01-02T13:08:24.697667836Z Request-Context: appId=cid-v1:9d7e501f-422d-4624-a8b7-5b542e8cb16f
2022-01-02T13:08:24.697698790Z Server: Microsoft-IIS/10.0
2022-01-02T13:08:24.697729380Z Www-Authenticate: Basic realm=powerfox
2022-01-02T13:08:24.697760675Z X-Aspnet-Version: 4.0.30319
2022-01-02T13:08:24.697790875Z X-Powered-By: ASP.NET
2022-01-02T13:08:24.697821432Z 
2022-01-02T13:08:24.697851982Z {"Message":"Die Autorisierung wurde für diese Anforderung verweigert."}
2022-01-02T13:08:24.805043251Z [http  ] TRACE 2022/01/02 14:08:24 GET https://backend.powerfox.energy/api/2.0/my/main/current
2022-01-02T13:08:24.845067709Z [http  ] TRACE 2022/01/02 14:08:24 
2022-01-02T13:08:24.845154921Z GET /api/2.0/my/main/current HTTP/1.1
2022-01-02T13:08:24.845188361Z Host: backend.powerfox.energy
2022-01-02T13:08:24.845269074Z User-Agent: Go-http-client/1.1
2022-01-02T13:08:24.845301725Z Accept-Encoding: gzip
2022-01-02T13:08:24.845332188Z 
2022-01-02T13:08:24.845361896Z HTTP/2.0 401 Unauthorized
2022-01-02T13:08:24.845393073Z Content-Length: 72
2022-01-02T13:08:24.845423488Z Access-Control-Allow-Headers: *
2022-01-02T13:08:24.845453765Z Access-Control-Allow-Methods: GET, POST, PUT, PATCH, DELETE, OPTIONS
2022-01-02T13:08:24.845485558Z Access-Control-Allow-Origin: *
2022-01-02T13:08:24.845516147Z Access-Control-Expose-Headers: Request-Context
2022-01-02T13:08:24.845546439Z Cache-Control: no-cache
2022-01-02T13:08:24.845577406Z Content-Type: application/json; charset=utf-8
2022-01-02T13:08:24.845608225Z Date: Sun, 02 Jan 2022 13:12:15 GMT
2022-01-02T13:08:24.845638521Z Expires: -1
2022-01-02T13:08:24.845669561Z Pragma: no-cache
2022-01-02T13:08:24.845699851Z Ratelimit-Limit: 700
2022-01-02T13:08:24.845730088Z Ratelimit-Remaining: 699
2022-01-02T13:08:24.845760980Z Request-Context: appId=cid-v1:9d7e501f-422d-4624-a8b7-5b542e8cb16f
2022-01-02T13:08:24.845792314Z Server: Microsoft-IIS/10.0
2022-01-02T13:08:24.845822552Z Www-Authenticate: Basic realm=powerfox
2022-01-02T13:08:24.845853427Z X-Aspnet-Version: 4.0.30319
2022-01-02T13:08:24.845883990Z X-Powered-By: ASP.NET
2022-01-02T13:08:24.845914178Z 
2022-01-02T13:08:24.845944573Z {"Message":"Die Autorisierung wurde für diese Anforderung verweigert."}
2022-01-02T13:08:25.113631226Z [http  ] TRACE 2022/01/02 14:08:25 GET https://backend.powerfox.energy/api/2.0/my/main/current
2022-01-02T13:08:25.138202265Z [http  ] TRACE 2022/01/02 14:08:25 
2022-01-02T13:08:25.138253242Z GET /api/2.0/my/main/current HTTP/1.1
2022-01-02T13:08:25.138286734Z Host: backend.powerfox.energy
2022-01-02T13:08:25.138318461Z User-Agent: Go-http-client/1.1
2022-01-02T13:08:25.138349453Z Accept-Encoding: gzip
2022-01-02T13:08:25.138380057Z 
2022-01-02T13:08:25.138410629Z HTTP/2.0 401 Unauthorized
2022-01-02T13:08:25.138441760Z Content-Length: 72
2022-01-02T13:08:25.138472765Z Access-Control-Allow-Headers: *
2022-01-02T13:08:25.138503729Z Access-Control-Allow-Methods: GET, POST, PUT, PATCH, DELETE, OPTIONS
2022-01-02T13:08:25.138535251Z Access-Control-Allow-Origin: *
2022-01-02T13:08:25.138566461Z Access-Control-Expose-Headers: Request-Context
2022-01-02T13:08:25.138597161Z Cache-Control: no-cache
2022-01-02T13:08:25.138628166Z Content-Type: application/json; charset=utf-8
2022-01-02T13:08:25.138659131Z Date: Sun, 02 Jan 2022 13:12:16 GMT
2022-01-02T13:08:25.138736301Z Expires: -1
2022-01-02T13:08:25.138768719Z Pragma: no-cache
2022-01-02T13:08:25.138799249Z Ratelimit-Limit: 700
2022-01-02T13:08:25.138830499Z Ratelimit-Remaining: 698
2022-01-02T13:08:25.138861345Z Request-Context: appId=cid-v1:9d7e501f-422d-4624-a8b7-5b542e8cb16f
2022-01-02T13:08:25.138892743Z Server: Microsoft-IIS/10.0
2022-01-02T13:08:25.138923998Z Www-Authenticate: Basic realm=powerfox
2022-01-02T13:08:25.138954885Z X-Aspnet-Version: 4.0.30319
2022-01-02T13:08:25.138985301Z X-Powered-By: ASP.NET
2022-01-02T13:08:25.139016218Z 
2022-01-02T13:08:25.139046504Z {"Message":"Die Autorisierung wurde für diese Anforderung verweigert."}
2022-01-02T13:08:25.139079562Z [site  ] ERROR 2022/01/02 14:08:25 updating grid meter: unexpected status: 401
2022-01-02T13:08:25.139111375Z [cache ] DEBUG 2022/01/02 14:08:25 error: updating grid meter: unexpected status: 401

andig, 2022-01-02:
HTTP/2.0 401 Unauthorized

Da werden keine Zugangsdaten mit geschickt

Clanchef, 2022-01-02:
Das passiert aber nur im Container :-), nicht so auf Kommandozeile

andig, 2022-01-02:
Sorry, aber es fehlen jegliche relevanten Informationen wie im Ticket angefragt. Hier gibt es NICHTS zu tun. Selbst von Container war bisher nicht die Rede. Mach gerne ein vernünftiges Ticket, aber das hier ist Zeitverschwendung.

Clanchef, 2022-01-02:
Hallo Andy,

Der Idiot bin wohl gerade ich. Welche relevanten Information fehlen denn? Ich habe wie angefordert die Log geliefert.

Grüße
Martin

Hofyyy, 2022-01-02:
@andig
Manchmal bist du so direkt und wertschätzend.
Früher gab es für sowas ein Sparschwein mit 50Cent Stücken.

Also ich habe mein EVCC mal im Trace gestartet und leider ist der Fehler weg. Auch sehe ich ihn nach einem evcc stop und start nicht mehr. Ist also bei mir nicht persistent und damit bei mir weg.

Ich hab evcc übrigens nativ auf dem Raspberry.

Annahme: Die haben genau wie BMW an den APIs gefummelt. Bei mir läuft gerade alles wieder. Bis auf das die Wallbe immer noch kommische Daten liefert. Jetzt hatte ich schon eine "-2" :-P

Sollte ich den Fehler noch mal sehen würde ich ein --log trace ziehen.

andig, 2022-01-02:
Wenn Du ein neues Ticket machst siehst Du alles was angefragt ist. U.a. gehört dazu auch die Konfiguration. Ich kann nichtmal erkennen, ob Du überhaupt irgendwo Zugangsdaten angegeben hast ;)

andig, 2022-01-02:
> Manchmal bist du so direkt und wertschätzend.

Fair Point. Damit genau sowas nicht passiert fragen wir im Ticket nach den notwendigen Informationen. Wertschätzend ggü. unserer Zeit ist es dann diese auch mitzugeben!

Clanchef, 2022-01-02:
Bei mir gehts im Container nicht, nativ gehts wie bei Hofyyy.
Ein Passwort steht natürlich in der Config drin, deswegen gehts ja auch nativ.