evcc-io/evcc #2187: Powerfox grid Meter Status 401

Clanchef, 2022-01-02:
**Describe the bug**
Seit den Updates vom 30.12./31.12. (bin mir nicht mehr 100% sicher seit wann genau) kann evcc die Grid Meter Daten von Powerfox nicht mehr auslesen. An der Konfiguration wurde nichts geändert.
Anzeige in der GUI:
updating grid meter: unexpected status: 401
Das Powerfox Backend funktioniert, in anderen Anwendung wird der Powerfox Status korrekt ausgelesen.

**To Reproduce**
Steps to reproduce the behavior:
evcc läuft im Docker Container (Ubuntu 21.10), letzte Version 0.77 gestartet mit docker-compose

**Expected behavior**
Powerfox Daten werden als Grid Meter Daten in evcc verwendet.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
<- 
Aus der Kommandozeile aufgerufen, nicht vom Container

```
config
======

grid: powerfox
--------------
Power: -453W

pv: pv
------
Power: 700W

loadpoint 1
===========

charger: wbecMQTT-LP1
---------------------
Charge status: F
Enabled:       false

loadpoint 2
===========

charger: wbecMQTT-LP2
---------------------
Charge status: F
Enabled:       false
```

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
<- paste here
```
```

uri: 0.0.0.0:7070 # uri for ui
interval: 5s # control cycle interval

log: error
levels:
  core: debug
  lp-1: debug
  lp-2: debug

meters:
- name: powerfox
  type: custom
[... 249 lines cut ...]
2022-01-02T13:08:25.138380057Z 
2022-01-02T13:08:25.138410629Z HTTP/2.0 401 Unauthorized
2022-01-02T13:08:25.138441760Z Content-Length: 72
2022-01-02T13:08:25.138472765Z Access-Control-Allow-Headers: *
2022-01-02T13:08:25.138503729Z Access-Control-Allow-Methods: GET, POST, PUT, PATCH, DELETE, OPTIONS
2022-01-02T13:08:25.138535251Z Access-Control-Allow-Origin: *
2022-01-02T13:08:25.138566461Z Access-Control-Expose-Headers: Request-Context
2022-01-02T13:08:25.138597161Z Cache-Control: no-cache
2022-01-02T13:08:25.138628166Z Content-Type: application/json; charset=utf-8
2022-01-02T13:08:25.138659131Z Date: Sun, 02 Jan 2022 13:12:16 GMT
2022-01-02T13:08:25.138736301Z Expires: -1
2022-01-02T13:08:25.138768719Z Pragma: no-cache
2022-01-02T13:08:25.138799249Z Ratelimit-Limit: 700
2022-01-02T13:08:25.138830499Z Ratelimit-Remaining: 698
2022-01-02T13:08:25.138861345Z Request-Context: appId=cid-v1:9d7e501f-422d-4624-a8b7-5b542e8cb16f
2022-01-02T13:08:25.138892743Z Server: Microsoft-IIS/10.0
2022-01-02T13:08:25.138923998Z Www-Authenticate: Basic realm=powerfox
2022-01-02T13:08:25.138954885Z X-Aspnet-Version: 4.0.30319
2022-01-02T13:08:25.138985301Z X-Powered-By: ASP.NET
2022-01-02T13:08:25.139016218Z 
2022-01-02T13:08:25.139046504Z {"Message":"Die Autorisierung wurde für diese Anforderung verweigert."}
2022-01-02T13:08:25.139079562Z [site  ] ERROR 2022/01/02 14:08:25 updating grid meter: unexpected status: 401
2022-01-02T13:08:25.139111375Z [cache ] DEBUG 2022/01/02 14:08:25 error: updating grid meter: unexpected status: 401

</pre></details>

Clanchef, 2022-01-02:
Das Formular ist ein wenig anstrengend

andig, 2022-01-02:
@DerAndereAndi haben wir für den Powerfox auch ein Template?

@Clanchef Du bist Dir 100%ig sicher, dass in Docker und nativ die gleiche Version läuft??? Kannst Du bitte mal die ersten paar Logzeilen lokal/Docker zeigen?

> Das Formular ist ein wenig anstrengend

https://docs.github.com/en/github/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax

andig, 2022-01-02:
@Clanchef kannst Du Deine Zugangsdaten mal mit cpuidle@gmx.de (mir) teilen?

andig, 2022-01-02:
Kannst Du ignorieren, hab den Fehler gefunden.

andig, 2022-01-02:
Du hast mit Sicherheit in Docker und Lokal eine unterschiedliche Version laufen. Das Problem in der Neueren ist mit https://github.com/evcc-io/evcc/commit/159e0304c5e00a45542e3543d2b0bc8df59bce76 behoben.

closed 2022-01-02 by andig

Clanchef, 2022-01-03:
Vielen Dank für die schnelle Behebung! Funktioniert wieder :-)