evcc-io/evcc #2198: Discovery Grid Power is zero, but PV Power is working

PanNwt, 2022-01-03:
**Describe the bug**
After update of evcc to 0.78 (docker) the grid power is always zero.

**To Reproduce**
Steps to reproduce the behavior:
1. Open evcc page with an grid and pv discovergy meter

**Expected behavior**
The grid power will be retrieved and used.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.78 (68cfebd)
```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
config
======

grid: grid
----------
Power:  0W
Energy: 10645854.6kWh

pv: pv
------
Power:  -2W
Energy: 74646.2kWh

loadpoint 1
===========

charger: garage
---------------
Power:          0W
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        false
Charged:        4.1kWh

vehicle 0
---------
SoC:            64%
Charge status:  A
Range:          161km
Odometer:       10730km
Finish time:    2022-01-03 21:16:00 +0000 UTC
Climate active: false
Outside temp:   0.0°C
Target temp:    20.0°C
Capacity:       52kWh
OnIdentified:   Mode:pv, MinCurrent:7, MaxCurrent:16, TargetSoC:80

vehicle 1
---------
SoC:           66%
Charge status: B
Range:         66km
Odometer:      84291km
Finish time:   2022-01-04 00:28:00 +0100 CET
Capacity:      17kWh
OnIdentified:  Mode:pv, MinCurrent:6, MaxCurrent:15, TargetSoC:100
</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
uri: 192.168.local.evcc:80 # uri for ui
interval: 10s # control cycle interval

sponsortoken: *****
[... 186 lines cut ...]
[cache ] DEBUG 2022/01/03 19:25:28 savingsChargedSelfConsumption: 0
[cache ] DEBUG 2022/01/03 19:25:28 savingsSelfPercentage: 0
[cache ] DEBUG 2022/01/03 19:25:28 savingsSince: 1.983004136s
[cache ] DEBUG 2022/01/03 19:25:28 currency: EUR
[cache ] DEBUG 2022/01/03 19:25:28 mode: pv
[lp-1  ] DEBUG 2022/01/03 19:25:28 charge power: 0W
[cache ] DEBUG 2022/01/03 19:25:28 chargePower: 0
[lp-1  ] DEBUG 2022/01/03 19:25:28 charge currents: [0 0 0]A
[cache ] DEBUG 2022/01/03 19:25:28 chargeCurrents: [0 0 0]
[cache ] DEBUG 2022/01/03 19:25:28 chargeCurrent: 0
[cache ] DEBUG 2022/01/03 19:25:28 chargedEnergy: 4086.2
[cache ] DEBUG 2022/01/03 19:25:28 chargeDuration: 0s
[lp-1  ] DEBUG 2022/01/03 19:25:28 charger status: B
[cache ] DEBUG 2022/01/03 19:25:28 connected: true
[cache ] DEBUG 2022/01/03 19:25:28 charging: false
[cache ] DEBUG 2022/01/03 19:25:28 enabled: false
[lp-1  ] DEBUG 2022/01/03 19:25:28 max charge current: 0A = 0A + -0A (0W @ 1p)
[lp-1  ] DEBUG 2022/01/03 19:25:28 charge power: 0W
[cache ] DEBUG 2022/01/03 19:25:28 chargePower: 0
[lp-1  ] DEBUG 2022/01/03 19:25:28 charge currents: [0 0 0]A
[savings] DEBUG 2022/01/03 19:25:28 0.0kWh charged since 3s
[savings] DEBUG 2022/01/03 19:25:28 0.0kWh own energy (0.0%)
[cache ] DEBUG 2022/01/03 19:25:28 homePower: 0
[cache ] DEBUG 2022/01/03 19:25:28 chargeCurrents: [0 0 0]
</pre></details>

andig, 2022-01-03:
magst mal Deine Zähler und Zugangsdaten an cpuidle@gmx.de schicken?

andig, 2022-01-03:
Danke. Ist in master behoben: https://github.com/evcc-io/evcc/commit/ae2c1bda77913d01277fe1be91a9c1f7e7211051. Da hab ich beim Umbau leider einiges kaputt gemacht. Als workaround kannst Du `scale: 1` setzen.

closed 2022-01-03 by andig

PanNwt, 2022-01-04:
Danke, die 0.79 funzt bei mir.