evcc-io/evcc #2222: Fehlermeldung 2. Ladepunkt "charge meter: json: cannot unmarshal number..."

pablopalec, 2022-01-06:
**Describe the bug**
Fehlermeldung an zweitem Ladepunkt

**To Reproduce**
Steps to reproduce the behavior:
Ich habe zum testen auf Basis einer jungfräulichen Raspberry OS Installation evcc gemäß Anleitung installiert und per evcc  configure konfiguriert. An dieser Stelle ein großes Lob für den Assistenten!

Am Ladepunkt "Hof" funktioniert evcc wie gewünscht, beim Ladepunkt "Garage" bekomme ich Fehlermeldungen, die ich nicht zuordnen kann. Das Laden kann aber z.B. über Anwahl des Modus "Sofort" gestartet werden.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.79 (ae2c1bd)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
Unsupported Config Type ""

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

log: info

meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.254.86
  name: grid1
- type: template
  template: sma-inverter
  usage: pv
  host: 192.168.254.79
  password: S0nnend1v1de  # Wird nur benötigt, wenn das Standardpasswort geändert wurde
  name: pv2

chargers:
- type: template
  template: go-e
  host: 192.168.254.125
  name: wallbox4
- type: template
  template: go-e
  host: 192.168.254.137
  name: wallbox5

vehicles:
- type: template
  template: seat
  title: Mii electric
  user: finger_seat@palec.de
  password: Hund3kuchen!
  vin: VSSZZZAAZMD509958
  capacity: 32
  cache: 5
  mode: pv
  minSoC: 40
  targetSoC: 95
  minCurrent: 6
  maxCurrent: 16
  name: ev3

loadpoints:
- title: Hof
  charger: wallbox4
  vehicles:
[... 75 lines cut ...]
Jan 06 12:11:49 steuermann evcc[853]: [lp-2  ] ERROR 2022/01/06 12:11:49 charge meter: json: cannot unmarshal number 186.8781 into Go struct field Card.Cards.Energy of type int
Jan 06 12:11:49 steuermann evcc[853]: [lp-2  ] ERROR 2022/01/06 12:11:49 charge meter: json: cannot unmarshal number 189.0313 into Go struct field Card.Cards.Energy of type int
Jan 06 12:11:49 steuermann evcc[853]: [lp-2  ] ERROR 2022/01/06 12:11:49 charge rater: json: cannot unmarshal number 189.0313 into Go struct field Card.Cards.Energy of type int
Jan 06 12:11:50 steuermann evcc[853]: [lp-2  ] ERROR 2022/01/06 12:11:50 charger: json: cannot unmarshal number 189.0313 into Go struct field Card.Cards.Energy of type int
Jan 06 12:12:09 steuermann evcc[853]: [lp-2  ] INFO 2022/01/06 12:12:09 car disconnected
Jan 06 12:12:10 steuermann evcc[853]: [lp-2  ] WARN 2022/01/06 12:12:10 charger out of sync: expected enabled, got disabled
Jan 06 12:12:28 steuermann evcc[853]: [lp-2  ] WARN 2022/01/06 12:12:28 charger out of sync: expected enabled, got disabled
Jan 06 12:12:51 steuermann evcc[853]: [lp-2  ] WARN 2022/01/06 12:12:51 charger out of sync: expected enabled, got disabled
Jan 06 12:13:09 steuermann evcc[853]: [lp-2  ] WARN 2022/01/06 12:13:09 charger out of sync: expected enabled, got disabled
Jan 06 12:13:31 steuermann evcc[853]: [lp-2  ] WARN 2022/01/06 12:13:31 charger out of sync: expected enabled, got disabled
Jan 06 12:13:58 steuermann evcc[853]: [lp-2  ] WARN 2022/01/06 12:13:58 charger out of sync: expected enabled, got disabled
Jan 06 12:14:00 steuermann evcc[853]: [lp-1  ] INFO 2022/01/06 12:14:00 car connected
Jan 06 12:14:08 steuermann evcc[853]: [lp-2  ] WARN 2022/01/06 12:14:08 charger out of sync: expected enabled, got disabled
Jan 06 12:14:18 steuermann evcc[853]: [lp-1  ] INFO 2022/01/06 12:14:18 start charging ->
Jan 06 12:14:29 steuermann evcc[853]: [lp-2  ] WARN 2022/01/06 12:14:29 charger out of sync: expected enabled, got disabled
Jan 06 12:14:52 steuermann evcc[853]: [lp-2  ] WARN 2022/01/06 12:14:52 charger out of sync: expected enabled, got disabled
Jan 06 12:15:08 steuermann evcc[853]: [lp-2  ] WARN 2022/01/06 12:15:08 charger out of sync: expected enabled, got disabled
Jan 06 12:15:28 steuermann evcc[853]: [lp-2  ] WARN 2022/01/06 12:15:28 charger out of sync: expected enabled, got disabled
Jan 06 12:17:58 steuermann evcc[853]: [lp-1  ] INFO 2022/01/06 12:17:58 stop charging <-
Jan 06 12:36:18 steuermann evcc[853]: [lp-1  ] INFO 2022/01/06 12:36:18 start charging ->
Jan 06 12:36:58 steuermann evcc[853]: [lp-1  ] INFO 2022/01/06 12:36:58 stop charging <-
Jan 06 12:36:58 steuermann evcc[853]: [lp-1  ] WARN 2022/01/06 12:36:58 charger out of sync: expected enabled, got disabled
Jan 06 12:37:18 steuermann evcc[853]: [lp-1  ] INFO 2022/01/06 12:37:18 start charging 

</pre></details>

andig, 2022-01-06:
@pablopalec erstmal Dickes Danke für das hervorragend beschriebene Ticket! Sieht hier aus, als würde Dein go-e float statt int liefern, ich korrigiere das...

andig, 2022-01-06:
Mit https://github.com/evcc-io/evcc/commit/759381a2f32de1f9cc0a3e5442d1c14a4871e39f behoben, kommt mit dem nächsten Release bzw. nightly build.

closed 2022-01-06 by andig

pablopalec, 2022-01-06:
Hallo andi und vielen Dank für den schnelle Fix! Ein großes Lob noch einmal für die Entwicklung von evcc. Ich hoffe du behältst den Spaß dabei und bin gespannt, wohin die Reise noch geht!

Gruß,

stephan

andig, 2022-01-06:
Der Spaß ist da und das Team machts. Ohne soviele Mitstreiter- sei es im Kernteam oder auch in der Community- wäre das Ergebnis nicht das was wir erreicht haben und es würde auch nicht soviel Spass machen...

pablopalec, 2022-01-17:
En kurzes Feedback, ich habe es nicht eher geschafft zu testen: der Fehler ist behoben, evcc läuft in meiner Konstellation tadellos. Ich bin begeistert, vielen Dank und weiter so!