evcc-io/evcc #2225: Calculate savings in backend

andig, 2022-01-06:
Currently this is done in FE using BE data:

	if site.tariffs.Grid != nil {
		if gridPrice, err := site.tariffs.Grid.CurrentPrice(); err == nil {
			site.publish("tariffGrid", gridPrice)
		}
	}
	if site.tariffs.FeedIn != nil {
		if feedInPrice, err := site.tariffs.FeedIn.CurrentPrice(); err == nil {
			site.publish("tariffFeedIn", feedInPrice)
		}
	}

The running total should be in the BE as tariffs can change over time when using Awattar and the like.

naltatis, 2022-01-06:
Ja, die aktuelle Implementierung ist bewusst simpel und geht von einem konstanten Strompreis aus (siehe Anmerkung in der README). Bin mir auch noch nicht sicher was hier wirklich der aussagekräftigste Wert ist den wir kommunizieren. Tendiere dazu die Ersparnis in € aus der Anzeige rauszunehmen und den effektiven Preis/kWh stärker in der Vordergrund zu stellen.

andig, 2022-01-06:
Ich hätte jetzt einfach das running total im Backend gemacht. Zu jedem Summenzeitpunkt kennen wir ja den richtigen Preis. Gerne im angefangenen PR.

closed 2022-01-09 by andig