evcc-io/evcc #2230: Determine 1p3p state on startup for supported chargers

andig, 2022-01-07:


naltatis, 2022-01-07:
Glaube, dass das nicht nur den Start betrifft. Hatte gestern auch den Fall, dass die minSoC Ladung nach dem Anstecken mit 1p gestartet wurde (und auch kein phase upgrade timer lief). Das war nicht der erste Ladevorgang, vorher wurde auf jeden Fall schon PV geladen, heißt evcc hätte eigentlich wissen müssen auf welcher Phase der Charger gerade steht.

Mein Log reicht momentan leider nicht weit genug zurück. Ich poste hier eins wenn es das nächste Mal auftritt.

bacanol, 2022-01-07:
Wäre es auch möglich direkt vor dem Start die Phasen nach dem verfügbaren PV einzustellen, um eine erneute Unterbrechung der Ladung zu verhindern? Bspw: 
pv: 1500W -> 1p
pv: 6000W -> 3p

Aktuell startet lp-1 (dreiphasig angeschlossen) bei mir immer 3p merkt dann, dass die Sonne gerade hinter dem Horizont aufgetaucht ist und nur 1400W da sind, stoppt dann und schaltet auf 1p runter und möchte dann wieder starten. Meistens hat dann der lp-2 auch mitbekommen, dass Sonne da ist und hat direkt einphasig angefangen (ist nur einphasig angeschlossen). lp-1 geht meistens leer aus, da dann die Sonne nicht mehr ausreicht. Wenn er nun gleich 1p anfangen würde, wäre es IMHO besser. Dann lieber hochschalten, wenn genug da ist.

andig, 2022-01-07:
> Aktuell startet lp-1 (dreiphasig angeschlossen) bei mir immer 3p, merkt dann, dass die Sonne gerade hinter dem Horizont aufgetaucht ist und nur 1400W da sind, stoppt dann und schaltet auf 1p runter und möchte dann wieder starten.

Das macht wenig Sinn. Wenn die Sonne hoch komme ist nur für 1p Strom da und er sollte daher NIE mit 3p starten. Er weiss ja dass es nicht reicht (es sei denn es hängt ein 1p Auto dran!). Hast Du mal ein Logfile davon?

bacanol, 2022-01-07:
Sehe ich so wie Du @andig, hier das Log von gestern:
```
Jan 06 10:05:48 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:05:48 ----
Jan 06 10:05:48 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:05:48 pv power: 2087W
Jan 06 10:05:48 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:05:48 grid power: -1331W
Jan 06 10:05:48 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:05:48 site power: -1231W
Jan 06 10:05:48 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:05:48 charge power: 0W
Jan 06 10:05:48 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:05:48 charge currents: [0 0 0]A
Jan 06 10:05:48 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:05:48 charger status: B
Jan 06 10:05:51 evcc evcc[438139]: [lp-1  ] ERROR 2022/01/06 10:05:51 vehicle soc: 408 Request Timeout
Jan 06 10:05:51 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:05:51 pv action: inactive
Jan 06 10:05:51 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:05:51 max charge current: 16A
Jan 06 10:05:51 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:05:51 charger enable
Jan 06 10:05:51 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:05:51 pv action: inactive
Jan 06 10:05:52 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:05:52 charge power: 0W
Jan 06 10:05:52 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:05:52 charge currents: [0 0 0]A
Jan 06 10:05:58 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:05:58 ----
Jan 06 10:05:59 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:05:59 pv power: 2086W
Jan 06 10:05:59 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:05:59 grid power: -1329W
Jan 06 10:05:59 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:05:59 site power: -1229W
Jan 06 10:05:59 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:05:59 charge power: 0W
Jan 06 10:05:59 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:05:59 charge currents: [0 0 0]A
Jan 06 10:05:59 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:05:59 charger status: B
Jan 06 10:06:01 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:06:01 vehicle status: B
Jan 06 10:06:01 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:06:01 vehicle soc: 50%
Jan 06 10:06:01 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:06:01 vehicle range: 133km
Jan 06 10:06:02 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:06:02 vehicle odometer: 5525km
Jan 06 10:06:02 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:06:02 max charge current: 5.34A = 0A + 5.34A (-1229W @ 1p)
Jan 06 10:06:03 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:06:03 climater active: false, target temp: 20.0°C, outside temp: 0.0°C
Jan 06 10:06:03 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:06:03 charge power: 0W
Jan 06 10:06:03 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:06:03 charge currents: [0 0 0]A
Jan 06 10:06:08 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:06:08 ----
Jan 06 10:06:08 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:06:08 pv power: 2087W
Jan 06 10:06:08 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:06:08 grid power: -1423W
Jan 06 10:06:08 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:06:08 site power: -1323W
Jan 06 10:06:08 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:06:08 charge power: 30W
Jan 06 10:06:08 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:06:08 charge currents: [0.6 0.8 0.9]A
Jan 06 10:06:08 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:06:08 charger status: C
Jan 06 10:06:08 evcc evcc[438139]: [lp-1  ] INFO 2022/01/06 10:06:08 start charging ->
Jan 06 10:06:08 evcc evcc[438139]: [lp-1  ] ERROR 2022/01/06 10:06:08 vehicle soc: 408 Request Timeout
Jan 06 10:06:08 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:06:08 pv action: inactive
Jan 06 10:06:08 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:06:08 charge power: 30W
Jan 06 10:06:08 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:06:08 charge currents: [0.6 0.8 0.9]A
Jan 06 10:06:18 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:06:18 ----
Jan 06 10:06:18 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:06:18 pv power: 2096W
Jan 06 10:06:18 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:06:18 grid power: -1237W
Jan 06 10:06:18 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:06:18 site power: -1137W
Jan 06 10:06:18 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:06:18 charge power: 0W
Jan 06 10:06:18 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:06:18 charge currents: [0 0 0]A
Jan 06 10:06:18 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:06:18 charger status: B
Jan 06 10:06:18 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:06:18 next soc poll remaining time: 59m41s
Jan 06 10:06:18 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:06:18 max charge current: 4.94A = 0A + 4.94A (-1137W @ 1p)
Jan 06 10:06:18 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:06:18 climater active: false, target temp: 20.0°C, outside temp: 0.0°C
Jan 06 10:06:18 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:06:18 charge power: 0W
Jan 06 10:06:18 evcc evcc[438139]: [lp-2  ] DEBUG 2022/01/06 10:06:18 charge currents: [0 0 0]A
Jan 06 10:06:28 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:06:28 ----
Jan 06 10:06:28 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:06:28 pv power: 2099W
Jan 06 10:06:28 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:06:28 grid power: 2620W
Jan 06 10:06:28 evcc evcc[438139]: [site  ] DEBUG 2022/01/06 10:06:28 site power: 2720W
Jan 06 10:06:28 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:06:28 charge power: 4490W
Jan 06 10:06:28 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:06:28 charge currents: [9.8 9.8 1.1]A
[... cut ...]
Jan 06 10:08:08 evcc evcc[438139]: [lp-1  ] ERROR 2022/01/06 10:08:08 vehicle soc: 408 Request Timeout
Jan 06 10:08:08 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:08:08 pv action: inactive
Jan 06 10:08:08 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:08:08 charge power: 10740W
Jan 06 10:08:08 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:08:08 charge currents: [15.6 15.8 15.8]A
Jan 06 10:08:08 evcc evcc[438139]: [lp-1  ] DEBUG 2022/01/06 10:08:08 detected phases: 3p [15.6 15.8 15.8]A
```

andig, 2022-01-07:
Aus Slack: die die Automatik zur Phasenerkennung schießt sich selbst in den Fuß:

    [lp-2  ] DEBUG 2022/01/07 09:50:47 phase disable timer remaining: 0s
    [cache ] DEBUG 2022/01/07 09:50:47 phaseAction: scale1p
    [lp-2  ] DEBUG 2022/01/07 09:50:57 switched phases: 1p @ -656W
    [cache ] DEBUG 2022/01/07 09:50:57 phases: 1
    [cache ] DEBUG 2022/01/07 09:50:57 activePhases: 1

aber dann:

    [lp-2  ] DEBUG 2022/01/07 09:50:58 charge currents: [6.07 6.12 6.05]A
    [cache ] DEBUG 2022/01/07 09:50:58 activePhases: 3

und damit:

    [lp-2  ] INFO 2022/01/07 09:51:07 stop charging <-
    [lp-2  ] WARN 2022/01/07 09:51:07 inconsistent phases: 1p configured < 3p observed active, updating internal state to 3p

Es dauert 20s bis wir das Ladeende bemerken, vorher kamen aber neue Meßwerte, die noch 3p suggerieren während evcc intern schon auf 1p umgestellt hat. Und dann in Folge wieder zurück korrigiert was falsch ist :(

andig, 2022-01-07:
> Sehe ich so wie Du @andig, hier das Log von gestern:

Kann ich leider nix mit anfangen weil die Ausgangslage nicht klar. Es fehlt das letzte Statement vor her zu phases/activePhases.

bacanol, 2022-01-07:
> > Sehe ich so wie Du @andig, hier das Log von gestern:
> 
> Kann ich leider nix mit anfangen weil die Ausgangslage nicht klar. Es fehlt das letzte Statement vor her zu phases/activePhases.

Habe jetzt noch cache mit in den Debuglog aufgenommen, mal sehen, wie es jetzt läuft. Was mir aufgefallen ist, seit dem aktuellen Master startet bei mir die Ladung in lp-1 beim Neustart von evcc, wenn vorher keine Ladung aktiv war mit 3p 16A, timer zählen nicht herunter. Ist das ein Seiteneffekt oder soll ich dafür ein neues Ticket öffnen?

bacanol, 2022-01-07:
> Habe jetzt noch cache mit in den Debuglog aufgenommen, mal sehen, wie es jetzt läuft. Was mir aufgefallen ist, seit dem aktuellen Master startet bei mir die Ladung in lp-1 beim Neustart von evcc, [...]

@andig , hier ist exakt das Verhalten nach dem Neustart, welches ich beschrieben habe:
[evcc-2.log](https://github.com/evcc-io/evcc/files/7828032/evcc-2.log)

Version:
evcc --version
evcc version 0.79 (96c6c625)

Kannst Du Dir darauf einen Reim machen?

andig, 2022-01-07:
Von welchem LP reden wir? Bitte etwas mehr Kontext!

bacanol, 2022-01-07:
> Von welchem LP reden wir? Bitte etwas mehr Kontext!

Wie geschrieben lp-1 .

andig, 2022-01-07:
> > Von welchem LP reden wir? Bitte etwas mehr Kontext!
> 
> Wie geschrieben lp-1 .

Gleiches Problem wie in Slack und Punkt 1 der Todos:

    phases: 1

Deine Config gaukelt 1p vor, der Charger kann aber und steht vmtl auf 3p. Daher Ladefreigabe die auf 3p läuft und dann wird irgendwann runter geschaltet.

andig, 2022-01-07:
@bacanol https://github.com/evcc-io/evcc/pull/2231 sollte Dein Problem bei gleicher Config erkennen/lösen.

bacanol, 2022-01-07:
> 

Habe phases jetzt rausgenommen beim lp-1. Er fängt nach dem Neustart direkt an mit 3p zu laden im PV-modus ohne Sonne  *verstehichnicht* werden den #2231 jetzt nochmal einspielen.

andig, 2022-01-08:
> Habe phases jetzt rausgenommen beim lp-1. Er fängt nach dem Neustart direkt an mit 3p zu laden im PV-modus ohne Sonne _verstehichnicht_ werden den #2231 jetzt nochmal einspielen.

Setz mal bitte 3p, genau so wie Deine WB angeschlossen ist.

andig, 2022-01-08:
Falls wir ein neues API benötigen wäre bisher (nur) Alfen, Easee und Go-E betroffen, der Aufwand also (bis auf die Easee) überschaubar.

premultiply, 2022-01-08:
Auch hier könnte ein Flussdiagramm helfen einen besser Überblick zu schaffen und zu behalten.

andig, 2022-01-09:
Ich hätte noch ein Experiment bzgl. Lösungssuche aus Slack:

> Die Phasenumschaltung scheint ja async zur Modbuskommunikation zu erfolgen. Könnt ihr mal bitte ausprobieren, was passiert wenn Du über Modbus umschaltest und dann- während die physische Trennung noch erfolgt- den Phaseszustand wieder ausliest? Wird der eingestellte (meine Befürchtung) oder der wahre (das wär gut) Zustand ausgegeben?

andig, 2022-01-09:
Solange wir den Phasenstatus nicht auslesen könnten (im Moment Write only) gibts folgendes Problem:
- falls 1p im Charger eingestellt aber 3p in evcc konfiguriert, werden wir _nie_ hoch schalten weil es immer so aussieht, als würde einfach ein 1p Auto am 3p Ladepunkt hängen
- falls 3p im Charger eingestellt aber 1p in evcc konfiguriert, finden wir die Inkonsistenz aufgrund der tatsächlichen 3p Ladung und korrigieren den internen Zustand entsprechend. Das ist #1572, #1580.

Genau #1580 macht jetzt aber die Probleme weil diese Korrektur auch zuschlägt wenn die Umschaltung async/langsam passiert.

Mein bester Lösungsvorschlag wäre jetzt:
- #1580 zurück zu nehmen: Warnungen weiter anzeigen, aber den internen State nicht anzupassen
- bei vorhandener Phasenumschaltung zum Start die Config immer auf 1p in evcc korrigieren. damit sind wir im 2. Fall oben und bekommen allenfalls einmalig eine Warnung

@bacanol @texperiri @naltatis klingt das sinnvoll? Ich würde das gerne nach Release erstmal in nightly ausprobieren.

premultiply, 2022-01-09:
Verbesserungsvorschlag:
Jedes Vehicle braucht eine optionale maxPhases-Eigenschaft damit vorab klar ist was bei einer Umschaltung passieren wird und ob sie überhaupt sinnvoll ist.

Theoretisch bräuchte auch noch maxCurrent pro Phasenkonfiguration am Vehicle den man mit dem maxCurrent und minCurrent des LPs abgleicht um sinnlose Schaltereien zu vermeiden und "Leistungslöcher" zu erkennen.

Es ist komplex und erfordert einen stimmigen Algorithmus.

@DerAndereAndi Könnte man wie andere Eigenschaften auch in Vehicle-Templates einbauen...

kaeferfreund, 2022-01-09:
Die maxPhases-Eigenschaft wäre auch für die Fahrzeugerkennung hilfreich 😃

bacanol, 2022-01-09:
> Solange wir den Phasenstatus nicht auslesen könnten (im Moment Write only) gibts folgendes Problem:
> 
> * falls 1p im Charger eingestellt aber 3p in evcc konfiguriert, werden wir _nie_ hoch schalten weil es immer so aussieht, als würde einfach ein 1p Auto am 3p Ladepunkt hängen
> * falls 3p im Charger eingestellt aber 1p in evcc konfiguriert, finden wir die Inkonsistenz aufgrund der tatsächlichen 3p Ladung und korrigieren den internen Zustand entsprechend. Das ist [Handle phases setting more elegantly when charger supports phase switching #1572](https://github.com/evcc-io/evcc/issues/1572), [Improve phase configuration handling for switchable chargers #1580](https://github.com/evcc-io/evcc/pull/1580).
> 
> Genau #1580 macht jetzt aber die Probleme weil diese Korrektur auch zuschlägt wenn die Umschaltung async/langsam passiert.
> 
> Mein bester Lösungsvorschlag wäre jetzt:
> 
> * [Improve phase configuration handling for switchable chargers #1580](https://github.com/evcc-io/evcc/pull/1580) zurück zu nehmen: Warnungen weiter anzeigen, aber den internen State nicht anzupassen
> * bei vorhandener Phasenumschaltung zum Start die Config immer auf 1p in evcc korrigieren. damit sind wir im 2. Fall oben und bekommen allenfalls einmalig eine Warnung
> 
> @bacanol @texperiri @naltatis klingt das sinnvoll? Ich würde das gerne nach Release erstmal in nightly ausprobieren.

Hallo, habe gerade den aktuellen Master nochmal probiert und dam lp-1 3 Phasen konfiguriert, er beginnt nach dem Neustart direkt mit dem Laden 3p 16A, irgendwie ist da was komisch. Habe evcc dann nochmal neu gestartet, dann bemerkt er, dass die Sonne quasi nicht scheint und beendet das Laden an lp-1. 

[evcc-3.log](https://github.com/evcc-io/evcc/files/7835126/evcc-3.log)

texperiri, 2022-01-09:
Leider hab' ich den 1p3p Code noch nicht 100% durchschaut ... daher bitte um Nachsicht wenn der Code das "eh" schon macht:
Aus meiner Sicht würde ich bei der 1p3p Umschaltung folgendermaßen vorgehen:
1) Es gibt einen Sollzustand: lp.phases
2) Es gibt einen "gut geratenen" Ist-Zustand: lp.activePhases
3) Es gibt einen "zuletzt geschaltet" Zustand der besagt welcher Befehl: 1p oder 3p zuletzt an das Device geschickt wurde.

Die Soll/Ist Zustände sollten unabhängig voneinander verwaltet werden.  
Der Sollzustand ergibt sich aus dem Mode - e.g. "now" oder bei min+PV oder PV aus dem Energieüberschuss der produziert wird.
Der Ist-Zustand aus der Energie / den Phasenströmen

Verhalten:
* Sollzustand != "zuletzt geschaltet" Zustand
  * Schaltbefehl erteilen
  * "Zuletzt geschaltet" Zustand auf den Sollzustand setzen
* Ist-Zustand 3p, Sollzustand 1p (und bereits geschaltet)
  *  nach (?konfigurierbar) 10 Minuten 3p Schaltbefehl erneut senden
  *  Problem bleibt über längeren Zeitraum bestehen (1-2 Schaltversuche auf 1p): Fehler ausgeben - vielleicht ist der 1p3p Schütz kleben geblieben => Anwender muss sich darum kümmen. ? Laden beenden vs. auf 3p fertig laden (keine weiteren Schaltversuche)
* Ist-Zustand: 1p, Sollzustand: 3p (obwohl schon auf 3p umgeschaltet wurde)
 * a) ein 1p Auto => OK
 * b) die 3p Umschaltung fehlgeschlagen sein ... Wahrscheinlichkeit ist eher gering. => keine Aktion
* Soll, Ist, und "zuletzt geschaltet" Zustand werden kontinuierlich überwacht (nicht nur bei Umschaltvorgängen)

Initial - würde ich "zuletzt geschaltet" auf "unknown" setzen.
Damit ergibt sich automatisch eine 1p oder 3p Umschaltung abhängig vom Mode / verfügbarer Energie

andig, 2022-01-09:
> Hallo, habe gerade den aktuellen Master nochmal probiert und dam lp-1 3 Phasen konfiguriert, er beginnt nach dem Neustart direkt mit dem Laden 3p 16A, irgendwie ist da was komisch. Habe evcc dann [...]

Soweit ich sehe steht bei Dir nach Neustart

    phases: 1

Also nix 3p

andig, 2022-01-09:
> Es gibt einen Sollzustand: lp.phases

@texperiri Da fängt das Elend an. Beim Programmstart wissen wir nicht, ob Konfiguration == Sollzustand. Config statisch, Sollzustand dynamisch. Wenn wir diesen Sollzustand laut Config jetzt immer forcieren würden dann würden laufende Ladevorgänge vmtl. alle erstmal gestoppt. Scheint keine gute Idee zu sein.

> Initial - würde ich "zuletzt geschaltet" auf "unknown" setzen. Damit ergibt sich automatisch eine 1p oder 3p Umschaltung abhängig vom Mode / verfügbarer Energie

Gleiches Problem beim Neustart wie oben. Alles Andere sind Folgeprobleme.

> Ist-Zustand 3p, Sollzustand 1p (und bereits geschaltet): nach (?konfigurierbar) 10 Minuten 3p Schaltbefehl erneut senden

Yep, das fehlt. Aktuell korrigieren wir den internen "Soll"Zustand wieder auf das Ist. Alle Dinge mit Zustand merken und Zeitverhalten einbauen sind aber schwer zu testen und recht komplex.

Ein ganz einfacher Workaround statt Deinem komplexeren Vorschlag wäre nach dem Umschalten einfach mal 30s zu warten- so macht das OpenWB. Dadurch können die Inkonsistenzen ziemlich sicher vermieden werden.

Sollen wir das mal probieren?

texperiri, 2022-01-09:
@andig
> Sollen wir das mal probieren?

... gern ... ich würd's mal durch den Simulator durchlassen, und könnte es wahrscheinlich heute noch an meiner Ladestation testen / spätestens am Dienstag.

> Da fängt das Elend an. Beim Programmstart wissen wir nicht, ob Sollzustand(Konfiguration) == Realität. Sollzustand(Config) statisch, Realität dynamisch. Wenn wir diesen Sollzustand laut Config jetzt immer forcieren würden dann würden laufende Ladevorgänge vmtl. alle erstmal gestoppt. Scheint keine gute Idee zu sein

... interessanter UseCase: das Programm während des Ladevorgangs neu zu starten ... das kann aus meiner Sicht nur mit Persistierung des Schaltzustands sauber gelöst werden (außer Du hast - wie oben schon erwähnt - die Möglichkeit den Phasen-Schaltzustand aus der Wallbox auszulesen). Bezüglich Docker und Persistierung muss man wahrscheinlich aufpassen ...

andig, 2022-01-09:
Ich versuche nochmal die Konstellationen zum Programmstart aufzumalen. Veilleicht finden wir ja für alle eine gute Antwort ohne neue APIs.

bacanol, 2022-01-10:
> Soweit ich sehe steht bei Dir nach Neustart
>
> ```
> phases: 1
> ```
>
> Also nix 3p

Das ist ja seltsam. Ist in der Config aber enthalten. Merkt er sich das irgendwo, so dass die Config nur für den 1.Start gezogen wird?

andig, 2022-01-10:
> Merkt er sich das irgendwo, so dass die Config nur für den 1.Start gezogen wird?

Ne. Das ist wirklich seltsam, ich schau mal ob ichs reproduzieren kann.

andig, 2022-01-10:
Nochmal ein paar Gedanken zum Startverhalten:
- problematisch sind nur die Situationen in denen die Config nicht zum Chargerzustand passt (1p vs 3p und umgekehrt)
- wenn wir _nicht_ gerade laden könnten wir bedenkenlos umschalten- sinnvollerweise immer auf den konfigurierten Wert der im Idealfall zum Auto/ zur PV passen sollte. Alternativ auf einen Wert abhängig PV Überschuss.
- wenn gerade geladen wird ist es schwieriger da wir Ladevorgänge vorzugsweise nicht unterbrechen wollen: 
  - 3p laden bei 1p Config ist als Fehler erkennbar und wir könnten einfach die richtige Config annehmen ohne umzuschalten
  - 1p laden bei 3p Config ist nicht erkennbar da das sowohl an Auto, wie auch an Chargerzustand liegen könnte. Da wäre zu entscheiden, ob wir auf 1p oder 3p umschalten wollen. Entweder wieder auf den konfigurierten Wert oder alternativ abhängig PV Überschuss oder abhängig der aktuellen Ladung. Meine Erwartung an die Hardware wäre, dass "Null"Umschaltungen, also wo Zielzustand schon dem internen Zustand entspricht, einfach nicht ausgeführt werden und damit auch keinen Fehler verursachen.
- das Problem aus #1572 bleibt separat zu lösen. Solange wir keine Zeitkonstante einführen würde ich lieber mit Warnungen leben als den internen Zustand, wie mit #1580 eingeführt, falsch zu korrigieren (heisst also #1580 zurück zu nehmen)

Was haltet ihr von diesem Lösungsvorschlag?

premultiply, 2022-01-10:
Die Konfig-Option `lp.phases` dürfte für umschaltbare Charger eigentlich keine Auswirkung haben. Die Option braucht es nur für manuelle Lösungen (wie Lasttrennschalter oder feste Konfigurationen ohne Chargemeter).

Sobald `vehicle.phases` angegeben ist und die automatische Phasennutzungserkennung Daten liefert ist die Situation klar.
Wenn ich nichts auf die Schnelle übersehen habe steht wenn `"Phasennutzung" == 1 && vehicle.phases > 1` die Box auf 1p, sonst immer auf 3p.

Geschaltet werden muss nach einem Neustart auch nicht zwinged, da sich die Bilanzsituation am Gridmeter im Idealfall überhaupt nicht maßgeblich verändert hat. Es braucht ja nur eine Umschaltung (nach Timeout) wenn es quasi nicht mehr anders rauf oder runter geht.

andig, 2022-01-10:
> Sobald vehicle.phases angegeben ist und die automatische Phasennutzungserkennung Daten liefert ist die Situation klar.

Das ist ein bisschen am Thema vorbei bzw. darüber hinaus:
- vehicle.phases haben wir noch nicht und im zweifel ist die nicht gepflegt, unbekanntes Fahrzeug
- automatische Phasennutzungserkennung kann nur 3p erkennen, 1p kann diverse Ursachen haben, sei es Charger oder 1p Auto

> Ich finde die Idee nicht schlecht. Ich denke aber, es wäre intuitiver, dass eine Phasenerkennung ausschließlich dann stattfindet wenn in der Config nichts angegeben ist 

Leider auch nicht korrekt. Die Erkennung passiert _immer_ beim Laden. Die Konfig ist (ich wiederhole mich da) NUR eine Rechenhilfe. Sie trifft keine Festlegung und hat keinen steuernden Charakter. Und wenn man sie weg ließe bliebe wieder genau das Originalproblem übrig. 

> Man könnte ja theoretisch sowohl das feste Setzen von 1p bei 3p-kompatiblen Geräten oder auch das 3-p-Setzen in der Konfiguration argumentieren.

Könnte man, ist aber heute nicht so und ein anderes Thema. Hier gehts um den konkreten Fehler auf Basis aktueller Funktionalität.

premultiply, 2022-01-10:
Ja, und es ist stand jetzt kaputt. :)
Aus dem Grund, weil relevante Infos fehlen. Daher braucht es hier vehicle.maxPhases und lp.maxPhases. Dann wäre es klar.

Wenn nix angegeben ist muss man einen Default festlegen der ggf. auch angewandt wird.
Hier würde ich dann eher 1p nehmen da dass schlimmstenfalls nur zu einer Unterbrechung führt aber weder die Regelung stört und auch sonst nichts kaputt macht.

andig, 2022-01-10:
> Aus dem Grund, weil relevante Infos fehlen. 

Es fehlt v.a. die Info vom Charger. Sorry, aber am Thema vorbei. Entweder scopen wir die Diskussion hier oder ich mache das Thema zu.

premultiply, 2022-01-10:
Danke, ich bin raus.

andig, 2022-01-12:
Der aktuelle Zustand ist ja ziemlich unbefriedigend. Seit #1572 kann eine unklare Konfiguration zudem das Schaltverhalten permanent in Mitleidenschaft ziehen. Ich würde deshalb #2258 in die `nightly` Releases mergen. Auf der Basis können wir das dann weiter beobachten und anpassen.

Wenn ihr mit Logfiles helfen wollt dann denkt bitte an Folgendes:
- Logfile ab Programmstart
- Konkreten Zeitpunkt nennen zu dem der Fehler auftritt
- Relevanten Ladepunkt benennen

andig, 2022-01-12:
Waiting for feedback from nightly builds

closed 2022-01-15 by andig